espressif/arduino-esp32 #3182: How to fix OTA failing on ESPAsyncWebServer?

moothyknight, 2019-09-05:
## Hardware:
Board:							WeMos LOLIN32 
Core Installation version:			1.0.3-rc3
IDE name:						Arduino IDE
Flash Frequency:					80MHz
PSRAM enabled:                                    I think so? Whatever the default is
Upload Speed:			               921600
Computer OS:                                        Windows 10

### Description:
Describe your problem here
I'm trying to use Web OTA via ESPAsyncWebServer with a big program and custom partition. The OTA updater crashes when attempting to upload. Not sure if it's a hardware issue or a code issue.

This is using a custom sized partition from the default and includes the ESPAsyncWebServer and BLE libraries, though they are not active while the WiFi is active. OTA worked before both on regular webserver and the ESPAsyncWebServer but after I changed partition size and added the BLE support it seems to be crashing, any ideas? My sketch is about 1.4MB and I increased the partition size to 2MB, and removed the other partition in default.csv.

New default.csv
boards.txt max size limits changed to reflect this. The sketch builds fine through the IDE, but I can't upload new .bins
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x200000,
spiffs,   data, spiffs,  0x290000,0x170000,

While I'm here, mDNS only seems to be working on apple products for me, any fix for android/windows?

### Sketch:  (leave the backquotes for [code formatting]
```cpp

//Change the code below by your sketch
#include <Arduino.h>
#include <WiFi.h>
#include <ESPAsyncWebServer.h>
//#include <WiFiClient.h>
//#include <WebServer.h>
#include <ESPmDNS.h>
#include <Update.h>

AsyncWebServer server(80);

//Enter your WiFi SSID and PASSWORD
const char* host = "esp32";

char received;

size_t content_len;
unsigned long t_start,t_stop;

void handleUpdate(AsyncWebServerRequest *request) {
  char* html = "<h4>Upload compiled sketch .bin file</h4><form method='POST' action='/doUpdate' enctype='multipart/form-data'><input type='file' name='update'><input type='submit' value='Update'></form>";
  AsyncWebServerResponse *response = request->beginResponse_P(200, "text/html", html);
  request->send(response);
}

void handleDoUpdate(AsyncWebServerRequest *request, const String& filename, size_t index, uint8_t *data, size_t len, bool final) {
  //uint32_t free_space = (ESP.getFreeSketchSpace() - 0x1000) & 0xFFFFF000;
  if (!index){
    Serial.println("Update");
    content_len = request->contentLength();
    // if filename includes spiffs, update the spiffs partition
    int cmd = (filename.indexOf("spiffs") > -1) ? U_SPIFFS : U_FLASH;
    if (!Update.begin(UPDATE_SIZE_UNKNOWN, cmd)){
      Update.printError(Serial);
    }
  }
  if (Update.write(data, len) != len) {
    Update.printError(Serial);
  }
  if (final) {
    if (!Update.end(true)){
      Update.printError(Serial);
[... 76 lines cut ...]
Decoding 18 results
0x400926ec: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 707
0x4009291d: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 707
0x401646f9: is_safe_write_address at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/flash_ops.c line 126
0x40082bcf: spi_flash_erase_sector at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/flash_ops.c line 204 (discriminator 2)
0x400e3339: EspClass::flashEraseSector(unsigned int) at C:\Users\Joshua\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Esp.cpp line 210
0x400dcc1e: UpdateClass::_writeBuffer() at C:\Users\Joshua\Documents\Arduino\hardware\espressif\esp32\libraries\Update\src/Updater.cpp line 360
0x400dcddd: UpdateClass::write(unsigned char*, unsigned int) at C:\Users\Joshua\Documents\Arduino\hardware\espressif\esp32\libraries\Update\src/Updater.cpp line 360
0x400d45c2: handleDoUpdate(AsyncWebServerRequest*, String const&, unsigned int, unsigned char*, unsigned int, bool) at C:\Users\Joshua\AppData\Local\Temp\arduino_build_564701\sketch/WiFi_API.h line 298
0x400d4735: std::_Function_handler ::_M_invoke(std::_Any_data const&, AsyncWebServerRequest*&&, String const&, unsigned int&&, unsigned char*&&, AsyncWebServerRequest*&&, bool&&) at C:\Users\Joshua\AppData\Local\Temp\arduino_build_564701\sketch/WiFi_API.h line 425
: (inlined by) _M_invoke at c:\users\joshua\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0/functional line 1871
0x401d8d17: std::function ::operator()(AsyncWebServerRequest*, String const&, unsigned int, unsigned char*, unsigned int, bool) const at C:\Users\Joshua\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/StringArray.h line 73
: (inlined by) AsyncCallbackWebHandler::handleUpload(AsyncWebServerRequest*, String const&, unsigned int, unsigned char*, unsigned int, bool) at C:\Users\Joshua\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/WebHandlerImpl.h line 107
0x401d8bd6: AsyncWebServerRequest::_handleUploadByte(unsigned char, bool) at C:\Users\Joshua\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/WebRequest.cpp line 990
0x400d8376: AsyncWebServerRequest::_parseMultipartPostByte(unsigned char, bool) at C:\Users\Joshua\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/WebRequest.cpp line 990
0x400d9b2e: AsyncWebServerRequest::_onData(void*, unsigned int) at C:\Users\Joshua\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/WebRequest.cpp line 990
0x400d9d09: std::_Function_handler ::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, std::_Any_data const&, unsigned int&&) at C:\Users\Joshua\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/WebRequest.cpp line 990
: (inlined by) _M_invoke at c:\users\joshua\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0/functional line 1871
0x400dbdd1: std::function ::operator()(void*, AsyncClient*, void*, unsigned int) const at C:\Users\Joshua\Documents\Arduino\libraries\AsyncTCP-master\src/AsyncTCP.cpp line 983
: (inlined by) AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at C:\Users\Joshua\Documents\Arduino\libraries\AsyncTCP-master\src/AsyncTCP.cpp line 878
0x400dbe51: AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) at C:\Users\Joshua\Documents\Arduino\libraries\AsyncTCP-master\src/AsyncTCP.cpp line 983
0x400dc45a: _async_service_task(void*) at C:\Users\Joshua\Documents\Arduino\libraries\AsyncTCP-master\src/AsyncTCP.cpp line 983
: (inlined by) _async_service_task at C:\Users\Joshua\Documents\Arduino\libraries\AsyncTCP-master\src/AsyncTCP.cpp line 180
0x4008e615: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 355 (discriminator 1)
```

sansillusion, 2019-09-05:
To make it work under windows install apple Bonjour service.

stickbreaker, 2019-09-05:
> This is using a custom sized partition from the default and includes the ESPAsyncWebServer and BLE libraries, though they are not active while the WiFi is active. OTA worked before both on regular [...]

>New default.csv
boards.txt max size limits changed to reflect this. The sketch builds fine through the IDE, but I can't upload new .bins
Name, Type, SubType, Offset, Size, Flags
nvs, data, nvs, 0x9000, 0x5000,
otadata, data, ota, 0xe000, 0x2000,
app0, app, ota_0, 0x10000, 0x200000,
spiffs, data, spiffs, 0x290000,0x170000,

For OTA to function, you have to have TWO app partitions APP0, APP1.  The update is using one while writing to the other.  After is successfully loads the new app, it marks it active and reboots.  When you removed the second APP partition it has no place to store the update.

and your `default.csv` has invalid values.
>Name, Type, SubType, Offset, Size, Flags
nvs, data, nvs, 0x9000, 0x5000,
otadata, data, ota, 0xe000, 0x2000,
app0, app, ota_0, 0x10000, 0x200000,
x,x,0x210000,0xF0000,  _**void, unused space??**_
spiffs, data, spiffs, 0x300000,0x170000, _**how big is your FLASH? this is 0x70000 past the end of a 4MB**_

Chuck.

moothyknight, 2019-09-06:
Thanks, this default.csv plus respective change in brds.txt did it for me. I still have 800kb of SPIFFs too, nice!

# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x180000,
app1,	  app,  ota_1,   0x190000, 0x180000
spiffs,   data, spiffs,  0x310000,0xC3500,

closed 2019-09-06 by moothyknight