espressif/arduino-esp32 #3183: I can't use rtos on esp32 module

Sanketd420, 2019-09-06:
Ok I have two esp32 module which i bought from banggood and I for the past couple of days I have been trying to implement rtos functionality in it which each board has built in.I upgraded the firmware via OTA but still I couldn't get it to work.So here is the problem
1.I have a esp 32 board from DOIT (ESP-F DEVKIT V4) and I use the board(in the boards manager)
NODEMCU 1.0(ESP-12E) beacuse i couldnt upload to the esp using any other boards from boards manager.I really need to implement rtos and using the Free_RTOS library from the arduino library manager I couldnt do it as it gave me an error that it couldnt find avr/io.h header file which is commonly used for arduino microprocessor and maybe the tensilica dont support avr/io.h.
2.The second board i have i tried to upload to it using the wrover board from the board manager and i succeded with that board(IT was an AI-thinker board).Now in this mess i cant find a way to access the rtos functionality of this board.I tried selecting this board to upload to the DO-IT board and i Failed with the following error.
############################################################
Arduino: 1.8.8 (Windows 10), Board: "ESP32 Wrover Module, Default, QIO, 80MHz, 921600, None"

Sketch uses 239600 bytes (18%) of program storage space. Maximum is 1310720 bytes.
Global variables use 13396 bytes (4%) of dynamic memory, leaving 314284 bytes for local variables. Maximum is 327680 bytes.
esptool.py v2.6
Serial port COM4
Connecting....

A fatal error occurred: Timed out waiting for packet header
A fatal error occurred: Timed out waiting for packet header

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
####################################
I cant seem to find out what the problem is 
This is the error i get if i use the DO-IT board with rtos functionality(bear in mind this code works if i select the wrover module in boards manager):
######################
Arduino: 1.8.8 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

core2duo_:5:1: error: 'TaskHandle_t' does not name a type

 TaskHandle_t Task1, Task2;

 ^

C:\Users\Rutuja\Documents\Arduino\core2duo_\core2duo_.ino: In function 'void setup()':

core2duo_:30:6: error: 'Task1' was not declared in this scope

     &Task1,                   /* Task handle to keep track of created task */

      ^

core2duo_:31:6: error: 'xTaskCreatePinnedToCore' was not declared in this scope

     1);                    /* Core */

      ^

core2duo_:40:6: error: 'Task2' was not declared in this scope

     &Task2,                   /* Task handle to keep track of created task */

      ^

C:\Users\Rutuja\Documents\Arduino\core2duo_\core2duo_.ino: In function 'void loop()':

core2duo_:48:46: error: 'xPortGetCoreID' was not declared in this scope

   Serial.printf("Core %d\n ", xPortGetCoreID());

                                              ^

exit status 1
'TaskHandle_t' does not name a type

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
###################################
It seems that rtos is not working for this board

Here is the rtos code(bear in mind this code doesn't even compile on the DO-IT board but compiles if i select wrover module)
###################################

#define LED_BUILTIN 14

TaskHandle_t Task1, Task2;

void codeForTask1( void * parameter )
{
  for (;;);
}

void codeForTask2( void * parameter )
{
  for (;;);
}

// the setup function runs once when you press reset or power the board
void setup() {
  Serial.begin(115200);
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);

  xTaskCreatePinnedToCore(
    codeForTask1,            /* Task function. */
    "Task1",                 /* name of task. */
    1000,                    /* Stack size of task */
    NULL,                     /* parameter of the task */
    1,                        /* priority of the task */
    &Task1,                   /* Task handle to keep track of created task */
    1);                    /* Core */
  delay(500);

  xTaskCreatePinnedToCore(
    codeForTask2,           /* Task function. */
    "Task2",        /* name of task. */
    1000,                    /* Stack size of task */
    NULL,                     /* parameter of the task */
    1,                        /* priority of the task */
    &Task2,                   /* Task handle to keep track of created task */
    1);                    /* Core */                  /* Core */
}

// the loop function runs over and over again forever
void loop() {
  Serial.printf("Core %d\n ", xPortGetCoreID());
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
#######################################

knifter, 2019-09-06:
Maybe you should include the FreeRTOS header files? For example:
    #include "freertos/FreeRTOS.h"
    #include "freertos/task.h"

atanisoft, 2019-09-06:
It sounds like you don't have esp32 boards installed in the ide. You don't need to include any special library for FreeRTOS as it is included in the esp32 builds automatically.

atanisoft, 2019-09-06:
The "doit" boards are possibly esp8266 board definitions. The WROVER is an esp32. You should use the esp32 dev module board type instead likely.

atanisoft, 2019-09-06:
FreeRTOS is bundle with the esp32 board types. Be sure to pick one of the esp32 boards and not esp8266 board types (esp-12, etc are esp8266 based).

closed 2019-11-19 by stale[bot]