espressif/arduino-esp32 #3185: Timed out waiting for packet header

jdhurwitz, 2019-09-06:
### Hardware:
Board:							ESP32 - wroom 
Core Installation version:			?
IDE name:						terminal
Flash Frequency:					40Mhz
PSRAM enabled:                                    ?
Upload Speed:					115200
Computer OS:                                        Mac OS X

### Description:
I'm getting the error message below. This is on a custom board we designed based off of the [adafruit huzzah32 feather board](https://learn.adafruit.com/adafruit-huzzah32-esp32-feather/downloads). I've tried holding down the reset button before issuing this command and then letting it go, but it makes no difference. 

I have some pins connected to voltage divider bridges (where a variable resistor is a pressure sensor), and these are currently pulled to ground since the sensors aren't connected. Are there some pins that need to be held high instead of low in order for this to boot and flash?

The thing I don't understand is that we're able to read the chip, its features,  and the MAC address.

```
Umars-MacBook-Pro-2:~ umar$ esptool.py --chip esp32 --port /dev/tty.SLAB_USBtoUART --baud 115200 --no-stub write_flash -z 0x1000 ~/Downloads/esp32-20190905-v1.11-274-g06661890d.bin
esptool.py v2.6
Serial port /dev/tty.SLAB_USBtoUART
Connecting........_
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 24:6f:28:46:31:7c
Enabling default SPI flash mode...
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Erasing flash...
Compressed 1204416 bytes to 756253...

A fatal error occurred: Timed out waiting for packet header
```

beegee-tokyo, 2019-09-06:
You can check for special GPIOS [here](https://desire.giesecke.tk/index.php/2018/07/06/reserved-gpios/)

jdhurwitz, 2019-09-06:
> You can check for special GPIOS [here](https://desire.giesecke.tk/index.php/2018/07/06/reserved-gpios/)

Why would 6-11 be unusable (according to the link) yet called GPIO? The point of GPIO is so that you can use them :) 

How are they reserved for the flash? So if I have these connected to a 10k pulldown, it won't work? They're not being driven by any signal right now, just passively connected.

lbernstone, 2019-09-06:
Would you like to argue about semantics, or would you like your board to work?  GPIO 6-11 are used by the SPI flash.  You cannot have them connected to pulldowns.

beegee-tokyo, 2019-09-06:
One of the mysteries, these GPIOs are used to connect the internal memory to the CPU and should not ever be exposed, but on most boards they are. If you connect these GPIOs to anything, the CPU might not be able to read or write to the Flash memory that will contain your application.

You have to live with it.

beegee-tokyo, 2019-09-06:
And to make it worse, if you use a Wrover version of the ESP32 with additional SPI RAM, there are two more GPIOs that you cannot connect to, but still most Wrover breakout  boards I got are exposing them as if they are available.
Get used to it and investigate before you make a custom board. I fall into the same trap with a custom board before, costing me 200 US$ to get new prototypes. Luckily my customer paid for it.

jdhurwitz, 2019-09-06:
Damn. Ok I'll try desoldering the pulldowns on those pins when I'm back home and checking.

beegee-tokyo, 2019-09-06:
Yeah I know, the rules are to not leave any "unused" pins open for EMC reasons. But these pins are not unused unfortunately.

stickbreaker, 2019-09-06:
@jdhurwitz 
>Why would 6-11 be unusable (according to the link) yet called GPIO? The point of GPIO is so that you can use them :)

From the hardware's point of view they are General Purpose Input Output. The Software environment IDF/Arduino has decided to use them for FLASH/PSRAM access.

>How are they reserved for the flash? So if I have these connected to a 10k pulldown, it won't work? They're not being driven by any signal right now, just passively connected.

The FLASH/PSRAM is accessed at up to 80MHZ, a 10k restive load on a 80MHZ is not the same load at zero frequency(DC). This graph is from [passive-components.eu](https://passive-components.eu/resistors-voltage-and-frequency-dependence/) it shows the relationship between Z(effective resistance) and Rdc(DC resistance), Z/Rdc is dependent on frequency.  I assume you are using carbon resistors (graph label 1).  So at 80MHz the effective resistance is less than 5% or 500 ohm, Way Way too much load for the 80MHZ bus to function.
[image]
 
Chuck.

jdhurwitz, 2019-09-08:
Thanks for the explanations! I'll try desoldering those pulldowns when I'm home next week.

jdhurwitz, 2019-09-08:
> Yeah I know, the rules are to not leave any "unused" pins open for EMC reasons. But these pins are not unused unfortunately.

Also, if I use BLE, are there any pins that have to be left open for this to function properly? I didn't see anything on the data sheet saying as much but I figured it would be good to check.

beegee-tokyo, 2019-09-09:
I am not aware of any GPIOs that needs to open for BLE.

jdhurwitz, 2019-09-14:
Leaving those GPIOs open allowed me to flash it! :)

closed 2019-09-14 by jdhurwitz