espressif/arduino-esp32 #3186: STA mode does not work

nightflyer88, 2019-09-06:
### Hardware:
Board:							ESP32 Pico Kit
Core Installation version:			1.0.3-rc3
IDE name:						Arduino IDE
Upload Speed:					921600
Computer OS:                                        Mac OSX

### Description:
STA mode does not work. If I use the example "SimpleWiFiServer" and enter my SSID and PW, the ESP32 can not connect to my network. I've checked SSID and PW many times.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

#include <WiFi.h>

const char* ssid     = "???";
const char* password = "???";

WiFiServer server(80);

void setup()
{
    Serial.begin(115200);
    pinMode(5, OUTPUT);      // set the LED pin mode

    delay(10);

    // We start by connecting to a WiFi network

    Serial.println();
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(ssid);

    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected.");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
    
    server.begin();

}

int value = 0;

void loop(){
 WiFiClient client = server.available();   // listen for incoming clients

  if (client) {                             // if you get a client,
    Serial.println("New Client.");           // print a message out the serial port
    String currentLine = "";                // make a String to hold incoming data from the client
    while (client.connected()) {            // loop while the client's connected
      if (client.available()) {             // if there's bytes to read from the client,
        char c = client.read();             // read a byte, then
        Serial.write(c);                    // print it out the serial monitor
        if (c == '\n') {                    // if the byte is a newline character

          // if the current line is blank, you got two newline characters in a row.
          // that's the end of the client HTTP request, so send a response:
          if (currentLine.length() == 0) {
            // HTTP headers always start with a response code (e.g. HTTP/1.1 200 OK)
            // and a content-type so the client knows what's coming, then a blank line:
            client.println("HTTP/1.1 200 OK");
            client.println("Content-type:text/html");
            client.println();

            // the content of the HTTP response follows the header:
            client.print("Click <a href=\"/H\">here</a> to turn the LED on pin 5 on.<br>");
            client.print("Click <a href=\"/L\">here</a> to turn the LED on pin 5 off.<br>");

            // The HTTP response ends with another blank line:
            client.println();
            // break out of the while loop:
            break;
          } else {    // if you got a newline, then clear currentLine:
            currentLine = "";
          }
        } else if (c != '\r') {  // if you got anything else but a carriage return character,
          currentLine += c;      // add it to the end of the currentLine
        }

        // Check to see if the client request was "GET /H" or "GET /L":
        if (currentLine.endsWith("GET /H")) {
          digitalWrite(5, HIGH);               // GET /H turns the LED on
        }
        if (currentLine.endsWith("GET /L")) {
          digitalWrite(5, LOW);                // GET /L turns the LED off
        }
      }
    }
    // close the connection:
    client.stop();
    Serial.println("Client Disconnected.");
  }
}
```

atanisoft, 2019-09-06:
@nightflyer88 Please enable verbose debug logging and retry your test. We will need to see the serial console to see why it is not connecting to your SSID.

nightflyer88, 2019-09-07:
```
Connecting to:Achtung Virus !! nicht anklicken
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[E][WiFiSTA.cpp:124] begin(): SSID too long or missing!
.......................
```
SSID is exactly 32 characters long. Other sketches with ESP8266 and the same WiFi are working properly.
If I connect the ESP32 to another WiFi that has a shorter SSID, it works.

beegee-tokyo, 2019-09-07:
I think your problem is that your SSID is 32 bytes long. I didn't check, but I think the ESP32 accepts only 31 bytes and the 32nd byte is 0.

stickbreaker, 2019-09-07:
@nightflyer88, @beegee-tokyo is correct.  the ESP32 WiFi treats the SSID as a 31byte cstr. The Spec IEEE 802.11-2007 specifies it as a 32 byte field. By specification it is 32 octets. By convention it is native language. 
The esp32 implementation treats it as cstr(zero terminated) if you need 32bytes of random characters you will have to rewrite the WiFi implementation.  And change all access to a 32byte octet field(no length, padded out to 32bytes with you favorite fill character).

Chuck.

beegee-tokyo, 2019-09-07:
And that limitation is not only on the ESP32. I have one cheap Chinese WiFi AP adapter where the SSID is limited to 31 bytes as well.

atanisoft, 2019-09-07:
File an issue against esp-idf if you want to have the full 32 octet non-terminated string for SSID.

nightflyer88, 2019-09-07:
So it should work: In the file WiFiSTA.cpp in line 123, change the value 31 to 32. In the ESP8266 lib, the value is also 32.

```
    if(!ssid || *ssid == 0x00 || strlen(ssid) > 32) {
        log_e("SSID too long or missing!");
        return WL_CONNECT_FAILED;
    }
```

It works for me:
```
Connecting to:Achtung Virus !! nicht anklicken
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
..[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
....[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.14, MASK: 255.255.255.0, GW: 192.168.1.1
.
WiFi connected.
IP address: 
192.168.1.14
```

atanisoft, 2019-09-07:
The reason it cuts off at 31 is due to the underlying ```wifi_config_t``` only stores 32 bytes. It may work for you but that is only because the string termination character overflows the ssid array into the password array. This is not a problem *ONLY* because password is copied next.

This should likely be fixed by switching from strcpy to memcpy with strlen() instead.

beegee-tokyo, 2019-09-08:
And what do you get after this change when you call `WiFi.SSID()`? 
Because this function returns the SSID as
`return String(reinterpret_cast<char*>(info.ssid));`
Without the trailing 0 it most likely returns a mix of SSID and password together. Because it returns everything until the first 0 and that would be at the end of your password which is the next entry in the structure.

atanisoft, 2019-09-08:
Right, it needs to be fixed at the esp-idf side first. I'd suggest an issue be filed there as this is an inconsistency in the API, wifi scan has 33 bytes as the ssid as example.

beegee-tokyo, 2019-09-08:
Until fixed in the IDF which can take a long time you could change WiFi.SSID() to
```
char _ssid[33];
snprintf(_ssid, 33,"%s",reinterpret_cast<char*>(info.ssid)),;
return String(_ssid) ;
```
Which would limit the number of used characters to 32 and add a 0 terminator. 

Not tested, just an idea.

atanisoft, 2019-09-08:
Shouldn't that be 32 in the snprintf?

stickbreaker, 2019-09-08:
@atanisoft No, the 33 is total buffer length(of _ssid).  It includes the terminator, `snprintf()` is smart enough to reserve space for the terminator.  By setting it to 33, it will only copy 32 characters, then stop and insert the terminator.  I think the logic of `snprintf()` is to copy until buffer filled or NULL encountered.

As an aside, what should the fill character be?  is `info` zero filled?  That would match with a cstr.

Chuck.

atanisoft, 2019-09-08:
Instead of snprintf it may be easier to use memcpy and add a call to set the last byte as '\0'

beegee-tokyo, 2019-09-08:
But then you might copy more than required, because you don't know how long the SSID is. Might be only 8 characters long. So you might need first to check the length. snprintf does that already for you.

atanisoft, 2019-09-08:
Perhaps, but the underlying data structure is not guaranteed to be null terminated and has a max limit of 32. Copying 8 or 32 bytes of memory isn't going to be a big difference as far as speed. But calling strlen or similar on an unterminated char array could be just as bad as limiting to 31 characters as it does today.

beegee-tokyo, 2019-09-08:
But before returning the SSID you still need to find the length somehow.
When the SSID is written into the structure and it is less then 32 bytes there must be a trailing 0. Otherwise you would always get garbage after the last character of the SSID. Because String(char[]) expects a trailing 0.

atanisoft, 2019-09-08:
The structure is set to zero before the ssid is copied into it initially. By using memcpy() with limit of 32 and forcing the last byte in the target array to '\0' (string null terminator) it will work just fine for string display etc.

So determining if the actual SSID is 8 or 32 bytes long is not relevant entirely.

beegee-tokyo, 2019-09-08:
Yep, you're right. So memcpy is faster.

stickbreaker, 2019-09-08:
Now here is a stupid idea:  Create a SSID with embedded NULLS?

beegee-tokyo, 2019-09-08:
You are cruel!

How would any phone, tablet, laptop be ever able to connect to it? You cannot enter NULL on a keyboard.

atanisoft, 2019-09-08:
Is there anything in the spec that says you can't have a null character in the SSID? It would be an interesting approach to securing the SSID...

beegee-tokyo, 2019-09-08:
What I know is from
https://serverfault.com/questions/45439/what-is-the-maximum-length-of-a-wifi-access-points-ssid

> According to the documentation of the standard, the length of an SSID should be a maximum of 32 characters (32 octets, normally ASCII letters and digits, though the standard itself doesn't exclude values).
> 
> Some access point/router firmware versions use null-terminated strings and accept only 31 characters

But I think ESP-IDF will cut the SSID at the first NULL it finds or after 32 chars

atanisoft, 2019-09-08:
> But I think ESP-IDF will cut the SSID at the first NULL it finds or after 32 chars

This would make sense, but it is hard to know for certain since the WiFi driver code is closed-source.

beegee-tokyo, 2019-09-08:
Easy to test. Setup an AP on one ESP32 with a NULL inside the SSID and try to scan it from another ESP32 and try to connect to it.

stickbreaker, 2019-09-08:
@beegee-tokyo Check out this page on WiKi [Help: Entering Special Characters #Windows - ALT](https://en.wikipedia.org/wiki/Help:Entering_special_characters#Windows_-_Alt_code) and [Help: Entering Special Characters #Windows](https://en.wikipedia.org/wiki/Help:Entering_special_characters#Windows).

beegee-tokyo, 2019-09-08:
Theoretically possible. As I said let's try it :smile:

stickbreaker, 2019-09-08:
This idea is equivalent to SQL injection attacks.

beegee-tokyo, 2019-09-08:
Back to the "real" issue, anybody willing to test and do a pull-request? The poor German guy reporting this issue must be going crazy over our brainstorming here.

negativekelvin, 2019-09-13:
https://github.com/espressif/esp-idf/issues/3762

closed 2019-11-26 by stale[bot]