espressif/arduino-esp32 #3192: Arduino board keeps resetting, cannot upload new sketch (NOT timeout)

Oliwae, 2019-09-09:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					40Mhz
PSRAM enabled:                                          ?no?
Upload Speed:						115200 / 230400 / 921600
Computer OS:                                            Windows 7

### Description:
First let me share the method I am used to upload sketch to Arduino ESP32 board
1. Connect Arduino board to computer through USB port
2. Press <Upload> in computer
3. Once "Connecting..." is displayed, hold <BOOT> on Arduino board
4. Release <BOOT> once "Connecting..." dissappear
5. Done
This way succeed in uploading new sketch to those ESP32 I bought online before July 2019
Due to some reasons, I bought 2 more last week
I am not able to upload new sketch to the 2 newly bought Arduino boards
It DOES pass the upload procedure (I think, as it does NOT come out timeout error)
But the serial monitor keeps writing the below message, no matter what sketch I uploaded

###
```
14:51:44.103 -> rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
14:51:44.103 -> configsip: 0, SPIWP:0xee
14:51:44.103 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
14:51:44.103 -> mode:DIO, clock div:1
14:51:44.103 -> load:0x3fff0018,len:4
14:51:44.103 -> load:0x3fff001c,len:1100
14:51:44.103 -> load:0x40078000,len:9232
14:51:44.103 -> load:0x40080400,len:6400
14:51:44.103 -> entry 0x400806a8
14:51:44.103 -> ets Jun  8 2016 00:22:57
14:51:44.103 -> 
14:51:44.103 -> rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
14:51:44.103 -> configsip: 0, SPIWP:0xee
14:51:44.136 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
14:51:44.136 -> mode:DIO, clock div:1
14:51:44.136 -> load:0x3fff0018,len:4
14:51:44.136 -> load:0x3fff001c,len:1100
14:51:44.136 -> load:0x40078000,len:9232
14:51:44.136 -> load:0x40080400,len:6400
14:51:44.136 -> entry 0x400806a8
14:51:44.136 -> ets Jun  8 2016 00:22:57
```
###

I am tried many ways found online like holding <BOOT> before <Upload>, holding <BOOT> then press <ENABLE> then release <BOOT>, lower the uploading frequency, etc.
but all fail to solve the problem

https://randomnerdtutorials.com/esp32-troubleshooting-guide/

Note that if I keep everything unchanged, just change the ESP32 Arduino board (newly bought) to the old one (bought before July) by removing in USB port
Everything works!

Thanks if anyone have some idea of comment.

me-no-dev, 2019-09-09:
which version of this core are you running?
You can try to change the flash speed in the board menu, try the new core (in case your esp is limited to 160MHz, but I doubt it). Flash Frequency set to 40 could also help... it halts too early to say

Oliwae, 2019-09-09:
> which version of this core are you running?
> You can try to change the flash speed in the board menu, try the new core (in case your esp is limited to 160MHz, but I doubt it). Flash Frequency set to 40 could also help... it halts too early to say

Thank you for your reply

May I ask how to check the version of the core? (sorry I am a newbie of arduino, especially hardware part)
The flash frequency is 40 MHz original and I do not change it.

lbernstone, 2019-09-09:
Try [erasing the flash](https://github.com/espressif/esptool#erase-flash-erase_flash--erase-region) before you upload.

me-no-dev, 2019-09-09:
how did you install the ESP32 support?

me-no-dev, 2019-09-09:
also, what board did you select?

Oliwae, 2019-09-10:
> Try [erasing the flash](https://github.com/espressif/esptool#erase-flash-erase_flash--erase-region) before you upload.

Thank you for your reply.
I will try it next time. (Are there any other ways do not need python but only arduino IDE to erase...?)

Oliwae, 2019-09-10:
> how did you install the ESP32 support?

I installed the esp32 library from Arduino IDE.

> also, what board did you select?

I think ESP32S?
I bought from this website
https://detail.tmall.com/item.htm?spm=a230r.1.14.9.139064d3ZPnvux&id=546381493593&cm_id=140105335569ed55e27b&abbucket=19

lbernstone, 2019-09-10:
The esp32 installer for arduino IDE includes esptool.  Search your hard drive for esptool.exe if you are on windows.

kapyaar, 2019-09-13:
I have had similar issues. Full erase seems to help. Here is how you can do it. There is the esptool.exe located on your machine. For me, it is at 

`C:\Users\AVC\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1/`

So you can open a command prompt (May be run as administrator if required), then Issue the command 

`C:\Users\AVC\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1/esptool.exe --chip esp32 --port COM4 erase_flash`

Oliwae, 2019-09-16:
> I have had similar issues. Full erase seems to help. Here is how you can do it. There is the esptool.exe located on your machine. For me, it is at [...]

Thanks it seems work to erase flash
The serial monitor now show
'''
10:50:38.376 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
10:50:38.376 -> flash read err, 1000
10:50:38.376 -> ets_main.c 371 
10:50:38.751 -> ets Jun  8 2016 00:22:57
10:50:38.751 -> 
10:50:38.751 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
10:50:38.751 -> flash read err, 1000
10:50:38.751 -> ets_main.c 371 
10:50:39.123 -> ets Jun  8 2016 00:22:57
10:50:39.123 -> 
10:50:39.123 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
10:50:39.123 -> flash read err, 1000
10:50:39.123 -> ets_main.c 371 
10:50:39.496 -> ets Jun  8 2016 00:22:57
'''

But the problem still here,  I do not need to hold/press BOOT to upload the sketch...It pass the uploading process but fail to upload the new sketch

closed 2019-11-29 by stale[bot]