espressif/arduino-esp32 #3193: WiFiClientSecure::write only sends the first 16KB

marcelstoer, 2019-09-09:
### Description:
Shouldn't [`WiFiClientSecure::write`](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp#L183) send the whole buffer even beyond SSL fragment length? Currently it only writes up to mbedTLS maximum fragment length which is 16KB by default.

According to [ARM mbed TLS](https://tls.mbed.org/api/ssl_8h.html#a5bbda87d484de82df730758b475f32e5) `mbedtls_ssl_write` has the following limitation

> If the requested length is greater than the maximum fragment length (either the built-in limit or the one set or negotiated with the peer), then:
>- with TLS, less bytes than requested are written.

It appears that `mbedtls_ssl_write` is only called once at https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFiClientSecure/src/ssl_client.cpp#L274

### Sketch
While the (simplified) workaround below works ok it does feel like such a use case should be handled by `WiFiClientSecure` internally. Opinions?

```cpp
size_t sent = 0; 
size_t remaining = fileSize; 
size_t offset = 0;

while (true) {
  sent = client->write(buffer + offset, remaining);
  remaining = remaining - sent;
  offset += sent;
  if (remaining <= 0 || sent <= 0) {
    break;
  }
}
```

marcelstoer, 2019-11-08:
@me-no-dev according to the commit history you might be familiar with the subject?

marcelstoer, 2020-01-08:
@me-no-dev what can I do to get some feedback from a developer to find out if my expectations are justified or not?

me-no-dev, 2020-01-20:
This should definitelly not be the case and I will put it on my list of things to check. Doing such while loops in FreeRTOS might not be a good idea, but I need to have a better look at what goes underneath to know if this can do or not.

marcelstoer, 2020-03-20:
"Hey bot, shut up. The grown-ups are talking." Not frequently but they do 😜

marcelstoer, 2020-05-19:
nudge

atanisoft, 2020-05-19:
@marcelstoer perhaps submit a PR with your proposal?

marcelstoer, 2020-05-19:
Re: PR; Normally not an issue but I'm not familiar with the inner workings of the affected class(es) to make a meaningful proposal.

Besides @me-no-dev stated that

> Doing such while loops in FreeRTOS might not be a good idea

This confirms my initial suspicion that there must be a better solution.

marcelstoer, 2020-07-18:
> This should definitelly not be the case and I will put it on my list of things to check.

@me-no-dev is it still there?

closed 2020-11-30 by stale[bot]

everslick, 2021-05-09:
hehe, I feel your pain. I had the pretty same experience in https://github.com/espressif/arduino-esp32/issues/2494