espressif/arduino-esp32 #3194: PWM question

tomiisp, 2019-09-10:
Hi,

maybe this is not right place but maybe someone help/provide clue.
I need (on ESP32) generate PWM signal (400Hz), this is fine but I need to sync this PWM to AC mains 50Hz (via optocoupler).
Can anyone point me how to do on ESP32/Arduino ? In old design (M2560) I have connected "sync" signal to INT pin and just reset/restart Timer what generates PWM on each rising edge. How to make it on ESP32?
And second question, I also need to measure duty cycle of PWM signal (again ~400Hz), any idea/example?
All what I need to do is convert/measure input PWM signal (not stable, ~400Hz, not synced to AC) to stable 400Hz PWM synced to AC frequency. (Duty cycle at input/output is recalculated/optimized but that is not important for this question)

Thanks for any help or idea
/Tomi

lbernstone, 2019-09-10:
Please ask on esp32.com.  This forum is for issues with code.

tomiisp, 2019-09-18:
Hi,

after some experiments I was able to set PWM/SYNC with mcpwm in Arudino, seems this works fine but I also try to setup Capture mode, I follow example for mcpwm but I stuck on reading/setting registers, problem is with this lines:

`static mcpwm_dev_t *MCPWM[2] = {&MCPWM0, &MCPWM1};`
it end with error: 'mcpwm_dev_t' does not name a type

and of course this lines is affected :
`mcpwm_intr_status = MCPWM[MCPWM_UNIT_0]->int_st.val;` 
`MCPWM[MCPWM_UNIT_0]->int_clr.val = mcpwm_intr_status;`
`MCPWM[MCPWM_UNIT_0]->int_ena.val = CAP0_INT_EN;`

have anyone idea what is wrong or how to fix this (in Arduino) ?

Thanks
/Tomi

markiv25, 2019-09-26:
Did you get success?  i am too tryng ac dimmer but theres lot of flickering

closed 2019-12-09 by stale[bot]