espressif/arduino-esp32 #3195: rst:0x1 (POWERON_RESET),boot:0x0 (DOWNLOAD_BOOT(UART0/UART1/SDIO_FEI_FEO_V2))

OmarAlkassab, 2019-09-10:
I used Bare ESP-WROOM-32, And I'm trying to upload the code using Arduino IDE (also tested with PlatformIO) but the module didn't respond and got the following message:

> Sketch uses 201244 bytes (15%) of program storage space. Maximum is 1310720 bytes.
> Global variables use 13516 bytes (4%) of dynamic memory, leaving 314164 bytes for local variables. Maximum is 327680 bytes.
> esptool.py v2.6
> Serial port COM4
> Connecting........_____....._____....._____....._____....._____....._____.....____An error occurred while uploading the sketch
> _
> 
> A fatal error occurred: Failed to connect to ESP32: Invalid head of packet (0x42)

please note that when I set GPIO0 to GND and reset the module, the serial of the module print the following message:

> ets Jun  8 2016 00:22:57
> 
> rst:0x1 (POWERON_RESET),boot:0x0 (DOWNLOAD_BOOT(UART0/UART1/SDIO_FEI_FEO_V2))

However when the GPIO0 is High and perform a Reset of the Chip I got this message:

> ␛[0;32mI (10) boot: ESP-IDF v3.0.1 2nd stage bootloader␛[0m
> ␛[0;32mI (10) boot: compile time 05:26:56␛[0m
> ␛[0;32mI (10) boot: Enabling RNG early entropy source...␛[0m
> ␛[0;32mI (13) boot: SPI Speed      : 40MHz␛[0m
> ␛[0;32mI (18) boot: SPI Mode       : DIO␛[0m
> ␛[0;32mI (22) boot: SPI Flash Size : 4MB␛[0m
> ␛[0;32mI (26) boot: Partition Table:␛[0m
> ␛[0;32mI (29) boot: ## Label            Usage          Type ST Offset   Length␛[0m
> ␛[0;32mI (36) boot:  0 phy_init         RF data          01 01 0000f000 00001000␛[0m
> ␛[0;32mI (44) boot:  1 otadata          OTA data         01 00 00010000 00002000␛[0m
> ␛[0;32mI (51) boot:  2 nvs              WiFi data        01 02 00012000 0000e000␛[0m
> ␛[0;32mI (59) boot:  3 at_customize     unknown          40 00 00020000 000e0000␛[0m
> ␛[0;32mI (66) boot:  4 ota_0            OTA app          00 10 00100000 00180000␛[0m
> ␛[0;32mI (74) boot:  5 ota_1            OTA app          00 11 00280000 00180000␛[0m
> ␛[0;32mI (81) boot: End of partition table␛[0m
> ␛[0;32mI (86) boot: No factory image, trying OTA 0␛[0m
> ␛[0;32mI (90) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x1fe00 (130560) map␛[0m
> ␛[0;32mI (145) esp_image: segment 1: paddr=0x0011fe28 vaddr=0x3ffc0000 size=0x001e8 (   488) load␛[0m
> ␛[0;32mI (145) esp_image: segment 2: paddr=0x00120018 vaddr=0x400d0018 size=0xddcc0 (908480) map␛[0m
> ␛[0;32mI (470) esp_image: segment 3: paddr=0x001fdce0 vaddr=0x3ffc01e8 size=0x03010 ( 12304) load␛[0m
> ␛[0;32mI (475) esp_image: segment 4: paddr=0x00200cf8 vaddr=0x40080000 size=0x00400 (  1024) load␛[0m
> ␛[0;32mI (477) esp_image: segment 5: paddr=0x00201100 vaddr=0x40080400 size=0x0eba0 ( 60320) load␛[0m
> ␛[0;32mI (510) esp_image: segment 6: paddr=0x0020fca8 vaddr=0x400c0000 size=0x00064 (   100) load␛[0m
> ␛[0;32mI (520) boot: Loaded app from partition at offset 0x100000␛[0m
> ␛[0;32mI (520) boot: Disabling RNG early entropy source...␛[0m
> Bin version(Wroom32):1.1.1
> I (626) wifi: wifi firmware version: bffcf7f
> I (626) wifi: config NVS flash: enabled
> I (626) wifi: config nano formating: disabled
> I (635) wifi: Init dynamic tx buffer num: 32
> I (636) wifi: Init data frame dynamic rx buffer num: 32
> I (636) wifi: Init management frame dynamic rx buffer num: 32
> I (642) wifi: wifi driver task: 3ffdf688, prio:23, stack:4096
> I (647) wifi: Init static rx buffer num: 10
> I (651) wifi: Init dynamic rx buffer num: 32
> I (655) wifi: wifi power manager task: 0x3ffe3d38 prio: 21 stack: 2560
> I (688) wifi: mode : softAP (3c:71:bf:6e:1d:11)
> I (695) wifi: mode : sta (3c:71:bf:6e:1d:10) + softAP (3c:71:bf:6e:1d:11)
> I (699) wifi: mode : softAP (3c:71:bf:6e:1d:11)

Anyone can help me please?
@igrr

lbernstone, 2019-09-10:
https://randomnerdtutorials.com/esp32-troubleshooting-guide

OmarAlkassab, 2019-09-10:
I realized what you said (pulling down GPIO0 while upload, and touch the enable pin to ground while the dots appears to "wake up" the board, and set upload parameters to DIO/40Mhz/115200 baud) and still get the error
[image]

lbernstone, 2019-09-10:
When you say "bare board", you mean there is nothing connected but 3V3, GPIO0, RX, TX, EN and GND (V0RTEG)?  What sort of USB->UART device are you using?

OmarAlkassab, 2019-09-10:
Yes I meant what you said.
I use Silicon Labs CP210x USB to UART Bridge.

lbernstone, 2019-09-10:
There isn't much more to it than that.  I would verify that your reset is working.  Watch in the serial monitor while you touch EN to ground and make sure it does reset.  I have many devices with all the different revisions of the chip which work like this.

OmarAlkassab, 2019-09-10:
I noticed according the following tutorial (https://randomnerdtutorials.com/esp32-pinout-reference-gpios/) in Strapping pins section, that I have 2 violations of the rule (I don’t know if these violations are the cause)
I have GPIO15 and GPIO5 pulled down by 10Kohm resistors to drive 2 transistors gates (I tested the upload  on my custom board).
Can you assert that the problem is that? Tomorrow I will desolder the then solder them to VCC voltage.

lbernstone, 2019-09-10:
This is what I meant by "bare board".  You really should prototype on a bare board (https://www.aliexpress.com/item/32845762242.html) before you make a pcb to avoid issues like this.  I believe having both 5 and 15 pulled low at boot puts the device into SDIO slave mode.

OmarAlkassab, 2019-09-11:
I have kept GPIO5 and GPIO15 floating, then trying the upload. the ESP32 also can't program!
but now the message when the GPIO0 is low is the following:

> ets Jun  8 2016 00:22:57
> 
> rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
> waiting for download

and still get the same error

OmarAlkassab, 2019-09-11:
I tried these connections and both not worked.

GPIO5 floating (connected to the gate of MOSFET)
GPIO15 floating (connected to the gate of MOSFET)
GPIO2 pulled down using 10k resistor
GPIO12 pulled down using 10k resistor
GPIO4 connected to led in series with 1k resistor
GPIO0 connected to normally open pin of the pushbutton (connected to GND if pushbutton pressed, and rest floating if released)

GPIO5 pulled down using 10k resistor (connected to the gate of MOSFET)
GPIO15 pulled down using 10k resistor (connected to the gate of MOSFET)
GPIO2 pulled down using 10k resistor (connected to the gate of MOSFET)
GPIO12 pulled down using 10k resistor (connected to the gate of MOSFET)
GPIO4 connected to led in series with 1k resistor
GPIO0 connected to normally open pin of the pushbutton (connected to GND if pushbutton pressed, and rest floating if released)

atanisoft, 2019-09-11:
@OmarAlkassab You will likely find your answer [here](https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection).

OmarAlkassab, 2019-09-11:
@atanisoft I read it carefully, but didn’t got the solution!

atanisoft, 2019-09-11:
remove all of the pull downs you have. Pull GPIO2 and GPIO0 down *ONLY* for uploading code. GPIO5, GPIO12 and GPIO15 have restrictions on use and should generally be avoided.

atanisoft, 2019-09-11:
Also, I'd highly recommend following one of the reference designs for your PCB.

OmarAlkassab, 2019-09-12:
@atanisoft thank you for your reply, I will test it then get back to you. And what abou the GPIO4?

atanisoft, 2019-09-12:
Gpio4 is fine. It is not one of the strapping pins.

OmarAlkassab, 2019-09-12:
Great, but I see in the following tutorial (https://randomnerdtutorials.com/esp32-pinout-reference-gpios/) that the GPIO4 is one of the strapping pins, is he wrong?

atanisoft, 2019-09-12:
Technically yes it could be a strapping pin but it is not used as such from any of the released versions of the chips. It is generally used for one of the  can controller pins, the other being gpio 5 typically.

OmarAlkassab, 2019-09-12:
@atanisoft Thank you so much dear.
actually I programmed the chip to support OTA update, then solder the chip into my PCB, because I need these configurations to drive the transistors. So I can test my PCB and debug my code.
I will close the issue now, then after finishing the beta test, I will edit the PCB to respect the strapping pins conditions, then let you know.
Thank you so much!

OmarAlkassab, 2019-10-01:
I respect the values of all strapping pins, and when I reset the module and GPIO0 in low voltage I get this message at 115200 baud:

> �ets Jun  8 2016 00:22:57
> 
> rst:0x1 (PO�ets Jun  8 2016 00:22:57
> 
> rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
> waiting for download

but unfortunately the module still not uploading the code!

atanisoft, 2019-10-01:
@OmarAlkassab GPIO0 = LOW will force the ESP32 to download mode. GPIO0 needs to be HIGH for normal operation.

OmarAlkassab, 2019-10-01:
Yes I know dear, I need to upload a new code to the module, but I can't

atanisoft, 2019-10-01:
It is waiting for you to upload via UART. If you want to upload via OTA you need to ensure GPIO0 is HIGH at boot time.

OmarAlkassab, 2019-10-01:
Yes I need to upload via UART, I don't need to upload via OTA, because If I made some mistakes on the code and the code cause the ESP to restart before OTA initialize, so I can't upload a new code again

OmarAlkassab, 2019-10-01:
I got the following error when using platformio:

> Connecting........_____....._____....._____....._____....._____....._____....._____
> 
> A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
> *** [upload] Error 2

atanisoft, 2019-10-01:
That is due to the incorrect PU/PD setting for GPIO0/2. You need to have both LOW until it starts uploading. Simply connecting RTS/DTR to the pins will only toggle the pin for a very brief moment. There is a specific "auto-reset" circuit that esptool uses to enter/exit the download mode.

OmarAlkassab, 2019-10-01:
I tried it, same erro

OmarAlkassab, 2019-10-01:
Please note also, that TX(GPIO1) is pulled low, and RX(GPIO3) is pulled high, I need to use them in my circuit (GPIO1 to drive a mosfet, and GPIO3 to read pushbutton).

lbernstone, 2019-10-01:
RTS != EN, CTS != GPIO0.  If you set up the transistors as in a dev board, there is some special functionality there.  Otherwise, just keep it simple.  GPIO0 down when you want to program. EN high all the time (I generally have it broken out so that I can touch it to ground and reboot if needed).

OmarAlkassab, 2019-10-01:
@lbernstone I did what you say:

> GPIO0 down when you want to program. EN high all the time (I generally have it broken out so that I can touch it to ground and reboot if needed).

but still not uploading

lbernstone, 2019-10-01:
Use a multimeter to check that the TX from you programmer is actually connecting to RX on the ESP.  Set the transfer rate to something really low (74880), DIO, 40MHz flash.

OmarAlkassab, 2019-10-01:
I checked them (and I can see in the serial terminal that the module is in download mode). but still not working

lbernstone, 2019-10-01:
You will need to try the command line esptool (<.arduino>/packages/esp32/tools/esptool_py/2.6.1/esptool.py).
1) See if you can [query the device](https://github.com/espressif/esptool#read-spi-flash-id-flash_id)
2) If so, [erase the flash](https://github.com/espressif/esptool#erase-flash-erase_flash--erase-region)
3) If you can do that, then try uploading again.

OmarAlkassab, 2019-10-03:
I have solved the problem...
The problem was the connection of the RX pin of the ESP32 to the INT pin of the PCA9555 (it is open drain output). once I cut the trace, the module upload without any problem.
Many thanks for you guys, I learned a lot with you.
@lbernstone 
@atanisoft

closed 2019-10-03 by OmarAlkassab