espressif/arduino-esp32 #3200: client.connected() returns error.

krutarthtrivedi, 2019-09-12:
Board: ESP32 Dev Module
IDE name: Arduino IDE
Flash Frequency: 80Mhz
PSRAM enabled: no
Upload Speed: 115200
Computer OS: Windows 10

Hey all, I am having the following error while trying to establishing connection with google IP Address.  First, I started with Domain name "google.com" instead of its IP address but it was showing "DNS failed for google.com" so that's why I switched to its IP Address.

`[D][WiFiClient.cpp:489] connected(): Disconnected: RES: -1, ERR: 113`

Here's the Arduino code.

```

void setup(){
  delay(1000);
  Serial.begin(115200);
  WiFi.begin("ssid","password");
  while(WiFi.status() != WL_CONNECTED){
  delay(100);
  }
  Server.begin();
}

void loop() {
  Server.handleClient();
  wifi_status = WiFi.status();
  if(wifi_status == WL_CONNECTED){
    testClient("172.217.167.238", 80);
  }
  else {
    Serial.println("No WiFi Connection");
  }
  delay(5000);
}

void testClient(const char * host, uint16_t port){
  Serial.print("\nconnecting to ");
  Serial.println(host);
  WiFiClient client;
  client.connect(host, port);
  server_status = client.connected();
  Serial.println(server_status);
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  while (client.connected() && !client.available());
  while (client.available()) {
    Serial.write(client.read());
  }
  Serial.println("closing connection\n");
  client.stop();
}
```

Here's the Debug message:

> connecting to 172.217.167.238
> [D][WiFiClient.cpp:489] connected(): Disconnected: RES: -1, ERR: 113
> 0                                        //This is the result of client.connected() 
> closing connection

I tested the same HTTP GET Request using telnet on windows 10, connected with the same router in which ESP32 is connected and It was giving the response back. Please find the below response.

> /********************************************************************************//
> HTTP/1.1 301 Moved Permanently
> Location: http://www.google.com/
> Content-Type: text/html; charset=UTF-8
> Date: Thu, 12 Sep 2019 07:35:46 GMT
> Expires: Sat, 12 Oct 2019 07:35:46 GMT
> Cache-Control: public, max-age=2592000
> Server: gws
> Content-Length: 219
> X-XSS-Protection: 0
> X-Frame-Options: SAMEORIGIN
> 
> <HTML><HEAD><meta http-equiv="content-type" content="text/html;charset=utf-8">
> <TITLE>301 Moved</TITLE></HEAD><BODY>
> <H1>301 Moved</H1>
> The document has moved
> <A HREF="http://www.google.com/">here</A>.
> </BODY></HTML>
> /********************************************************************************//

 So, at this point of time I guess the problem is somewhat associated with ESP32 TCP/IP connection establishment.

Kindly help me out to resolve this asap.

Thanks,
Krutarth Trivedi.

kapyaar, 2019-09-13:
@krutarthtrivedi tested on an ESP32 Dev module with 1.0.3rc3. Seems to be working. 

```
connecting to 172.217.167.238
1
HTTP/1.1 301 Moved Permanently
Location: http://www.google.com/
Content-Type: text/html; charset=UTF-8
Date: Fri, 13 Sep 2019 19:09:46 GMT
Expires: Sun, 13 Oct 2019 19:09:46 GMT
Cache-Control: public, max-age=2592000
Server: gws
Content-Length: 219
X-XSS-Protection: 0
X-Frame-Options: SAMEORIGIN

<HTML><HEAD><meta http-equiv="content-type" content="text/html;charset=utf-8">
<TITLE>301 Moved</TITLE></HEAD><BODY>
<H1>301 Moved</H1>
The document has moved
<A HREF="http://www.google.com/">here</A>.
</BODY></HTML>
closing connection
```

But I do think that there is some DNS related flunkiness going on. I have seen similar behaviour as 
https://github.com/espressif/arduino-esp32/issues/3203 albeit sporadically. Cannot repeat it at will. 
Can you check and print the dns server your router using?

stickbreaker, 2019-09-13:
@kapyaar This 301 moved error is caused by using a numeric server address.  The redirect to "www.google.com" is required because of server pooling/load balancing.  The numeric address resolves down to ONE specific piece of hardware.  "www.google.com" is not ONE server.  That specific address actually resolves to a load balancer that transparently forwards the request to next available server.  The "name" is needed because a single piece of hardware can support multiple virtual domains. "www.yourdomain.com:80" and "www.google.com:80" could resolve down to the same numeric address/hardware.  The HTTP server needs the "name" to decide how to handle the request. "http://172.217.167.238" is not the same as "http://www.google.com".

Chuck.

kapyaar, 2019-09-13:
@stickbreaker Yes, I totally understand.
1. When I said it is working for me, I meant the ESP32 is doing its job, and bringing the server response correctly. I understand why the 301. But issue above is that client.connected() was returning 0. In my test, it is showing 1 as it should.

2. When I said flunkiness, I meant in some other tests I have seen the ESP32 fails to make a connection to an external server ( I was using a domain name which has its load balancers and stuff on aws, and not an IP address). 

My point was to state that 1.0.3rc3 seems to be working as far as #3200 is concerned, but I also wanted to point out some observations from my testings which may mean something to #3203

stickbreaker, 2019-09-13:
@kapyaar ok, I miss understood your comment. I just wanted to explain the 301.  #3203 is too vacuous.  I don't see any way to help because the poster did not follow the issue template, no code, just "a plea for help".

chuck.

krutarthtrivedi, 2019-09-16:
@kapyaar 
> tested` on an ESP32 Dev module with 1.0.3rc3. Seems to be working.
>
> ```
> connecting to 172.217.167.238
> 1
> HTTP/1.1 301 Moved Permanently
> Location: http://www.google.com/
> Content-Type: text/html; [...]

I am using static IP assignment and followings are the data. Besides, what I have analyzed is, client.connected() returns 0 and ERR:113 and sometimes 1 without any kind of error. So, I am bit confused regarding this kind of behavior. I have tried using different ESP32 and Routers as well.

> IPAddress local_IP(192, 168, 31, 100);
> IPAddress gateway(192, 168, 31, 1);
> IPAddress subnet(255, 255, 255, 0);
> IPAddress DNS1(8, 8, 8, 8);
> IPAddress DNS2(8, 8, 4, 4);

kapyaar, 2019-09-16:
Same results, working correctly for me with static IP as well. Here is the code I used since you did nt provide the whole code. Also, I use IDE 1.8.9

```
/*
    This sketch sends a message to a TCP server

*/

#include <WiFi.h>
#include <WiFiMulti.h>

WiFiMulti WiFiMulti;

void setup() {
  delay(1000);
  Serial.begin(115200);
  IPAddress local_IP(192, 168, 0, 100);
  IPAddress gateway(192, 168, 0, 1);
  IPAddress subnet(255, 255, 255, 0);
  IPAddress DNS1(8, 8, 8, 8);
  IPAddress DNS2(8, 8, 4, 4);
  WiFi.config(local_IP, gateway, subnet, DNS1);
  WiFi.begin("SSID", "PASSWORD");
  while (WiFi.status() != WL_CONNECTED) {
    delay(100);
  }  
}

void loop() {
  
  
  if (WiFi.status() == WL_CONNECTED) {
    testClient("172.217.167.238", 80);
  }
  else {
    Serial.println("No WiFi Connection");
  }
  delay(5000);
}

void testClient(const char * host, uint16_t port) {
  Serial.print("\nconnecting to ");
  Serial.println(host);
  WiFiClient client;
  client.connect(host, port);
  //server_status = client.connected();
  Serial.println(client.connected());
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  while (client.connected() && !client.available());
  while (client.available()) {
    Serial.write(client.read());
  }
  Serial.println("closing connection\n");
  client.stop();
}
```
Some thoughts:
1. Try with the above code and see if it makes any difference. I used 
[image]
2. Delete your entire arduino installation (after keeping your own files and libraries) and reinstall it and start over.

Between to two, you should get some insights as to what might be going on.

krutarthtrivedi, 2019-09-26:
@kapyaar Thanks!
I tried reinstalling everything and It solved my problem. 

One more thing I found is, If internet connectivity goes down in between then I am getting "false"  from client.connected() but this is taking too much time and it's slowing down other functions performing inside void  loop(). What can be the possible reason for this?

My code is as follows:
1. Taking sensor data at every 10 seconds. (This 10 seconds is a non-blocking delay produced using millis()).
2. Connect to the server using  client.connect(). if true, then proceed else exit.
3. Check for connectivity using client.connected(). if true, post data to server else store them to SD card.
4. If client.connected() returns true then data posting is taking place at every 10 seconds flawlessly without any error.
5. If client.connected() returns false then it is storing data to SD card but taking 20 seconds to respond instead of 10 seconds. I assume, It's client.connected() which is taking this much of time to figure out what kind of error is present there. I am getting "Software caused connection abort" during false return. 

This image is taken suddenly after data connectivity is lost. It can be seen clearly that it's taking around 30 seconds to show that internet/server connection is lost and data is getting stored to SD card.

[image]

During run-time, it's taking around 20 seconds to show that data is getting transferred to SD card. As far as, program is concerned. I assure you that there's no blocking delay inside the void loop() and this is coming into the picture when client.connected() returns false.

[image]

As per my application requirement. It's necessary to check for server connectivity along with data reading at every 10 seconds. Kindly look into it and share your thoughts.

Thanks,
Krutarth Trivedi.

kapyaar, 2019-09-26:
I would just put a timeout on the client.connect() check

krutarthtrivedi, 2019-09-27:
> I would just put a timeout on the client.connect() check
@kapyaar 
Sorry, but I don't understand your point.

Thanks,
Krutarth Trivedi.

krutarthtrivedi, 2019-09-30:
@kapyaar Thanks! 
I got the answer, I was looking for!
I'll update the main code and test. I will get back to you if I need any further help.

Thanks,
Krutarth Trivedi.

closed 2019-10-14 by krutarthtrivedi

Israelmx, 2021-02-11:
Hi @krutarthtrivedi, how do you solve the slow the timing when the client disconnect?