espressif/arduino-esp32 #3203: [E][WiFiGeneric.cpp:654] hostByName(): DNS Failed for server_domain.com

OmarAlkassab, 2019-09-13:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.3
IDE name:						Arduino IDE and Platform.io
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:					921600
Computer OS:                                         Windows 10

### Description:
I used the BasicHttpClient code to make a POST request to a server. the code run fine, but
when I use the same code library in my code it generate the following error:
### Debug Messages:
```
[E][WiFiGeneric.cpp:654] hostByName(): DNS Failed for server_domain.com
```
note1: I used PubSubClient.h and many libraries.
note2: some times the code run fine, but rarely
note3: server_domain.com is a dummy, not the real server which i use

anyone can help me please?

OmarAlkassab, 2019-11-13:
@kapyaar, I currently working with PlatformIO on VSCode and the error still occurred when the Internet connectivity lost

OmarAlkassab, 2019-12-26:
Closing as the issue solved after update to the newest firmware

closed 2019-12-26 by OmarAlkassab

davidkjackson54, 2020-09-02:
This problem is still occurring.
I have just moved code from  an ESP8266 to ESP32.  
I have MQTT running on a Raspberry p[i whcih is also running avahi. 
I can ping the RPi  from my Windows 10 machine using: 
raspberrypi
raspberrypi.local 
raspberroypi.localdomain

My code:
if (!MDNS.begin("esp32")) {             // Start the mDNS responder for esp32.local
    Serial.println("Error setting up MDNS responder!");
    while(1) {
            delay(1000);
        }
  } 
  Serial.println("mDNS responder started");
 
  client.setServer("**raspberrypi.localdomain**", 1883);        // set ip address of Raspberry pi and default mqtt port number
  client.setCallback(callback);               // set callback for messages that arrive on a topic

This works but if I change it to raspberrypi.local , I get
[**E][WiFiGeneric.cpp:654] hostByName(): DNS Failed**

This worked fine on ESP8266 but fails on ESP32.

Avahi on the pi is using defaults and expects  .local. SO I would have expected a MQTT connect ot raspberrypi.local to have worked from the esp32.    AsI understand it, .local is the required suffix for mDNS. and that is assumed. IF I haev to hardcode .localdomain , then it is no better than hard coding an IP address.

I am using Unifi router and wifi .