espressif/arduino-esp32 #3204: compile error - in function 'esp_err_t capture_handler(httpd_req_t*)':

die-tueftelei, 2019-09-13:
### Hardware:
Board:							espressif32 - esp32cam
IDE name:						Platform.io
Upload Speed:					115200
Computer OS:                                        Mac OSX 10.14.2

### Description:

Compiling .pio/build/esp32cam/lib316/WiFi/WiFiGeneric.cpp.o
src/app_httpd.cpp: In function 'esp_err_t capture_handler(httpd_req_t*)':
src/app_httpd.cpp:286:25: error: 'box_array_t {aka struct tag_box_list}' has no member named 'score'
         free(net_boxes->score);
                         ^

me-no-dev, 2019-09-14:
The build is passing, so something in your PIO install did not go well :)
https://travis-ci.org/espressif/arduino-esp32/jobs/584357979#L1015-L1099

@ivankravets maybe files did not update properly?

valeros, 2019-09-16:
Hi @die-tueftelei ! Does it work with Arduino IDE? If so, please upload your project somewhere so we can reproduce the issue. Thanks!

die-tueftelei, 2019-09-16:
Hi @valeros I just copied the following example: https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/Camera/CameraWebServer

die-tueftelei, 2019-09-16:
& I did not try Arduino IDE - I use Atom & Platform.IO

atanisoft, 2019-09-16:
@die-tueftelei using "master" branch with the current release available in PIO is the problem. PIO has not updated to 1.0.3 release, it will soon though.

you can fix it in your project by adding the following to your platformio.ini file:
```
[env]
platform_packages =
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
```

This will force override the version of the arduino-esp32 code that PIO uses to the latest version available in GIT. This is a temporary solution until PIO has packaged the latest arduino-esp32 1.0.3 release for PIO.

die-tueftelei, 2019-09-16:
perfect - yep that solves it !

closed 2019-09-16 by die-tueftelei