espressif/arduino-esp32 #3205: Serial.end() not reattaching pin to GPIO matrix

luketanti, 2019-09-13:
After trying to switch back two UART pins to GPIO by using the Serial.end() function, the pins do not respond to digitalWrite commands.

I have found out that by using "gpio_matrix_out(gpio_number, SIG_GPIO_OUT_IDX, false, false);" I am able to use the pin again after ending the Serial.

stickbreaker, 2019-09-13:
@luketanti after using `Serial.end()` did you do a `pinMode(pin,OUTPUT)` before doing a `digitalWrite(pin,value)`?

Chuck.

luketanti, 2019-09-14:
> @luketanti after using `Serial.end()` did you do a `pinMode(pin,OUTPUT)` before doing a `digitalWrite(pin,value)`?
> 
> Chuck.

Yes, offcourse.

Here are two snippets. The first one is when the issue occurs and the second is how I managed to get the pin to respond to digitalWrite commands.

```
  Serial2.end();
  pinMode(K_TX, OUTPUT);
  pinMode(K_RX, INPUT);
  digitalWrite(K_TX, HIGH);
  delay(298);
  digitalWrite(K_TX, LOW);
  delay(23);
  digitalWrite(K_TX, HIGH);
  delay(23);

  Serial2.begin(10417, SERIAL_8N1, K_RX, K_TX);  // K-LINE
```

```
  Serial2.end();
  gpio_matrix_out(K_TX, SIG_GPIO_OUT_IDX, false, false);
  pinMode(K_TX, OUTPUT);
  pinMode(K_RX, INPUT);
  digitalWrite(K_TX, HIGH);
  delay(298);
  digitalWrite(K_TX, LOW);
  delay(23);
  digitalWrite(K_TX, HIGH);
  delay(23);

  Serial2.begin(10417, SERIAL_8N1, K_RX, K_TX);  // K-LINE
```

atanisoft, 2019-09-14:
Did you try gpio_reset(K_TX); instead of gpio_matrix_out?

luketanti, 2019-09-14:
I have tried to compile with gpio_reset(K_TX); but it failed.

> exit status 1
> 'gpio_reset' was not declared in this scope

Shouldn't Serial.end() automatically release the pins allocated to other use?

atanisoft, 2019-09-14:
Sorry that should have been gpio_reset_pin. And yes I'd think that Serial.end() should release the pin to normal/default state. I believe there was an older issue for the same.

luketanti, 2019-09-14:
Yes, I can confirm that it also works.

`gpio_reset_pin(GPIO_NUM_17);` in my case.

Can this line be added to the UART HAL?
And in future Arduino core updates this would be included?

atanisoft, 2019-09-14:
@luketanti send in a PR that does this for both RX and TX in the esp32-hal-uart.c file in the uartEnd() method and I'm sure @me-no-dev will consider it.

me-no-dev, 2019-09-14:
huh? this is rather really strange. this is already called in `uartEnd()` when the pins are detached from the matrix (`gpio_matrix_out(gpio_number, SIG_GPIO_OUT_IDX, false, false);`)

me-no-dev, 2019-09-14:
- `SIG_GPIO_OUT_IDX ` equals 0x100 (256). see here: https://github.com/espressif/esp-idf/blob/aaf12390eb14b95589acd98db5c268a2e56bb67e/components/soc/esp32/include/soc/gpio_sig_map.h#L421
- Arduino's function to detach from the matrix is here: https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-matrix.c#L28
- `MATRIX_DETACH_OUT_SIG` in Arduino also equals 0x100. see here: https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-matrix.c#L19
- The call to detach is coming from here: https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-uart.c#L127-L142
- And those are called in `uartEnd()` here: https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-uart.c#L242-L243
- `uartEnd()` is called by Arduino's `HardwareSeral::end()` here: https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HardwareSerial.cpp#L87

stickbreaker, 2019-09-14:
@me-no-dev my 2 cents: The only difference I see between I2C(working) and UART de-init  is a call to `pinMode(pin,INPUT | PULLUP);` 

Has anyone other than the OP validated this behavior? 

That is why I asked my silly question above.

Chuck.

stickbreaker, 2019-09-14:
@luketanti do you call `Serial.flush();` before you call `Serial.end();`?
Could it be a case where the UART is still using (interrupting)?

Chuck.

bertmelis, 2019-09-15:
flush blocks untill uart's FSM is idle.

stickbreaker, 2019-09-15:
@bertmelis yep, that is its purpose. I believe reconfiguring UART should not happen until it is idle.

closed 2019-11-28 by stale[bot]

WayneKeenan, 2020-02-01:
This is still an issue on  Pin16 (U2RX) on  ESP-CAM using  Arduino 1.0.4.