espressif/arduino-esp32 #3207: BT client disconnection while SerialBT.write() cause ESP32 to hang

Benik3, 2019-09-14:
### Hardware:
Board: Wemos ESP32 mini
Core Installation version: 1.0.3
IDE name: Arduino IDE 1.8.9
Flash Frequency: 40MHz
PSRAM enabled: No
Upload Speed: 921600
Computer OS: Win10 1903

### Description:
If BlueTooth client disconnects while SerialBT.write(); function sends data, ESP32 hangs here for ever.
Even that ESP32 is hanged, the BT is still registering that client connected/disconnected:
18:19:58.328 -> [I][BluetoothSerial.cpp:174] esp_spp_cb(): ESP_SPP_SRV_OPEN_EVT
18:20:02.270 -> [I][BluetoothSerial.cpp:184] esp_spp_cb(): ESP_SPP_CLOSE_EVT

To get the BTserial works again I have to restart the ESP32.
I use this for sending GPS NMEA data to mobile phone over BT.

### Sketch: 
```cpp
#include "BluetoothSerial.h"
BluetoothSerial SerialBT;

void setup() {
  Serial1.begin(115200, SERIAL_8N1, 16, 17);
  Serial1.setRxBufferSize(4096);
  SerialBT.begin("BT_SPP");
}

void loop() { 
  byte buff[4096];
  int num = Serial1.available();
  
  if(num > 0 && SerialBT.hasClient()){
    Serial1.readBytes(buff, num);
    SerialBT.write(buff, num);
    SerialBT.flush();
  }
}

```

### Debug Messages:
```
18:13:04.184 -> load:0x3fff0018,len:4
18:13:04.184 -> load:0x3fff001c,len:928
18:13:04.184 -> ho 0 tail 12 room 4
18:13:04.184 -> load:0x40078000,len:8740
18:13:04.184 -> load:0x40080400,len:5800
18:13:04.184 -> entry 0x4008069c
18:13:04.527 -> ⸮⸮⸮U⸮⸮⸮2-hal-cpu.c:178] setCpuFrequencyMhz(): PLL: 320 / 4 = 80 Mhz, APB: 80000000 Hz
18:13:05.814 -> [I][BluetoothSerial.cpp:160] esp_spp_cb(): ESP_SPP_INIT_EVT
18:13:05.814 -> [I][BluetoothSerial.cpp:227] esp_spp_cb(): ESP_SPP_START_EVT
18:18:31.494 -> [I][BluetoothSerial.cpp:174] esp_spp_cb(): ESP_SPP_SRV_OPEN_EVT
18:18:31.595 -> [V][BluetoothSerial.cpp:201] esp_spp_cb(): ESP_SPP_WRITE_EVT: 330 FREE
...

```

closed 2019-11-27 by stale[bot]

emmabeebe, 2020-01-27:
I believe I am also having this issue so if anyone figures it out that would be great.

Benik3, 2020-01-27:
Thanks, I can confirm this bug also from my friend, which also tried to play with BT on ESP32...

emmabeebe, 2020-01-27:
https://github.com/espressif/arduino-esp32/issues/3657
There’s another issue out that I think applies to this one as well.

emmabeebe, 2020-01-28:
Also if you add a watchdog to your program you can find ways around this bug until they fix it.

Benik3, 2020-01-28:
Yeah, that's exactly what I did :)

max5555, 2020-05-31:
> 
> 
> Yeah, that's exactly what I did :)

Can you give an example?

Benik3, 2020-05-31:
I used this example: https://medium.com/@supotsaeea/esp32-reboot-system-when-watchdog-timeout-4f3536bf17ef