espressif/arduino-esp32 #3208: Cannot read ADC2 while using BLE

jdhurwitz, 2019-09-14:
# Hardware:
Board:							ESP-32 wroom on custom PCB
Flash Frequency:					40MHz
Upload Speed:					115200
Computer OS:                                        Mac OS X

### Description:

I'm trying to read sensor values on ADC2 GPIOs while BLE is enabled, but all of the pins on ADC2 read high (0xffff...fff). When BLE isn't used to read ADC2 GPIOs, all the pins can be read through a serial connection. 

The only GPIOs I can read using BLE are those on ADC1. Is there any way to make it so that the GPIOs on ADC2 can be read using BLE? Has anyone else run into this issue? I know there are known issues with reading GPIOs on ADC2 while WiFi is enabled, but not necessarily when BLE is enabled. I know they share the same frequency band and controller, but I don't understand why the pins would be inaccessible.

I have tried to turn off WiFi during setup, but including the WiFi.h library makes the binary too large since BLE's library is already massive as is. Below is some super basic GPIO pin polling/reading. If I'm doing something critically wrong, please let me know! Thank you!

### Sketch:  (leave the backquotes for [code formatting]
```
// main.cpp
#ifdef ENABLE_BLUETOOTH
  #include "ble.h"
#endif

#include "sensor.h"
#include "helpers.h"

void setupBluetoothPins() 
{
  for (int i = 0; i < pcbSensorCount; i++) 
  {
    pinMode(pcbPins[i], OUTPUT);
  }
}

void bluetoothSensorLoop()
{
  for (int i = 0; i < pcbSensorCount; i++)
  {
    uint8_t val = analogRead(pcbPins[i]);
    pcbSensors[i] = (uint8_t)val;
  }
  setSensorCharacteristic(pcbSensors, pcbSensorCount*sizeof(uint8_t));
}

void loop()
{
 while(true) {
   bluetoothSensorLoop();
   delay(300);
  }
}

void setup() 
{
  // put your setup code here, to run once:
  setupPins();

  Serial.begin(115200);
  #ifdef ENABLE_BLUETOOTH
    setupBLE();
  #endif
}
```

atanisoft, 2019-09-14:
ADC2 is used by the radio for WiFi and likely BLE. So when either are active it is highly unlikely that ADC2 is usable.

jdhurwitz, 2019-09-14:
Will calling btStop() before reading the sensor values and then btStart() to reinitialize bluetooth work? https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-bt.c

Wondering if the ADC2 will still be allocated to the radio even after calling btStop()

atanisoft, 2019-09-14:
That might work, not surr

closed 2019-11-27 by stale[bot]