espressif/arduino-esp32 #3211: HTTPClient returns error send header failed when using HTTPS

mlukasek, 2019-09-15:
### Hardware:
Board:							M5Stack Core
Core Installation version:			1.0.3
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:
HTTPClient fails with error "send header failed" when using HTTPS. It was working ok with ES32 1.0.2, but after update to ESP32 1.0.3 it fails. Even the simple standard sample HTTPClient/BasicHttpsClient fails. When I go back to 1.0.2, it works. Included code is just a pure BasicHttpsClient sample with changed SSID a wifi pass.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
/**
   BasicHTTPSClient.ino

    Created on: 14.10.2018

*/

#include <Arduino.h>

#include <WiFi.h>
#include <WiFiMulti.h>

#include <HTTPClient.h>

#include <WiFiClientSecure.h>

// This is GandiStandardSSLCA2.pem, the root Certificate Authority that signed 
// the server certifcate for the demo server https://jigsaw.w3.org in this
// example. This certificate is valid until Sep 11 23:59:59 2024 GMT
const char* rootCACertificate = \
"-----BEGIN CERTIFICATE-----\n" \
"MIIF6TCCA9GgAwIBAgIQBeTcO5Q4qzuFl8umoZhQ4zANBgkqhkiG9w0BAQwFADCB\n" \
"iDELMAkGA1UEBhMCVVMxEzARBgNVBAgTCk5ldyBKZXJzZXkxFDASBgNVBAcTC0pl\n" \
"cnNleSBDaXR5MR4wHAYDVQQKExVUaGUgVVNFUlRSVVNUIE5ldHdvcmsxLjAsBgNV\n" \
"BAMTJVVTRVJUcnVzdCBSU0EgQ2VydGlmaWNhdGlvbiBBdXRob3JpdHkwHhcNMTQw\n" \
"OTEyMDAwMDAwWhcNMjQwOTExMjM1OTU5WjBfMQswCQYDVQQGEwJGUjEOMAwGA1UE\n" \
"CBMFUGFyaXMxDjAMBgNVBAcTBVBhcmlzMQ4wDAYDVQQKEwVHYW5kaTEgMB4GA1UE\n" \
"AxMXR2FuZGkgU3RhbmRhcmQgU1NMIENBIDIwggEiMA0GCSqGSIb3DQEBAQUAA4IB\n" \
"DwAwggEKAoIBAQCUBC2meZV0/9UAPPWu2JSxKXzAjwsLibmCg5duNyj1ohrP0pIL\n" \
"m6jTh5RzhBCf3DXLwi2SrCG5yzv8QMHBgyHwv/j2nPqcghDA0I5O5Q1MsJFckLSk\n" \
"QFEW2uSEEi0FXKEfFxkkUap66uEHG4aNAXLy59SDIzme4OFMH2sio7QQZrDtgpbX\n" \
"bmq08j+1QvzdirWrui0dOnWbMdw+naxb00ENbLAb9Tr1eeohovj0M1JLJC0epJmx\n" \
"bUi8uBL+cnB89/sCdfSN3tbawKAyGlLfOGsuRTg/PwSWAP2h9KK71RfWJ3wbWFmV\n" \
"XooS/ZyrgT5SKEhRhWvzkbKGPym1bgNi7tYFAgMBAAGjggF1MIIBcTAfBgNVHSME\n" \
"GDAWgBRTeb9aqitKz1SA4dibwJ3ysgNmyzAdBgNVHQ4EFgQUs5Cn2MmvTs1hPJ98\n" \
"rV1/Qf1pMOowDgYDVR0PAQH/BAQDAgGGMBIGA1UdEwEB/wQIMAYBAf8CAQAwHQYD\n" \
"VR0lBBYwFAYIKwYBBQUHAwEGCCsGAQUFBwMCMCIGA1UdIAQbMBkwDQYLKwYBBAGy\n" \
"MQECAhowCAYGZ4EMAQIBMFAGA1UdHwRJMEcwRaBDoEGGP2h0dHA6Ly9jcmwudXNl\n" \
"cnRydXN0LmNvbS9VU0VSVHJ1c3RSU0FDZXJ0aWZpY2F0aW9uQXV0aG9yaXR5LmNy\n" \
"bDB2BggrBgEFBQcBAQRqMGgwPwYIKwYBBQUHMAKGM2h0dHA6Ly9jcnQudXNlcnRy\n" \
"dXN0LmNvbS9VU0VSVHJ1c3RSU0FBZGRUcnVzdENBLmNydDAlBggrBgEFBQcwAYYZ\n" \
"aHR0cDovL29jc3AudXNlcnRydXN0LmNvbTANBgkqhkiG9w0BAQwFAAOCAgEAWGf9\n" \
"crJq13xhlhl+2UNG0SZ9yFP6ZrBrLafTqlb3OojQO3LJUP33WbKqaPWMcwO7lWUX\n" \
"zi8c3ZgTopHJ7qFAbjyY1lzzsiI8Le4bpOHeICQW8owRc5E69vrOJAKHypPstLbI\n" \
"FhfFcvwnQPYT/pOmnVHvPCvYd1ebjGU6NSU2t7WKY28HJ5OxYI2A25bUeo8tqxyI\n" \
"yW5+1mUfr13KFj8oRtygNeX56eXVlogMT8a3d2dIhCe2H7Bo26y/d7CQuKLJHDJd\n" \
"ArolQ4FCR7vY4Y8MDEZf7kYzawMUgtN+zY+vkNaOJH1AQrRqahfGlZfh8jjNp+20\n" \
"J0CT33KpuMZmYzc4ZCIwojvxuch7yPspOqsactIGEk72gtQjbz7Dk+XYtsDe3CMW\n" \
"1hMwt6CaDixVBgBwAc/qOR2A24j3pSC4W/0xJmmPLQphgzpHphNULB7j7UTKvGof\n" \
"KA5R2d4On3XNDgOVyvnFqSot/kGkoUeuDcL5OWYzSlvhhChZbH2UF3bkRYKtcCD9\n" \
"0m9jqNf6oDP6N8v3smWe2lBvP+Sn845dWDKXcCMu5/3EFZucJ48y7RetWIExKREa\n" \
"m9T8bJUox04FB6b9HbwZ4ui3uRGKLXASUoWNjDNKD/yZkuBjcNqllEdjB+dYxzFf\n" \
"BT02Vf6Dsuimrdfp5gJ0iHRc2jTbkNJtUQoj1iM=\n" \
"-----END CERTIFICATE-----\n";
[... 120 lines cut ...]
[E][ssl_client.cpp:35] handle_error(): MbedTLS message code: -28928
[E][ssl_client.cpp:33] handle_error(): SSL - Bad input parameters to function
[E][ssl_client.cpp:35] handle_error(): MbedTLS message code: -28928
[E][ssl_client.cpp:33] handle_error(): SSL - Bad input parameters to function
[E][ssl_client.cpp:35] handle_error(): MbedTLS message code: -28928
[E][ssl_client.cpp:33] handle_error(): SSL - Bad input parameters to function
[E][ssl_client.cpp:35] handle_error(): MbedTLS message code: -28928
[E][ssl_client.cpp:33] handle_error(): SSL - Bad input parameters to function
[E][ssl_client.cpp:35] handle_error(): MbedTLS message code: -28928
[W][HTTPClient.cpp:1318] returnError(): error(-2): send header failed
[E][ssl_client.cpp:33] handle_error(): SSL - Bad input parameters to function
[E][ssl_client.cpp:35] handle_error(): MbedTLS message code: -28928
[E][ssl_client.cpp:33] handle_error(): SSL - Bad input parameters to function
[E][ssl_client.cpp:35] handle_error(): MbedTLS message code: -28928
[D][HTTPClient.cpp:1320] returnError(): tcp stop
[E][ssl_client.cpp:33] handle_error(): SSL - Bad input parameters to function
[E][ssl_client.cpp:35] handle_error(): MbedTLS message code: -28928
[E][ssl_client.cpp:33] handle_error(): SSL - Bad input parameters to function
[E][ssl_client.cpp:35] handle_error(): MbedTLS message code: -28928
[E][ssl_client.cpp:33] handle_error(): SSL - Bad input parameters to function
[E][ssl_client.cpp:35] handle_error(): MbedTLS message code: -28928
[D][HTTPClient.cpp:370] disconnect(): tcp stop

Waiting 10s before the next round...
```

vlastahajek, 2019-09-15:
The same for me. No TLS communication works. After downgrade to 1.0.2 it works again.

meltdown03, 2019-09-15:
Same here, check out #2670 for a workaround

Jeroen88, 2019-09-15:
Maybe calling HTTPClient::setConnectTimeout() with some decent value helps? Looking at the code the timeout is in ms, so something in the 5000 to 15000 range might work?

meltdown03, 2019-09-15:
> Maybe calling HTTPClient::setConnectTimeout() with some decent value helps? Looking at the code the timeout is in ms, so something in the 5000 to 15000 range might work?

No luck with setting timeouts for me.

me-no-dev, 2019-09-16:
should be fixed in current master

me-no-dev, 2019-09-16:
if you can confirm, I can release it :)

meltdown03, 2019-09-16:
Yes, I can confirm it works for me. Thank you!

mlukasek, 2019-09-17:
It works great, thank you!

vlastahajek, 2019-09-18:
I can also confirm the fix works. Thanks.

vlastahajek, 2019-09-25:
New finding. It hangs when issuing HTTPS post:
```
Checking db..
[D][HTTPClient.cpp:276] beginInternal(): host: <edited>
[D][HTTPClient.cpp:1025] connect():  connected to <edited>
[D][HTTPClient.cpp:1158] handleHeaderResponse(): code: 200
[D][HTTPClient.cpp:1165] handleHeaderResponse(): Transfer-Encoding: chunked
[D][HTTPClient.cpp:797] writeToStream():  read chunk len: 243
[D][HTTPClient.cpp:1295] writeToStreamDataBlock(): connection closed or file end (written: 243).
[D][HTTPClient.cpp:797] writeToStream():  read chunk len: 0
[D][HTTPClient.cpp:361] disconnect(): still data in buffer (2), clean up.

[D][HTTPClient.cpp:368] disconnect(): tcp keep open for reuse

[D][HTTPClient.cpp:383] disconnect(): tcp is closed

Writing to db
[D][HTTPClient.cpp:276] beginInternal(): <edited>
[I][ssl_client.cpp:156] start_ssl_client(): WARNING: Use certificates for a more secure communication!
[D][HTTPClient.cpp:1025] connect():  connected to <edited>
[D][HTTPClient.cpp:1158] handleHeaderResponse(): code: 204
```

cyberman54, 2019-09-30:
same here, see
https://github.com/espressif/arduino-esp32/issues/3273

mlukasek, 2019-10-02:
@me-no-dev It is working OK in 1.0.4 RC1, so you can close this. Thank you.

me-no-dev, 2019-10-02:
nice :)

closed 2019-10-02 by me-no-dev