espressif/arduino-esp32 #3213: output error stack smashing.

Mangulomx, 2019-09-16:
### Hardware:
Board:							TTGO-LORA32-V2.0
Core Installation version:			
IDE name:						Arduino IDE
Flash Frequency:					80 MHZ
PSRAM enabled:                                      no
Upload Speed:					15200
Computer OS:                                        Mac OSX

### Description:
My board rebooting code output error stack smashing. I am trying mesasure temperature with sensor DS18b20. 

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
/*******************************************************************************
 * Copyright (c) 2015 Thomas Telkamp and Matthijs Kooijman
 *
 * Permission is hereby granted, free of charge, to anyone
 * obtaining a copy of this document and accompanying files,
 * to do whatever they want with them without any restriction,
 * including, but not limited to, copying, modification and redistribution.
 * NO WARRANTY OF ANY KIND IS PROVIDED.
 *
 *******************************************************************************/

#include <Arduino.h>

#include <lmic.h>
#include <hal/hal.h>
#include <SPI.h>
#include <U8x8lib.h>
#include <OneWire.h>
#include <DallasTemperature.h>

// the OLED used
// ux8x(clock, data, reset)
U8X8_SSD1306_128X64_NONAME_SW_I2C u8x8 (/* clock=*/ 22, /* data=*/ 21, /* reset=*/ 16);

// LoRaWAN NwkSKey, network session key
static const PROGMEM u1_t NWKSKEY[16] = { 0xF7, 0xCE, 0xFD, 0x86, 0x3A, 0x6B, 0x0C, 0x3D, 0x05, 0x76, 0xE8, 0x1A, 0xDC, 0xD5, 0x05, 0x07 };

// LoRaWAN AppSKey, application session key
static const u1_t PROGMEM APPSKEY[16] = { 0x8B, 0xDB, 0x05, 0xBE, 0x4A, 0x86, 0xED, 0x6C, 0xF8, 0x3C, 0xF9, 0xA3, 0xDA, 0x3F, 0x05, 0xD8 };

// LoRaWAN end-device address (DevAddr)
static const u4_t DEVADDR = 0x26011245;

// These callbacks are only used in over-the-air activation, so they are
// left empty here (we cannot leave them out completely unless
// DISABLE_JOIN is set in config.h, otherwise the linker will complain).
void os_getArtEui(u1_t *buf) {}
void os_getDevEui(u1_t *buf) {}
void os_getDevKey(u1_t *buf) {}

#define ONE_WIRE_BUS 13 // GPIO 13
#define LED_BUILTIN 25

OneWire oneWire(ONE_WIRE_BUS);
DallasTemperature sensors(&oneWire);
DeviceAddress tempSensor;
static float tempC;

static uint8_t mydata[16];

static osjob_t sendjob;
static osjob_t triggerjob;
static osjob_t readjob;

void do_temp_read(osjob_t *j);

[... 302 lines cut ...]

abort() was called at PC 0x400da3e8 on core 1

Backtrace: 0x4008b8f4:0x3ffb1ec0 0x4008bb21:0x3ffb1ee0 0x400da3e8:0x3ffb1f00 0x400d0f69:0x3ffb1f20 0x400d1213:0x3ffb1f50 0x400d6897:0x3ffb1fb0 0x40087a71:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9344
load:0x40080400,len:6348
entry 0x400806a8
Node v0.2
Trigger Temps
Read Temps = 25.00
Read Temps = 25.00
Send Temp: 25.00
Packet queued
235922: EV_TXCOMPLETE (includes waiting for RX windows)
```

Mangulomx, 2019-09-17:
Someone say me something. What can I modify? Regards

stickbreaker, 2019-09-17:
@mangulomx
You are over writing your temp display buffer.

"=%0.2f" creates "=-127.00"+NULL.  Eight characters plus null is NINE.

Use ~svnprintf()~

Mangulomx, 2019-09-17:
Are there some plugin ti know errors exactly?

Mangulomx, 2019-09-18:
@stickbreaker could you put me an example?. I am very novice

stickbreaker, 2019-09-18:
[snprintf()](http://www.cplusplus.com/reference/cstdio/snprintf/)
```c++
void readTemps()
{
    tempC = sensors.getTempC(tempSensor);
    if((tempC > 85.0)||(tempC < -40.0)) { // invalid temperature reading
      Serial.printf("invalid temp reading %0.2fC\n",tempC);
      tempC = -127.0;
      return;
    }
    Serial.printf("Read Temps = %0.2fC\n", tempC);
#define BUFFMAX 8 // this will fail if tempC is out of range, but it will not crash program, 
   // just print error messages.  if overflow error message prints, just increase this value.

    char buf[BUFFMAX];
    int len = snprintf(buf, BUFFMAX,"=%0.2f", tempC); //len will be length of resultant string.
    // if len <0 then error, if >= BUFFMAX then would have overflowed storage location (buf)

    if( len<0) { // invalid conversion
      snprintf(buf,BUFFMAX,"ERR CONV");
     Serial.printf("Conversion Error\n");
    }
    else if ( len>=BUFFMAX) { // conversion exceeded buffer length
      snprintf(buf,BUFFMAX,"ERR OF");
     Serial.printf("temp Won't fit in buffer[%d] =%0.2f\n",BUFFMAX,tempC); 
    }   
    u8x8.drawString(0, 0, buf);
}
```

Chuck.

Mangulomx, 2019-09-18:
Very thanks by your effort. help me very much!

closed 2019-09-18 by Mangulomx