espressif/arduino-esp32 #3214: abort() was called at PC 0x40081e6b on core 1

mehrdad987, 2019-09-16:
I couldn't find any reason for the crash
is it for a code or any other reason

volatile int interruptCounter;
int totalInterruptCounter;
RTC_DS3231 rtc;
char daysOfTheWeek[7][12] = {"Sunday", "Monday", "Tuesday", "Wednesday", "Thursday", "Friday", "Saturday"};
RTC_DATA_ATTR int readingID = 0;
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;
void writeFile(fs::FS &fs, const char * path, const char * message){
    Serial.printf("Writing file: %s\n", path);
    File file = fs.open(path, FILE_WRITE);
    if(!file){
        Serial.println("Failed to open file for writing");
        return;
    }
    if(file.print(message)){
        Serial.println("File written");
    } else {
        Serial.println("Write failed");
    }
    //file.close();
}
void appendFile(fs::FS &fs, const char * path, const char * message){
    Serial.printf("Appending to file: %s\n", path);
    File file = fs.open(path, FILE_APPEND);
    if(!file){
        Serial.println("Failed to open file for appending");
        return;
    }
    if(file.print(message)){
        Serial.println("Message appended");
    } else {
        Serial.println("Append failed");
    }
    file.close();
}
void setup(){
  Serial.begin(115200);
  if(!SD.begin()){
        Serial.println("Card Mount Failed");
        return;
    }
  uint8_t cardType = SD.cardType();
  if(cardType == CARD_NONE){
        Serial.println("No SD card attached");
        return;
    }
  if (!rtc.begin()) {
  Serial.println("Couldn't find RTC");
  while (1);
    }
  if (rtc.lostPower()) {
    Serial.println("RTC lost power, lets set the time!");
    // following line sets the RTC to the date &amp; time this sketch was compiled
    rtc.adjust(DateTime(F(__DATE__), F(__TIME__)));
    // This line sets the RTC with an explicit date &amp; time, for example to set
    // January 21, 2014 at 3am you would call:
    //rtc.adjust(DateTime(2019, 8, 6, 16, 15, 0));
  }
  setset();
}
void IRAM_ATTR onTimer() {  
    Serial.print("An interrupt as occurred. Total number: ");
    Serial.println(totalInterruptCounter); 
    writeFile(SD, "/hello.txt", "Hello\n ");  
}
void setset(){
    timer = timerBegin(0, 80, true);
    timerAttachInterrupt(timer, &onTimer, true);
    timerAlarmWrite(timer, 1000000, true);
    timerAlarmEnable(timer);
}
void loop(){
    totalInterruptCounter++;
    DateTime now = rtc.now();
    String a = String(now.year());
    String b = String("-");
    String c = String(now.month());
    String d = String("-");
    String e = String(now.day());
    String f = String(".");
    String i = String(now.hour());
    String j = String(".");
    String k = String(now.minute());
    String l = String(".");
    String m = String(now.second());  
    String timeSec = (a + c + e + f + i +j + k + l + m);
    Serial.println(timeSec);
    delay(2000);
}

serial monitor{{

abort() was called at PC 0x40081e6b on core 1
 
Backtrace: 0x4008c738:0x3ffbbbd0 0x4008c965:0x3ffbbbf0 0x40081e6b:0x3ffbbc10 0x40081f65:0x3ffbbc40 0x400db19f:0x3ffbbc60 0x400dbc51:0x3ffbbdd0 0x4000bcc5:0x3ffbbdf0 0x400dc91d:0x3ffbbe10 0x400d1bfa:0x3ffbbe30 0x400d133b:0x3ffbbeb0 0x400d1233:0x3ffbbee0 0x40080ee5:0x3ffbbf20 0x40081cb1:0x3ffbbf40 0x40082355:0x3ffbbf60 0x400f1afb:0x3ffbc020 0x400dde7f:0x3ffbc040 0x4008a5fa:0x3ffbc060 0x4008874d:0x3ffbc080
 
 Rebooting...
 ets Jun  8 2016 00:22:57
 
 rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
 configsip: 0, SPIWP:0xee
 clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
 mode:DIO, clock div:1
 load:0x3fff0018,len:4
 load:0x3fff001c,len:1100
 load:0x40078000,len:9232
 load:0x40080400,len:6400
 entry 0x400806a8

beegee-tokyo, 2019-09-16:
Don't do
```
Serial.print("An interrupt as occurred. Total number: ");
Serial.println(totalInterruptCounter);
writeFile(SD, "/hello.txt", "Hello\n ");
```
from within an interrupt routine. That will 100% crash your application.
Instead set a flag and handle it from the loop() or from a task assigned to do it.

mehrdad987, 2019-09-16:
thank you 
i need append data to my file every 1 second
is there any other way?

beegee-tokyo, 2019-09-16:
You could run a task that is writing to a file. Inside the tasks loop you wait for a semaphore. In your interrupt routine you give the semaphore which will trigger your write event in the task.

Try to search for ESP32 tasks and semaphores to get examples

closed 2019-11-15 by mehrdad987