espressif/arduino-esp32 #3215: Build failed on ESP_IDF

boonkey, 2019-09-16:
### Hardware: (irrelevant for this issue)
Board: M5 ESP32CAM
Core Installation/update date: latest
IDE name: IDF-with Arduino
Flash Frequency: 40
Flash size: 4 MB
PSRAM enabled: ?no?
Upload Speed: ?115200?
Computer OS: Ubuntu 18.04.3 LTS

I'm trying to add the HTTPClient header file (and all associated files, libraries, methods, etc.) to M5's WIFI_STA project [(link)](https://github.com/m5stack/m5stack-cam-psram/tree/master/wifi/wifi_sta) in order to be able to send GET requests.

I followed the steps in the docs [(doc)](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md):
1. cloned the repo into components/ as arduino/
2. tried to make
I had the exact same issue as [WiFiClientSecure make issue #2154](https://github.com/espressif/arduino-esp32/issues/2154), so I followed the instructions there to solve. 
3. `Component Config -> mbedTLS -> TLS Key Exchange Methods -> 
  [*] Enable pre-shared-key ciphersuits
    [*] Enable PSK based ciphersuite modes`
I now still have building issues, appearing unrelated to the previous one. 
### LOG: 
`In file included from /home/user/esp/m5stack-cam-psram/wifi/wifi_sta/main/main.c:18:0:
/home/user/esp/m5stack-cam-psram/wifi/wifi_sta/components/arduino/libraries/HTTPClient/src/HTTPClient.h:32:18: fatal error: memory: No such file or directory
compilation terminated.
/home/user/esp/esp-idf/make/component_wrapper.mk:285: recipe for target 'main.o' failed
make[1]: *** [main.o] Error 1
/home/user/esp/esp-idf/make/project.mk:530: recipe for target 'component-main-build' failed
make: *** [component-main-build] Error 2
`
please help me to solve this issue.
Thanks

me-no-dev, 2019-09-16:
you forgot to mention the ESP-IDF version, but I will guess it's not 3.2 :)
Arduino tracks IDF branch release/3.2 currently

closed 2019-09-16 by me-no-dev

boonkey, 2019-09-18:
I changed ESP-IDF version to release/3.2
It did not solve the issue 
`In file included from /home/user/esp/m5stack-cam-psram/wifi/wifi_sta/main/main.c:18:0:
/home/user/esp/m5stack-cam-psram/wifi/wifi_sta/components/arduino/libraries/HTTPClient/src/HTTPClient.h:32:18: fatal error: memory: No such file or directory
compilation terminated.
/home/user/esp/esp-idf/make/component_wrapper.mk:285: recipe for target 'main.o' failed
make[1]: *** [main.o] Error 1
/home/user/esp/esp-idf/make/project.mk:530: recipe for target 'component-main-build' failed
make: *** [component-main-build] Error 2
`

boonkey, 2019-09-18:
just to be sure, I've **DELETED EVERYTHING** and started on a clean slate.
I have done most of my checking `make` commands without extra threads

Maybe I had something done wrong. I'm writing this as I'm doing it, so everything I write is what has just happend:

### Download ESP-IDF
1. `cd ~/esp`
2. `git clone -b release/v3.2 --recursive https://github.com/espressif/esp-idf.git`
3. `cd ~/esp/esp-idf`
4. `git submodule update --init --recursive`

### Clone the M5stack repo on which I'm basing my project:
1. `cd ~/esp`
2. `git clone https://github.com/m5stack/m5stack-cam-psram.git`
3. `cd ~/esp/m5stack-cam-psram/wifi/wifi_sta`
4. changed WiFi credentials to connect to my WiFi.
5. `make menuconfig && make flash`

after setting menuconfig as required, everything has compiled and ran as expected

### Clone this library into my project (following instruction from  [To use as a component of ESP-IDF](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md) :
1. `cd components && \
git clone https://github.com/espressif/arduino-esp32.git arduino && \
cd arduino && \
git submodule update --init --recursive && \
cd ../.. && \
make menuconfig`

2. I got a build error: 
`/home/dan/esp/m5stack-cam-psram/wifi/wifi_sta/components/arduino/libraries/WiFiClientSecure/src/ssl_client.cpp: In function 'int start_ssl_client(sslclient_context*, const char*, uint32_t, int, const char*, const char*, const char*, const char*, const char*)':`
`/home/dan/esp/m5stack-cam-psram/wifi/wifi_sta/components/arduino/libraries/WiFiClientSecure/src/ssl_client.cpp:149:67: error: 'mbedtls_ssl_conf_psk' was not declared in this scope
                  (const unsigned char *)pskIdent, strlen(pskIdent));`
`                                                                                              ^`
`/home/dan/esp/esp-idf/make/component_wrapper.mk:285: recipe for target 'libraries/WiFiClientSecure/src//ssl_client.o' failed`
`make[1]: *** [libraries/WiFiClientSecure/src//ssl_client.o] Error 1`
`make[1]: *** Waiting for unfinished jobs....`
`/home/dan/esp/esp-idf/make/project.mk:530: recipe for target 'component-arduino-build' failed`
`make: *** [component-arduino-build] Error 2`

I used the solution in [#2154](https://github.com/espressif/arduino-esp32/issues/2154)
3. `make menuconfig`
4. `Component Config -> mbedTLS -> TLS Key Exchange Methods -> `
`  [*] Enable pre-shared-key ciphersuits`
`    [*] Enable PSK based ciphersuite modes`

5. `make clean && make`

The make command failed with this error:
`Python requirements from /home/dan/esp/esp-idf/requirements.txt are satisfied.
/home/dan/esp/esp-idf/components/partition_table/Makefile.projbuild:60: warning: undefined variable 'quote'`
`Building partitions from /home/dan/esp/m5stack-cam-psram/wifi/wifi_sta/components/arduino/tools/partitions/default.csv...`
`Partitions defined in '/home/dan/esp/m5stack-cam-psram/wifi/wifi_sta/components/arduino/tools/partitions/default.csv' occupy 4.0MB of flash (4194304 bytes) which does not fit in configured flash size 2MB. Change the flash size in menuconfig under the 'Serial Flasher Config' menu.`
`/home/dan/esp/esp-idf/components/partition_table/Makefile.projbuild:60: recipe for target '/home/dan/esp/m5stack-cam-psram/wifi/wifi_sta/build/default.bin' failed`
`make: *** [/home/dan/esp/m5stack-cam-psram/wifi/wifi_sta/build/default.bin] Error 2`

I've verified that my device (ESP32CAM) has enough flash to 4MB, which it does.
6. `make menuconfig`

7. `Serial flasher config -> Flash size -> 4MB`

8. `make clean && make`

compilation passed without errors. I re-checked by running step 8 numerous times, all passing.

9. `make -j6 flash monitor`

run passed without errors

### Add use of this library to my code:
using [instructions from here](https://techtutorialsx.com/2017/05/19/esp32-http-get-requests/)
[... 61 lines cut ...]
                 from /home/dan/esp/m5stack-cam-psram/wifi/wifi_sta/components/arduino/libraries/WiFi/src/WiFi.h:32,
                 from /home/dan/esp/m5stack-cam-psram/wifi/wifi_sta/main/main.c:2:
/home/dan/esp/m5stack-cam-psram/wifi/wifi_sta/components/arduino/libraries/WiFi/src/WiFiGeneric.h:28:22: fatal error: functional: No such file or directory
compilation terminated.
/home/dan/esp/esp-idf/make/component_wrapper.mk:285: recipe for target 'main.o' failed
make[1]: *** [main.o] Error 1
/home/dan/esp/esp-idf/make/project.mk:530: recipe for target 'component-main-build' failed
make: *** [component-main-build] Error 2
`
3. I found bug [#116](https://github.com/espressif/arduino-esp32/issues/116) and followed instructions: change all my C files to C++

4. `make clean && make`

resulted in lots of `/home/dan/esp/m5stack-cam-psram/wifi/wifi_sta/main/main.cpp:73:1: sorry, unimplemented: non-trivial designated initializers not supported` errors. solving them included:
a. change all non-trivial designated initializers
b. adding `extern "C" ` to `void app_main()`
c. changing CMakeLists.txt

so far so good, the code compiled and ran

5. adding code section from the instructions, and `make`.

code is compiling. This bug is over.

@me-no-dev for future times, please don't dismiss so easily. the problem wasn't compiler version (but a whole different sets of problems)

Boldie, 2019-10-15:
Thanks for this "tutorial", I ran to the same problem with my port.

atanisoft, 2019-10-15:
> Thanks for this "tutorial", I ran to the same problem with my port. Since this is a config problem, why is this not catched in the code with a #ifndef XY #error A GOOD DESCRIPTION ....

@Boldie Why not send in a PR to add those sort of config checks to the client code ....