espressif/arduino-esp32 #3218: Unable to connect to ap with ssid of 32 characters length

jstammi, 2019-09-16:
#2746 # Hardware:
Board:							ESP32-WROVER-B
Core Installation version:			??? (n/a)
IDE name:						Platform.io
Flash Frequency:					??? (n/a)
PSRAM enabled:                                     yes
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
My ap's ssid is of 32 characters length, the max allowed AFAIK.

On connecting the serial console shows an error message: "[E][WiFiSTA.cpp:124] begin(): SSID too long or missing!"

WiFiSTA.cpp check length ssid > 31 ... seems to assume a trailing '\0'. But IMHO such is not enforced by the spec.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>
#include <WiFi.h>

void setup() {
  Serial.begin(115200);
  Serial.println(WiFi.begin("123456789A123456789B123456789C12", "passphrase"));
}

void loop() {
}

### Debug Messages:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6412
entry 0x400806a8
[E][WiFiSTA.cpp:124] begin(): SSID too long or missing!
4

atanisoft, 2019-09-16:
Per ESP-IDF this is intentional: https://github.com/espressif/esp-idf/issues/1176

closed 2019-11-30 by stale[bot]

jstammi, 2021-03-20:
> 
> 
> Per ESP-IDF this is intentional: [espressif/esp-idf#1176](https://github.com/espressif/esp-idf/issues/1176)

What is intenional? That it is impossible to connect to an AP with SSID length of 32 characters? But this is the allowed maximum SSID length ...

Today I came over this once again. But this time without an error message, the connection could not be established without giving any reasons on the console. With using shorter SSID everything works as expected (but I cannot change it in target environment!).

Please note that with using a ESP8266 instead, this does NOT face this problem.

lbernstone, 2021-03-20:
In 1.0.5 this will generate an error - https://github.com/espressif/arduino-esp32/blob/1.0.5/libraries/WiFi/src/WiFiSTA.cpp#L124
This can be revisited, as the functionality is there to support 32 characters now.

jstammi, 2021-03-21:
Yes, I can confirm that this causes an error. For a long time already. But only for ESP32. Not for ESP8266. And the "if" in referenced line is simply not correct as the SSID max length per standard is 32 characters. See §7.3.2.1 in https://www.iith.ac.in/~tbr/teaching/docs/802.11-2007.pdf

me-no-dev, 2021-03-23:
Fixed in master

jeffrey-meyer, 2022-12-16:
I get same error with an 18 character ssid.