espressif/arduino-esp32 #3222: esp_wifi_set_channel() not working in promiscuous mode without esp_wifi_start()

cyberman54, 2019-09-17:
According to API description esp_wifi_set_channel()

> @brief     Set primary/secondary channel of ESP32
> @attention 1. This is a special API for sniffer
> @attention 2. This API should be called after esp_wifi_start() or esp_wifi_set_promiscuous()

should work in promiscuous() mode without a preceding esp_wifi_start(), but it doesn't.
I did not see this behaviour before SDK Version v3.2.3, thus assuming that it came in with v3.2.3

closed 2019-11-30 by stale[bot]

rdmueller, 2021-02-13:
could it be that This has been fixed with newer versions of the board?
https://heltec-automation-docs.readthedocs.io/en/latest/esp32+arduino/wifi_kit_32/hardware_update_log.html#v2-1

I have boards wich still say "v2" on the PCB but the given battery read example with the PIN37 works.

cyberman54, 2021-02-13:
This references to issue #102 i guess.