espressif/arduino-esp32 #3225: v1.0.3 lost ability to connect to an access point; v1.0.2 works

ssilverman, 2019-09-18:
I'm trying to run a simple program that connects to a Wi-Fi access point. Quite simply, v1.0.2 can connect to a 2.4GHz access point but v1.0.3 cannot. I'm using the following code snippet, but the `WiFiClient` example works just as well for testing, after changing the SSID and password.

```c++
  Serial.printf("Connecting to \"%s\"...", kSSID);
  WiFi.begin(kSSID, kPassword);
  while (!WiFi.isConnected()) {
    Serial.print(".");
    delay(500);
  }
  Serial.println("connected.");
```

atanisoft, 2019-09-18:
@ssilverman do you have any logs to show why it failed? I can't reproduce this issue with any of my testing, nor can a lot of others who have also tested 1.0.3.

ssilverman, 2019-09-18:
I just tried these steps:
1. Brand new Arduino 1.8.10 install.
2. Deleted `~/Library/Arduino15` folder and reinstalled the latest ESP32 v1.0.3.
3. I'm using a _SparkFun ESP32 Thing Plus_, so I'm using the _Adafruit ESP32 Feather_ as my selected board.

I'm using the exact same code between a v1.0.2 test and a v1.0.3 test. The only difference is that the dots keep printing forever with v1.0.3 and this is printed once when I add "Info" debugging:
`[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT`

ssilverman, 2019-09-18:
I just turned on the "Debug" level and I get this:

```
Connecting to "Scrumtrelescent"...[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
.................[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
.............................
```

... and dots forevermore...

atanisoft, 2019-09-18:
Turn on the highest verbosity of the log level and rerun your test. Use the ESP32 DevKit-C board type (base board type), there is no meaningful difference for the SparkFun/Adafruit boards for this test.

The 4WAY_HANDSHAKE_TIMEOUT indicates that the esp32 is not able to complete the connection process and there may be additional details in the more verbose logs.

ssilverman, 2019-09-18:
I turned on the "Verbose" level and now it works. Changing back to the less verbose debug levels doesn't change now; it works. It didn't start working until I turned on this logging level, and now I can't get it to fail. This is the first time I've seen it work since I installed v1.0.3 (yesterday and lots and lots of tests). Weird.

Well, hopefully this helps someone who can't get their WiFi to connect: turn on Verbose logging, and then optionally back to where you had it. Maybe this was a coincidence with my specific setup? I'm at a loss to explain... :)

atanisoft, 2019-09-18:
It sounds like it is/was failing to complete the connection process and store the SSID data in NVS for later use. Perhaps the logging caused a timing difference which would be odd but not entirely unheard of.

ssilverman, 2019-09-18:
For kicks, I created a program to just do `WiFi.disconnect(true, true)` to erase the stored information, and then re-ran the original program. Now I can't get it to fail. Some internal state must have been different. Hmm...

atanisoft, 2019-09-18:
Something must have been corrupted in the NVS in that case, that is the only thing I can think of.

kapyaar, 2019-09-18:
I believe is related to https://github.com/espressif/arduino-esp32/issues/2025.

JC-85, 2019-11-19:
I just ran into this issue on a Adafruit Huzzah32. Out of three boards one would timeout with the  4WAY_HANDSHAKE_TIMEOUT error. Downgrading from version 1.0.4 to 1.0.2 in the Arduino IDE solved the problem.

Lopton, 2019-11-27:
I want to add that I am having this problem also on my Adafruit Huzzah32's. I just flashed 4 of them and all present this issue! It is super annoying because everything you flash them there is a chance that it will work, then once they are working then next flash may cause it to fail again.

[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT

Here is the weird part for me though, it will only fail when connecting to my home wifi which is a Ubiquiti Unifi system. If I turn my phone on as a hotspot it will always connect. However when I connect to my home network it fails, unless I flash it a ton of times and once in a random flash it will start connecting again. Once it connects, powering it off / on doesn't matter it will always connect, however then next time I flash it, it will roll the dice and not connect.

Lopton, 2019-11-27:
@ssilverman Thank you for the clue to go back to 1.0.2 that solved it and lets me flash my boards!!! 

If anyone needs me to test this I can pretty much reproduce it 100% of the time, but I'm still pretty new at this arduino thing, ie I looked for "ESP32 DevKit-C board type" mentioned earlier and I couldn't even find it :(

closed 2020-02-09 by stale[bot]

witnessmenow, 2020-03-14:
I'm after running into this problem on V1.0.4 using a Feather Huzzah32.

Using "HelloServer" as the example in all scenarios, only thing I change is my SSID and password

On V1.0.4, swapped to a different board and the exact same code worked fine.

Created a portable version of the Arduino IDE and installed V1.0.2, connects no problem on the Feather Huzzah

Then tried V1.0.4 on Huzzah again, didn't connect.

Any ideas what could be wrong?

Lopton, 2020-03-19:
I am just staying on V1.0.2. Maybe @atanisoft could provide further guidance on what we should be doing to troubleshoot this.

atanisoft, 2020-03-19:
@witnessmenow @Lopton about the only advice I can give is use a different board. A lot of people have a challenging time with the Feather line of boards, often with WiFi which appears to be signal strength or interference related

Gadgit83, 2020-04-30:
I've had a similar issue - code was working fine for quite a while, hadn't made any changes affecting wifi connection, uploaded, failed with reason 15, found this thread, downgraded to v1.0.2, re-uploaded and was able to connect no problem. I'm not using the feather board.

Arduino 1.8.12
Lolin Wemos D32 Mini clone

HamzaHajeir, 2020-09-15:
I think we'll need to murder this bot.

cagney, 2020-11-07:
<aol>me to</aol>
Blew away the entire .arduino directory and downloaded 1.0.2 and my Huzzah32 magically works :-(  Switch to 1.0.4 and broken.

jeanlucpacaud, 2020-11-21:
I had  also
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT

 I have  downgraded ESP2 to V.1.0.0 and it works!!!
(it didn't work with V.1.0.4 or with V.1.0.2)
now  it works with arduino V1.18.13  and ESP32 V.1.0.0  but i don't know why  ???

JEAN LUC     21 nov 2020