espressif/arduino-esp32 #3228: Navigate keys available in opened preferences

14geronimo, 2019-09-18:
In the Preferences library, is it possible to iterate through the keys that are available in opened preferences?

I would like to be able to do something like this, "iterator-style" if you will:

`Preferences preferences;`
`preferences.begin("config_01", true);`
`while (preferences.hasMoreKeys()) {`
`   char* key = preferences.nextKey();`
`   Serial.println(preferences.getString(key));`
`}`
`preferences.end();`

I have gone through the code for Preferences.cpp, and there isn't something obvious there that looks like a candidate for this functionality. Besides, there is no README.md doc for this library (is it missing? shall I write one up and submit a pull request?).

See also #2179, and pls indicate forum where to raise feature requests if not here.

lbernstone, 2019-09-19:
This is not currently possible in nvs, but is coming in esp-idf v4.0.  I'll try to have something ready for when that is available.
If you want to work on documentation, please feel free.  However, the proper way to document the library will be [doxygen](http://www.doxygen.nl/) tags, rather than a static README.md.  You can see how those are done in the [header files](https://github.com/espressif/esp-idf/blob/master/components/nvs_flash/include/nvs.h#L498-L528) and the end results in [esp-idf docs](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/storage/nvs_flash.html#).

14geronimo, 2019-09-20:
That's great news, thank you! And I'll look up Doxygen about the doc.

closed 2019-09-20 by 14geronimo

Xylopyrographer, 2022-05-05:
@dhavals1212 The API doc may help: https://docs.espressif.com/projects/arduino-esp32/en/latest/api/preferences.html
Associated tutorial: https://docs.espressif.com/projects/arduino-esp32/en/latest/tutorials/preferences.html
As long as you pass a `namespace` of a type that matches the `begin()` method, (and is no greater than 15 char) should be good to go.

If you're looking for a way to enumerate the namespaces and keys, all the nitty gritty details of ESP32 NVS are here: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/storage/nvs_flash.html

If you're looking to store or manipulate a large amount of data, might be better to look at using LittleFS or such.

mspieth, 2024-03-30:
What about something like this

```
class IterablePreferences : public Preferences {
public:
  IterablePreferences() = default;

  using Callback = void(const char* key, nvs_type_t type);
  void foreach (const char* nvNamespace, Callback cb, const char* nvPartition = "nvs") {
    nvs_iterator_t it = nvs_entry_find(nvPartition, nvNamespace, NVS_TYPE_ANY);
    while (it) {
      nvs_entry_info_t info{};
      nvs_entry_info(it, &info);  // Can omit error check if parameters are guaranteed to be non-NULL
      if (cb) {
        cb(info.key, info.type);
      }
      it = nvs_entry_next(it);
    }
    nvs_release_iterator(it);
  }
};
```
Use
```
IterablePreferences prefs;
prefs.foreach ("ns1", [](const char* key, nvs_type_t type) {
        String password = preferences.getString(key);
        Serial.printf(" found %s=%s\n", key, password.c_str());
      });
```