espressif/arduino-esp32 #3229: Don`t exist a bool confirmation of sucess WiFiClientSecure::setCACert()

JoaquimFlavio, 2019-09-19:
Hi, i`am Joaquim Flávio!

### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.3
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:					916000
Computer OS:                                        Mac OSX

### Description:
I need sucess/failed return of method `WiFiClientSecure::setCACert()`, `WiFiClientSecure::setPrivateKey()` and `WiFiClientSecure::setCertificate`. This return is present on **ESP8266** but don`t exist in **esp32**

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
  if(espClient.setPrivateKey(privkey)){
     Serial.println("sucess");
   }else{
     Serial.println("failed");
   }
}

void loop() {
}
```

### Debug Messages:
```
error: no matching function for call to 'WiFiClientSecure::setPrivateKey(fs::File&)'
     if (espClient.setPrivateKey(privKey)){ 
```

Jeroen88, 2019-09-19:
::setPrivateKey(const char *private_key) can't fail, because it just copies the pointer to the private key into a class variable. 
You can not call ::setPrivateKey() with a File &. If you want to use a key from a file, you should read it from the file into a char buffer[] and make sure the buffer is available throughout the life time of your espClient.

lbernstone, 2019-09-19:
The ::load* methods are available, but expect a stream and a length.  You should be able to call them with eg `wfcs.loadPrivateKey(keyfile, keyfile.size());`

JoaquimFlavio, 2019-09-19:
Very well, usin `WiFiClienteSecure::loadPrivateKey(keyfile, keyfile.size());` my code work!
Thanks @lbernstone!

JoaquimFlavio, 2019-09-19:
Very well, usin WiFiClienteSecure::loadPrivateKey(keyfile, keyfile.size()); my code work!
Thanks @lbernstone!

closed 2019-09-19 by JoaquimFlavio