espressif/arduino-esp32 #3230: ESP32 sketch data upload for fFAT

thedub2001, 2019-09-19:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

lbernstone, 2019-09-19:
See https://github.com/marcmerlin/esp32_fatfsimage.

thedub2001, 2019-09-19:
Thanks for your quick answer!
The tool looks complete, is there a possibility to turn it as a plugin ?

lbernstone, 2019-09-19:
Sure. Are you offering?

thedub2001, 2019-09-19:
No sorry :smiley:
I just wanted to point out the need for this tool !
Thanks a lot for your for your work!!

zekageri, 2019-09-25:
Ah i wanted to switch to FATFS so bad but because there are no plugins for it for arduino ide or PlatformIO i just can't use it. :( If there was a way to upload files to FATfs from browser like with SPIFFS i would be fine with that too. 
Btw i have an ESP with 16MB flash and with SPIFFS it is slow A.F. I had to reduce the SPIFFS size to 1mb to work properly.

closed 2019-12-08 by stale[bot]

marcmerlin, 2020-01-01:
@thedub2001 sorry, the tool I published only works on linux (and I didn't write it, I just packaged it). For a plugin it should have to be done in java I think.
Obviously it fixes my problem, and I have had other things to keep me busy. Hopefully someone else can do this for the java IDE.

magicmicros, 2020-04-18:
I just wrote a small python script to generate a filesystem e.t.c.
Should work on all platforms.
[https://github.com/magicmicros/mk_espfat](https://github.com/magicmicros/mk_espfat)

lbernstone, 2020-04-18:
@magicmicros : Your script will not work on esp32, as it does not include the [wear levelling](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/wear-levelling.html) layer, used by the fatfs components.

magicmicros, 2020-04-18:
I'm afraid you're mistaken, it's been working beautifully for a couple of weeks on a number of ESP32 boards.
The wear levelling might come into play AFTER a system has been created.
The initial filesystem will always be "clean".
I haven't read up on the wear levelling, but I'm pretty sure it's all done "outside" FAT, otherwise it would not be a FAT filesystem anymore.

There is no difference between the regular FFat.format() and the way I create the filesystem.
Well, except for my code not needed to to the same size calculations that the regular format does.

lbernstone, 2020-04-18:
very interesting.  I'll see what I can do about a plugin.