espressif/arduino-esp32 #3232: Missing requestFrom overloaded function

jrockinl, 2019-09-19:
### Hardware:
Board:							ESP32 and ESP 8266
Core Installation version:			all
IDE name:						Arduino IDE
Flash Frequency:					n/a
PSRAM enabled:                                    n/a
Upload Speed:					n/a
Computer OS:                                        n/a

### Description:
While trying to fix Arduino Due problem with a touch sensor MPR121, I found a function in the Wire library that handled the condition that failed.  The MPR121 library did not handle I2C Restart bits properly and actually created a stop and a start.  There are i2C devices that toss away data to be read if the read (preceded by a write of desired address) is not followed by Restart bit.

The fixed MPR121 library (Adafruit_MPR121) works for many boards now but seems if fails on ESP32 and ESP8266 because the Wire library doesn't have the following Arduino function:
uint8_t TwoWire::requestFrom(uint8_t address, uint8_t quantity, uint32_t iaddress, uint8_t isize, uint8_t sendStop)

### Sketch: 

[MPR121test.zip](https://github.com/espressif/arduino-esp32/files/3632683/MPR121test.zip)

stickbreaker, 2019-09-19:
@jrockinl the official Arduino reference [arduino.cc WireRequestFrom](https://www.arduino.cc/en/Reference/WireRequestFrom) does not list that variant?  Where is it defined?

It looks to me that adafruit made a variant of the `Wire()` object.

Chuck.

jrockinl, 2019-09-20:
No, it wasn't adafruit that created it.  I found it in the arduino wire.cpp.h in the official Arduino files for at very least, Uno, Mega2560 (1.6.23) and Due (1.6.12).

stickbreaker, 2019-09-20:
@jrockinl It's not part of documented [arduino/ArduinoCore-API](https://github.com/arduino/ArduinoCore-API/issues/92) .
Someone snuck it in, they are arguing about officially including it.

Chuck.

closed 2019-12-03 by stale[bot]