espressif/arduino-esp32 #3233: OLED and RTC DS3231 not working great.

JoanPedro, 2019-09-19:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do? 

Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board: ESP32 TTGO LoRa OLED
Core Installation version: 1.0.1-rc4
IDE name: Arduino IDE 1.8.09
Computer OS: Windows 10

### Description:
I have a communication problem between my ESP device. I am using Wifi, LoRa, OLED, and RTC DS3231. Follows: The device cannot keep OLED or RTC fully operational. Either works either.

Tested:
1. RTC LIBRARY EXAMPLE, OK
2. OLED LIBRARY EXAMPLE, OK
3. WIFI + LORA + RTC, OK
4. WIFI + LORA + OLED, OK
5. WIFI + LORA + OLED + RTC, NOT OK.

### Sketch:  
```cpp

                         //----------------*Includes of Librarys*----------------\\

    // Load Wi-Fi library.
    #include <WiFi.h>
    // Load SPI library.
    #include <SPI.h>
    //Load LoRa library.
    #include <LoRa.h>
    // Load Wire and 3Wire library.
    #include <Wire.h> 
    // Loard RTC - Clock library .
    #include <RtcDS3231.h> // DS3231 Module -  
    // Load Json utitily library
    #include <ArduinoJson.h>
    // Load SSD1306 Display library.
    #include "SSD1306.h" 
    // Load pack images library .
   // #include "images.h"

                         //----------------*Defines of Pins usage:*----------------\\
                   /*in LoRa module*/
  #define SCK     5    // GPIO5  -- SX1276 SCK.
  #define MISO    19   // GPIO19 -- SX1276 MISnO.
  #define MOSI    27   // GPIO27 -- SX1276 MOSI.
  #define SS      18   // GPIO18 -- SX1276 CS.
  #define RST     14   // GPIO14 -- SX1276 RESET.
  #define DI0     26   // GPIO26 -- SX1276 IRQ(Interruption Requirement).
  // Band bound rate 915x10^6
  #define BAND  915E6
  // Json usage 
  #define countof(a) (sizeof(a) / sizeof(a[0]))
  
                    /*in Wire module*/
                    /*in RTC - Clock module*/
  RtcDS3231<TwoWire> Rtc(Wire);
                    /*in SSD1306 Display - module*/
[... 400 lines cut ...]
    // you may also get the temperature as a float and print it
    // Serial.print(temp.AsFloatDegC());
    Serial.println("C");
    */
    Serial.println("Core2");
    delay(1000);

  }
}
```

### Debug Messages:
```
In this Code, the OLED Display crash with message in display:  LoRa Initial success!
1. LoRa message sent by other ESP: Received
2. Command by WiFi: Working
3. RTC in other core: Working
4. OLED: Crashed

```

[DS3231_RL.txt](https://github.com/espressif/arduino-esp32/files/3633449/DS3231_RL.txt)
[CASA STATION.txt](https://github.com/espressif/arduino-esp32/files/3633450/CASA.STATION.txt)
[REMOTE STATION.txt](https://github.com/espressif/arduino-esp32/files/3633451/REMOTE.STATION.txt)
[esp8266-oled-ssd1306-master.zip](https://github.com/espressif/arduino-esp32/files/3633452/esp8266-oled-ssd1306-master.zip)

stickbreaker, 2019-09-20:
@JoanPedro What value of pullup resistors are you using on SCL and SDA?

Chuck.

stickbreaker, 2019-09-20:
@JoanPedro I use a SSD1306, 3 MCP23008 Expanders, 8 24LCxx EEPROMS, a DS1307 RTCC, MMA8452 Accel, SHT25 temp/hum all on the same I2C bus. I use 3.3k pullups on the 3.3v branch and 10k on the 5v (MCP23008's) branch.

Chuck.

stickbreaker, 2019-09-20:
You do realize that `Wire()` is single threaded? and you can't do two different things at the same time with one piece of hardware?  If I am understanding your program you are trying to display on a SSD1306 and read from a RTC using the same hardware without any synchronization between two separate tasks?  Either interleave your access or use both I2C peripherals (`Wire()` and `Wire1()`). 
Also, you init the SSD1306 using the Default SDA and SCL pins, then RECONFIGURE `Wire()` with custom pins at the bottom of your setup?  How do you expect the SSD1306 to magically receive communications when you disconnect it?

Chuck.

JoanPedro, 2019-09-21:
I want apologize for the inconvenience. Beginner in this world. Thank you @stickbreaker.
Problem solved.

closed 2019-09-21 by JoanPedro