espressif/arduino-esp32 #3237: adding new libs

lagunax, 2019-09-21:
hi
trying to make project for TTGO-T-Display.
how can i add TFT_eSPI to libraries? and is there any way to simply add library without manualy editing arduino-esp32 config files?
when i cloning tft_espi repo into library folder it is do not appeer in menuconfig until i modify those configs.
thanks

lbernstone, 2019-09-21:
Your libraries go into <Arduino>/libraries, not <esp32>/libraries.  Your best bet is to use the library manager in the Arduino IDE (Sketch->Include Library->Manage Libraries)

lagunax, 2019-09-21:
> Your libraries go into <Arduino>/libraries, not <esp32>/libraries. Your best bet is to use the library manager in the Arduino IDE (Sketch->Include Library->Manage Libraries)

i'm using Eclipse (also jast console with 'make && make flash') and pasting files in arduino/libraries (in [project]/components/arduino/libraries).

one way i found to make this easer is to make script to modify CMakeLists.txt anf Kconfig.projbuild so i can do it with script. but i think that it mast be another way for adding library (i thought it mast be autoadded due menuconfig or something like that, but nop, without editing those 2 files menuconfig whon't see new library)

atanisoft, 2019-09-21:
If you are using Eclipse you are likely compiling arduino as a component of IDF. In this case you would need to have the libraries themselves enhanced to include CMakeLists.txt and KConfig related files to have them picked up automatically by the IDF build process.

This is not a responsibility of arduino-esp32 to support 100% of arduino like libraries. There are only a handful that are packaged with arudino-esp32 as they are customized or otherwise specific to the esp32.

lagunax, 2019-09-21:
ok. will make script to automate adding libs later on the week. will post link to git repo here.

lagunax, 2019-09-22:
started making script.
for now on linux it installs in current dirrectory: esp-idf, arduino-esp32, xtensa. then it installs TFT_eSPI as library (also adds [https://github.com/Xinyuan-LilyGO/TTGO-T-Display](https://github.com/Xinyuan-LilyGO/TTGO-T-Display) to it) but TFT_eSPI is bougus and need to be edited (later will remake it and store in my repo with new name)
[https://github.com/lagunax/SPIDisplay](https://github.com/lagunax/SPIDisplay)

closed 2019-09-22 by lagunax