espressif/arduino-esp32 #3238: ESP-NOW to Android SDK comms

skippa, 2019-09-21:
Are there any plans to build an Android library so I can use ESP-NOW to connect directly with an Android phone? If not, where can I start? Are there any documents which describe the ESP-NOW packets / frames?

atanisoft, 2019-09-21:
ESP-NOW is only for the esp8266 from what I can tell. It is not supported/used on the esp32.

atanisoft, 2019-09-21:
of course, you could also use google to find an arduino library that might work...

lbernstone, 2019-09-21:
It is definitely available in esp32 (https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/ESPNow), but is a proprietary espressif protocol.

skippa, 2019-09-21:
I know there's an Android app for connecting to ESP-MESH networks (which uses ESP-NOW-like frames AFAIK). I was hoping this meant that somewhere is an Android library I can use for ESP-NOW comms

beegee-tokyo, 2019-09-22:
Not sure if it is a library or just some example code:

https://github.com/EspressifApp/IOT-Espressif-Android/blob/master/java/TYPE/com/espressif/iot/type/device/status/IEspStatusEspnow.java

closed 2019-12-05 by stale[bot]

kakopappa, 2020-03-11:
@skippa Did you manage to send data from Android <-> ESP ? I am working on something similar.

vitorrussi, 2020-05-14:
@kakopappa hey, having some progress? I want to start something with esp32 and flutter, and i was wondering if its possible or not.

kakopappa, 2020-05-14:
Not clear what’s your question here

 Sinric Pro doesn’t have a Android SDK (yet)

Last time I checked ESP-NOW works between two ESP devices only and Android
or iOS is not supported yet

Best option was to start a web server in the ESP module and communicate via
GET or POST requests. Another option is UDP, which is a hit and a miss

gauravarya83, 2020-09-23:
@kakopappa Socket is another option, right ?