espressif/arduino-esp32 #3239: gcc 8.2.0 with latest esp-idf makes arduino-esp32 with many errors

lagunax, 2019-09-21:
### Hardware:
Board:							TTGO-T-Display (but not metter, becous it is not hardware specific)
Core Installation version:			git clone https://github.com/espressif/arduino-esp32.git (21.september.2019)
IDE name:						Eclipse/console make + esp-idf 3.0.2
Computer OS:                                        Linux 4.9.0-9-rt-amd64 #1 SMP PREEMPT RT Debian 4.9.168-1+deb9u5 (2019-08-11) x86_64 GNU/Linux

### Description:
when i 'make' i getting next error:

`
make
Toolchain path: /home/producer/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp32-2019r1
Compiler version: 8.2.0
Python requirements from /usr/media/Programming/esp/esp-idf/requirements.txt are satisfied.

App "app-template" version: 0080e9d-dirty
/usr/media/Programming/esp/Projects/homeauto/components/arduino/component.mk:9: предупреждение: неопределённая переменная «CONFIG_ARDUINO_SELECTIVE_HTTPUpdate»
Arduino libraries in build:  AsyncUDP  BLE BluetoothSerial  EEPROM ESP32  FFat FS   NetBIOS Preferences   SPI SPIFFS SimpleBLE TFT_eSPI   WebServer WiFi WiFiClientSecure Wire
CC build/arduino/cores/esp32/wiring_shift.o
In file included from /usr/media/Programming/esp/Projects/homeauto/components/arduino/cores/esp32/wiring_private.h:31,
                 from /usr/media/Programming/esp/Projects/homeauto/components/arduino/cores/esp32/wiring_shift.c:21:
/usr/media/Programming/esp/Projects/homeauto/components/arduino/cores/esp32/Arduino.h:121:6: error: conflicting types for 'random'
 long random(long, long);
      ^~~~~~
In file included from /usr/media/Programming/esp/Projects/homeauto/components/arduino/cores/esp32/esp32-hal.h:30,
                 from /usr/media/Programming/esp/Projects/homeauto/components/arduino/cores/esp32/wiring_shift.c:20:
/home/producer/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/xtensa-emake
Toolchain path: /home/producer/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp32-2019r1
Compiler version: 8.2.0
Python requirements from /usr/media/Programming/esp/esp-idf/requirements.txt are satisfied.

App "app-template" version: 0080e9d-dirty
/usr/media/Programming/esp/Projects/homeauto/components/arduino/component.mk:9: предупреждение: неопределённая переменная «CONFIG_ARDUINO_SELECTIVE_HTTPUpdate»
Arduino libraries in build:  AsyncUDP  BLE BluetoothSerial  EEPROM ESP32  FFat FS   NetBIOS Preferences   SPI SPIFFS SimpleBLE TFT_eSPI   WebServer WiFi WiFiClientSecure Wire
CC build/arduino/cores/esp32/wiring_shift.o
In file included from /usr/media/Programming/esp/Projects/homeauto/components/arduino/cores/esp32/wiring_private.h:31,
                 from /usr/media/Programming/esp/Projects/homeauto/components/arduino/cores/esp32/wiring_shift.c:21:
/usr/media/Programming/esp/Projects/homeauto/components/arduino/cores/esp32/Arduino.h:121:6: error: conflicting types for 'random'
 long random(long, long);
      ^~~~~~
In file included from /usr/media/Programming/esp/Projects/homeauto/components/arduino/cores/esp32/esp32-hal.h:30,
                 from /usr/media/Programming/esp/Projects/homeauto/components/arduino/cores/esp32/wiring_shift.c:20:
/home/producer/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:267:6: note: previous declaration of 'random' was here
 long random (void);
      ^~~~~~
/usr/media/Programming/esp/esp-idf/make/component_wrapper.mk:291: ошибка выполнения рецепта для цели «cores/esp32/wiring_shift.o»
make[1]: *** [cores/esp32/wiring_shift.o] Ошибка 1
/usr/media/Programming/esp/esp-idf/make/project.mk:584: ошибка выполнения рецепта для цели «component-arduino-build»
make: *** [component-arduino-build] Ошибка 2
sp32-elf/sys-include/stdlib.h:267:6: note: previous declaration of 'random' was here
 long random (void);
      ^~~~~~
/usr/media/Programming/esp/esp-idf/make/component_wrapper.mk:291: ошибка выполнения рецепта для цели «cores/esp32/wiring_shift.o»
make[1]: *** [cores/esp32/wiring_shift.o] Ошибка 1
/usr/media/Programming/esp/esp-idf/make/project.mk:584: ошибка выполнения рецепта для цели «component-arduino-build»
make: *** [component-arduino-build] Ошибка 2
`

when i comments line "long random(long, long);" in arduino.h, i got another error:

`
/usr/media/Programming/esp/Projects/homeauto/components/arduino/libraries/BLE/src/BLEDevice.cpp: In static member function 'static void BLEDevice::whiteListAdd(BLEAddress)':
/usr/media/Programming/esp/Projects/homeauto/components/arduino/libraries/BLE/src/BLEDevice.cpp:493:75: error: too few arguments to function 'esp_err_t esp_ble_gap_update_whitelist(bool, uint8_t*, esp_ble_wl_addr_type_t)'
  esp_err_t errRc = esp_ble_gap_update_whitelist(true, *address.getNative()); //True to add an entry.
                                                                           ^
In file included from /usr/media/Programming/esp/Projects/homeauto/components/arduino/libraries/BLE/src/BLEDevice.cpp:17:
/usr/media/Programming/esp/esp-idf/components/bt/host/bluedroid/api/include/api/esp_gap_ble_api.h:956:11: note: declared here
 esp_err_t esp_ble_gap_update_whitelist(bool add_remove, esp_bd_addr_t remote_bda, esp_ble_wl_addr_type_t wl_addr_type);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/media/Programming/esp/Projects/homeauto/components/arduino/libraries/BLE/src/BLEDevice.cpp: In static member function 'static void BLEDevice::whiteListRemove(BLEAddress)':
/usr/media/Programming/esp/Projects/homeauto/components/arduino/libraries/BLE/src/BLEDevice.cpp:507:76: error: too few arguments to function 'esp_err_t esp_ble_gap_update_whitelist(bool, uint8_t*, esp_ble_wl_addr_type_t)'
  esp_err_t errRc = esp_ble_gap_update_whitelist(false, *address.getNative()); //False to remove an entry.
                                                                            ^
In file included from /usr/media/Programming/esp/Projects/homeauto/components/arduino/libraries/BLE/src/BLEDevice.cpp:17:
/usr/media/Programming/esp/esp-idf/components/bt/host/bluedroid/api/include/api/esp_gap_ble_api.h:956:11: note: declared here
 esp_err_t esp_ble_gap_update_whitelist(bool add_remove, esp_bd_addr_t remote_bda, esp_ble_wl_addr_type_t wl_addr_type);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
`
editing BLEDevice.cpp
`        esp_err_t errRc = esp_ble_gap_update_whitelist(true, *address.getNative()), BLE_WL_ADDR_TYPE_PUBLIC);  // True to add an entry.`

next same with _update_blacklist (below this function in same file). editing - same (add ", BLE_WL_ADDR_TYPE_PUBLIC")

next:
`
/usr/media/Programming/esp/Projects/homeauto/components/arduino/libraries/BLE/src/BLEAdvertisedDevice.cpp:` In member function 'std::__cxx11::string BLEAdvertisedDevice::toString()':
/usr/media/Programming/esp/Projects/homeauto/components/arduino/libraries/BLE/src/BLEAdvertisedDevice.cpp:509:30: error: 'snprintf' output may be truncated before the last format character [-Werror=format-truncation=]
   snprintf(val, sizeof(val), "%d", getTXPower());
`

modifing BLEAdvertisedDevice.cpp:
adding
`
#ifndef snprintf_nowarn(...)
#define snprintf_nowarn(...) (snprintf(__VA_ARGS__) < 0 ? abort() : (void)0)
#endif
`
adding _nowarn to all snprintf

Next:
`
CXX build/arduino/libraries/BLE/src//BLEUUID.o
/usr/media/Programming/esp/Projects/homeauto/components/arduino/libraries/BLE/src/BLEUUID.cpp: In member function 'std::__cxx11::string BLEUUID::toString()':
/usr/media/Programming/esp/Projects/homeauto/components/arduino/libraries/BLE/src/BLEUUID.cpp:372:22: error: '%02x' directive output truncated writing 2 bytes into a region of size 1 [-Werror=format-truncation=]
  snprintf(hex, size, "%02x%02x%02x%02x-%02x%02x-%02x%02x-%02x%02x-%02x%02x%02x%02x%02x%02x",
`

doing same as last? but getting next... (not helped)
`
/usr/media/Programming/esp/Projects/homeauto/components/arduino/libraries/BLE/src/BLEUUID.cpp:19:39: error: '%02x' directive output truncated writing 2 bytes into a region of size 1 [-Werror=format-truncation=]
 #define snprintf_nowarn(...) (**snprintf(__VA_ARGS__)** < 0 ? abort() : (void)0)
`
and now i'm stuck
i know that downgrade of gcc and removing some flags will fix issue, but, realy, let prepare code to above vertions of compillers. how to fix last issue, who knows?

lbernstone, 2019-09-21:
The newest supported version of esp-idf is v3.2.3.  This uses the older gcc toolchain.  Once v4.0 gets to stable, the new toolchain will be supported.  This is a downstream, community project.  We do not do experimental builds.

lagunax, 2019-09-21:
> The newest supported version of esp-idf is v3.2.3. This uses the older gcc toolchain. Once v4.0 gets to stable, the new toolchain will be supported. This is a downstream, community project. We do not support experimental builds.

where to get toolchain for v3.2.3? espressif have link only for latest (that i have), and it makes many errors :(
https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_2_0-esp32-2019r1-linux-amd64.tar.gz

lbernstone, 2019-09-21:
There already is a script:
https://github.com/espressif/arduino-esp32/blob/master/tools/get.py

lagunax, 2019-09-22:
> There already is a script:
> https://github.com/espressif/arduino-esp32/blob/master/tools/get.py

thanks

now making script to automate full installation with esp-idf and adding lib TFT_eSPI (later will be edited copy)

here link
[https://github.com/lagunax/SPIDisplay](https://github.com/lagunax/SPIDisplay)

closed 2019-09-22 by lagunax

lbernstone, 2019-09-22:
There's a script for that too :smile:
https://github.com/espressif/esp32-arduino-lib-builder.git

lagunax, 2019-09-23:
> There's a script for that too smile
> https://github.com/espressif/esp32-arduino-lib-builder.git

im late ((((((
XD