espressif/arduino-esp32 #3247: WiFi.status() returns WL_CONNECTED - even when no longer connected

marrold, 2019-09-22:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.3
IDE name:						Platform.io
Platformio Platform                                 espressif32@1.10.0
Flash Frequency:					?
Upload Speed:					115200
Computer OS:                                        Mac OSX

### Description:

I have been troubleshooting a WiFi issue where by my ESP32 _thinks_ it's connected , even when it isn't, and discovered `WiFi.status()` returns `WL_CONNECTED` - even when no longer connected.

I am using a Mikrotik hAP AC router, and if to replicate the issue I can disable the WiFi or apply an ACL to ban the ESP32. WiFi.status() will continue to indicate it's connected. This sounds similar to #1879

It appears to be related to the Access Point, as when I connect to another device, the ESP32 is able to detect that the Wifi has disconnected.

Thanks

lbernstone, 2019-09-22:
Does it work properly if you unplug the router instead of software disable?

marrold, 2019-09-22:
If the access point is totally powered off the ESP32 successfully detected disconnection, but in other scenarios where it becomes disconnected from WiFi for whatever reason, it does not.

marrold, 2019-09-22:
One observation, if I reboot the ESP32 and it tries to connect to the WiFi whilst its banned, it fills the serial console with `E (646775) wifi: Set status to INIT`

Which makes me wonder if the Wireless is _connected_ but not _authenticated_ and `WiFi.status()` is not making this distinction.

martinius96, 2019-09-28:
@marrold have you tried?

closed 2019-12-11 by stale[bot]