espressif/arduino-esp32 #3248: Using files (SPIFFS) to setCertificate.

JoaquimFlavio, 2019-09-23:
I need get file in the flash of ESP32 to set certificate for connect to my Broker MQTT, but the code return me a connecte error. I search in the network for sollution but don`t found...

I try using `loadCertificate` and `setCertificate` but both return the ssl error in the monitor: 
```
rc=-2
  WiFiClientSecure SSL error: X509 - The CRT/CRL/CSR format is invalid, e.g. different type expected
```

ferologics, 2019-10-14:
Hey @JoaquimFlavio have you been able to resolve your issue? I'm receiving the same error

JoaquimFlavio, 2019-10-14:
@ferologics I don't resolving this complete problem. The `loadCertificate` method don't working in ESP32, but work in ESP8266! I do not know the reason occurs....
For resolve my problem i need copy the file to char array and poiter to null after end file. I write a simple example:
```
char array[max lenght of file + 1];//this variable need be global
int i=0;
while(begin to end file){
    array[i] = file.read();
    i++;
}
array[i] = '\0';

esp. setCertificate(array);
````

KevinHunter12, 2019-11-21:
I also have the same problem :(
[V][ssl_client.cpp:56] start_ssl_client(): Free internal heap before TLS 257816
[V][ssl_client.cpp:58] start_ssl_client(): Starting socket
[V][ssl_client.cpp:93] start_ssl_client(): Seeding the random number generator
[V][ssl_client.cpp:102] start_ssl_client(): Setting up the SSL/TLS structure...
[V][ssl_client.cpp:115] start_ssl_client(): Loading CA cert
[E][ssl_client.cpp:33] _handle_error(): [start_ssl_client():122]: (-8576) X509 - The CRT/CRL/CSR format is invalid, e.g. different type expected
[E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -8576
[V][ssl_client.cpp:248] stop_ssl_socket(): Cleaning SSL connection.
---> mqtt failed, rc=-2

If i connect to the same server using the same certificate but HTTPS rather than MQTTS it works !

Im reading the certificate from SPIFFS and loading using wifiClient.loadCACert(ca, ca.size())

Any clues??
Thanks
Kevin

closed 2020-02-03 by stale[bot]

pulquero, 2020-06-01:
I also have this problem, specifically when trying to connect to AWS IoT with cacert, cert and private key.

pulquero, 2020-06-03:
Found the problem! Fix is edit C:\Users\<...>\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFiClientSecure\src\WiFiClientSecure.cpp and modify
```
char *WiFiClientSecure::_streamLoad(Stream& stream, size_t size) {
  /* original code uses a static ptr so the next load overwrites the previous certificate!!!
  static char *dest = nullptr;
  if(dest) {
      free(dest);
  }
  */
  char* dest = (char*)malloc(size);
  if (!dest) {
    return nullptr;
  }
  if (size != stream.readBytes(dest, size)) {
    free(dest);
    dest = nullptr;
  }
  return dest;
}
```

@me-no-dev please re-open and mark as a bug.

lbernstone, 2020-06-03:
Please submit as a PR.  Bugs don't fix themselves around here.

JaronrH, 2020-06-17:
Thank you, @pulquero!  

This fixed my issue with trying to connect to MQTT using PubSubClient and cert verification.

cjkarande, 2020-12-20:
@pulquero , well i am still getting the same error even after the said code change. I am using esp32s2 code branch. Am i missing something?

> Found the problem! Fix is edit C:\Users<...>\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFiClientSecure\src\WiFiClientSecure.cpp and modify
>
> ```
> char [...]

NarinLab, 2021-03-09:
Same on here, how can i load spiffs ca.cert.pem to pass to setCACert method:

```
#include <WiFiClientSecure.h>
#include <MQTTClient.h>

//short code
ssl.setCACert("/ca.cert.pem");
iot.setOptions(900, true, 10000);
```

pulquero, 2021-03-09:
I use something like this to load the certificates:

```
template<typename L> void loadFromFile(const char* fname, L&& load) {
  if (SPIFFS.exists(fname)) {
    File f = SPIFFS.open(fname);
    bool rc = load(f, f.size());
    f.close();
  }
}

void loadCertificates(WiFiClientSecure* client) {
  SPIFFS.begin();
  loadFromFile("/ca.cert.pem", [client](Stream& stream, size_t size){return client->loadCACert(stream, size);});
  loadFromFile("/client.cert.pem", [client](Stream& stream, size_t size){return client->loadCertificate(stream, size);});
  loadFromFile("/private.key.pem", [client](Stream& stream, size_t size){return client->loadPrivateKey(stream, size);});
  SPIFFS.end();
}

`