espressif/arduino-esp32 #3250: wifi library using freeRTOS

vignesh13404387, 2019-09-23:
can we use WiFi.h library in FreeRTOS while creating a separate task?

me-no-dev, 2019-09-23:
yes. wifi already runs in it's own task though ;)

closed 2019-09-23 by me-no-dev

vignesh13404387, 2019-09-23:
#include <WiFi.h>
void wifi(void *pvParameters);
TaskHandle_t wifi_connect;
void setup() {
  Serial.begin(115200);
  xTaskCreate(wifi, "wifi connection", 1024, NULL, 1, &wifi_connect);
}

void loop() {
  vTaskDelay(2000);
}

void wifi(void *pvParameters) {
  for ( ;; ) {
    const char* ssid = "ACT02";
    const char* pass = "fbwelcome123";
    WiFi.begin(ssid, pass);
    while (WiFi.status() != WL_CONNECTED ) {
      vTaskDelay(500);
      Serial.print(".");
    }
    Serial.println("WiFi connected");
    Serial.print("IP address is ");
    Serial.println(WiFi.localIP());
  }
}

me-no-dev, 2019-09-23:
why? because you are running connect in a busy loop? why do you connect, then connect again and again and again?

vignesh13404387, 2019-09-23:
what will happen if I create a task without for loop?

me-no-dev, 2019-09-24:
Why do you use vTaskDelay? it depends on tick configuration ;) 
```cpp
#include <WiFi.h>
void wifi(void *pvParameters);
TaskHandle_t wifi_connect;
void setup() {
  Serial.begin(115200);
  xTaskCreate(wifi, "wifi connection", 4096, NULL, 1, &wifi_connect);
}

void loop() {
  delay(2000);
}

void wifi(void *pvParameters) {
  const char* ssid = "ACT02";
  const char* pass = "fbwelcome123";
  WiFi.begin(ssid, pass);
  while (WiFi.status() != WL_CONNECTED ) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("WiFi connected");
  Serial.print("IP address is ");
  Serial.println(WiFi.localIP());
  for ( ;; ) {
    delay(1000);
  }
}
```

Rafinha29, 2022-03-21:
> yes. wifi already runs in it's own task though ;)

in which core is running?