espressif/arduino-esp32 #3251: created separate task to connect with wifi

vignesh13404387, 2019-09-23:
#include <WiFi.h>
void wifi(void *pvParameters);
TaskHandle_t wifi_connect;

const char* ssid = "ssid";
const char* pass = "pass";
void setup() {
  Serial.begin(115200);
  xTaskCreate(wifi, "wifi_connection", 1024, NULL, 1, &wifi_connect);
}

void loop() {
  vTaskDelay(2000);
}

void wifi(void *pvParameters) {
  for ( ;; ) {
    WiFi.enableSTA(true);
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, pass);
    while (WiFi.status() != WL_CONNECTED ) {
      vTaskDelay(500);
      Serial.print(".");
    }
    Serial.println("WiFi connected");
    Serial.print("IP address is ");
    Serial.println(WiFi.localIP());
  }
}

me-no-dev, 2019-09-23:
no need to open more than one ticket ;) you are having and issue with your code and we are discussing this in another ticket.

closed 2019-09-23 by me-no-dev

rgov, 2024-09-26:
For other search engine wanderers: The discussion happened in https://github.com/espressif/arduino-esp32/issues/3250 and the conclusion was that Wi-Fi already runs in its own task.