espressif/arduino-esp32 #3253: Missing Partition Scheme Options for Adafruit Feather HUZZA32 in boards.txt

tpitman, 2019-09-23:
### Hardware: Adafruit HUZZAH32
Board:							Adafruit HUZZAH32
Core Installation version:			1.0.3 (all actually)
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                    yes
Upload Speed:					115200
Computer OS:                                        All

### Description:
The boards.txt file doesn't have the partition scheme section for the board type:
featheresp32.name=Adafruit ESP32 Feather

Here is an example of one I use that could be included:

`
featheresp32.menu.PartitionScheme.default=Default
featheresp32.menu.PartitionScheme.default.build.partitions=default
featheresp32.menu.PartitionScheme.minimal=Minimal (2MB FLASH)
featheresp32.menu.PartitionScheme.minimal.build.partitions=minimal
featheresp32.menu.PartitionScheme.no_ota=No OTA (Large APP)
featheresp32.menu.PartitionScheme.no_ota.build.partitions=no_ota
featheresp32.menu.PartitionScheme.no_ota.upload.maximum_size=2097152
featheresp32.menu.PartitionScheme.min_spiffs=Minimal SPIFFS (Large APPS with OTA)
featheresp32.menu.PartitionScheme.min_spiffs.build.partitions=min_spiffs
featheresp32.menu.PartitionScheme.min_spiffs.upload.maximum_size=1966080
featheresp32.menu.PartitionScheme.fatflash=16M Fat
featheresp32.menu.PartitionScheme.fatflash.build.partitions=ffat
`

atanisoft, 2019-09-23:
feel free to send in a PR with the suggested updates. "fatflash" may not work on the Feather boards unless they come in a 16MB variant (which I don't believe it does)

tpitman, 2019-09-23:
Will do. You are probably right. I copied that from another section because I specifically needed the min SPIFFS.

I will remove fatflash and do a PR.

Thank you.

tpitman, 2019-09-23:
I don't think I have the right permissions to create a pull request. I made my changes locally in a branch and tried to push the branch. That gave a permissions error.

I then followed some instructions for creating a pull request and when I do the create pull request in github desktop it takes me to a web page on the project where you have to select the branch to compare. Since my branch isn't pushed I can't pick it.

Sorry for being a noob. I have never contributed to a github project that wasn't my own.

Can you give me some instructions? I am on macos with github desktop. I cloned a clean copy, made my changes and created my branch.

atanisoft, 2019-09-23:
go to https://github.com/espressif/arduino-esp32/blob/master/boards.txt and click the pencil (edit) icon. Make the edits and at the bottom of the page it will have an option to create a PR from your edits.

tpitman, 2019-09-23:
Easy. Thanks. It is processing now.

closed 2019-12-06 by stale[bot]