espressif/arduino-esp32 #3256: Build against latest idf 4.1 master

rbiernat, 2019-09-24:
Hi

Im using idf 4.1 from git for some dev. The new cmake ninja based build which works really well.  I now need a couple of arduino libs and dont have time to port them so I thought why not use arduino as an idf component. Unfortunately it looks like the master is still tied to idf 3 and older gnu make.  Is there an equivalent branch in this project that is compiling against latest idf? 

I managed to get core compiled but SD library which I need isn't compiling. Something todo with fatfs and diskio. Editing the makelists.txt is easy enough to work out but it appears between idf 3 and idf4 maybe the disk and vfs stuff was refactored.

Rob

me-no-dev, 2019-09-24:
yeah... new IDF requires you to specify dependencies and stuff. those are probably the ones missing from CMakeLists.txt. Will open a new branch for new IDF soon. Need to have this stable enough and IDF4 is far from it (in release means)

rbiernat, 2019-09-25:
Its seems ok for me so i will stick with 4.1

I dont think this is a case of component require dependencies but more a refactoring of the diskio stuff. 

Ill do a diff on the 3.2 vs 4.1 branches to see whats changed. I doubt its a hard fix so for now I will do it myself

closed 2019-12-08 by stale[bot]