espressif/arduino-esp32 #3260: get_wakeup_reason() changed values in 1.0.3?

robertgregor, 2019-09-24:
I have migrated my code to the latest stable 1.0.3. But it doesn't work as it worked in the past. The reason is the get_wakeup_reason(). I have this:

byte get_wakeup_reason() {
  esp_sleep_wakeup_cause_t wakeup_reason = esp_sleep_get_wakeup_cause();
  switch(wakeup_reason) {
    case 1  : return 1;
    case 2  : return 2;
    case 3  : return 3;
    case 4  : return 4;
    case 5  : return 5;
    default : return 6;
  }
}
void print_wakeup_reason(){
  byte wakeup_reason = get_wakeup_reason();
  switch(wakeup_reason) {
    case 1  : logSketch('I',PSTR("Wakeup caused by external signal using RTC_IO")); break;
    case 2  : logSketch('I',PSTR("Wakeup caused by external signal using RTC_CNTL")); break;
    case 3  : logSketch('I',PSTR("Wakeup caused by timer")); break;
    case 4  : logSketch('I',PSTR("Wakeup caused by touchpad")); break;
    case 5  : logSketch('I',PSTR("Wakeup caused by ULP program")); break;
    case 6  : logSketch('I',PSTR("Wakeup was not caused by deep sleep")); break;
  }
}

However, now it seems, that i.e. the RTC_IO interrupt (It means external PIN interrupt) has the value 4???

atanisoft, 2019-09-24:
You should not use numbers but instead use the ENUM values defined in esp_sleep.h: https://github.com/espressif/esp-idf/blob/v3.2/components/esp32/include/esp_sleep.h#L57-L67

These will change from version to version as the IDF version may get updated from one release to the next.

robertgregor, 2019-09-24:
Yes, OK, I did and it fixed the issue:

byte get_wakeup_reason() {
  esp_sleep_wakeup_cause_t wakeup_reason = esp_sleep_get_wakeup_cause();
  switch(wakeup_reason) {
    case ESP_SLEEP_WAKEUP_EXT0 : return 1;
    case ESP_SLEEP_WAKEUP_EXT1 : return 2;
    case ESP_SLEEP_WAKEUP_TIMER : return 3;
    case ESP_SLEEP_WAKEUP_TOUCHPAD : return 4;
    case ESP_SLEEP_WAKEUP_ULP : return 5;
    default : return 6;
  }
}
void print_wakeup_reason(){
  byte wakeup_reason = get_wakeup_reason();
  switch(wakeup_reason) {
    case 1  : logSketch('I',PSTR("Wakeup caused by external signal using RTC_IO")); break;
    case 2  : logSketch('I',PSTR("Wakeup caused by external signal using RTC_CNTL")); break;
    case 3  : logSketch('I',PSTR("Wakeup caused by timer")); break;
    case 4  : logSketch('I',PSTR("Wakeup caused by touchpad")); break;
    case 5  : logSketch('I',PSTR("Wakeup caused by ULP program")); break;
    case 6  : logSketch('I',PSTR("Wakeup was not caused by deep sleep")); break;
  }
}

Thanks for the help!!!

closed 2019-09-24 by robertgregor