espressif/arduino-esp32 #3261: ESP32 suddenly stopped recognizing port COM3

ChristosPalisidis, 2019-09-24:
Hello people, 
I've been working for a long time with my esp32 azure IoT Kit and everything was fine, when suddenly, after a firmware update, it stopped recognizing the port COM3
I switched USB cables, i checked the usb-uart drivers and everything was fine. Then i used another board and it was functional with no problems.
What may went wrong? Is it possible that the firmware that i used was dangerous? Personaly i doubt it, as the changes that i made and updated was minor, in simple words, the board use to work on that firmware the previous day.
What can i do if something is fault about the CP2102 usb to uart bridge attached on the board?
Any advices?

lbernstone, 2019-09-24:
Things break.  Your software had no effect on the USB/UART chip.  Be happy that it was a $15 device instead of your water heater.  On your working board, make sure that the cp2102 chip does not get extremely hot in normal use.

ChristosPalisidis, 2019-09-26:
Hello, i have a weird update,
I tried the same firmaware on a new device. The COM port was detectable and i managed to upload the firmware, but when it completed i lost connection with the device and it is not detectable from the port that it is connected any more.
That tells me that the problem is firmware based, but what may be the problem. I was using the device with almost the same firmware for some days past week with no problem. The only thing i've changes was related with an OLED display.
Please help!

atanisoft, 2019-09-26:
@ChristosPalisidis you haven't provided anything to help us help you (no code, no schematic of how you connected things, etc).

I would suspect you are using one of the various GPIO pins that should not be used as they are connected to flash *OR* you are short circuiting something in how you are connecting components together which is leading to the board not starting fully.

ChristosPalisidis, 2019-09-26:
Hi @atanisoft  ! Thanks for the response.
I am connecting my esp32 azure IoT KiT in COM port 3 through USB cable. Something that i would guess that may provoke the problem now that i know that it the problem is firmware based, is that i attach an interrupt to the GPIO 0 (boot button), so as to use it to change the oled displays. 
Was that silly from me? Is the problem reversible to the devices?

atanisoft, 2019-09-26:
I couldn't tell you as there is no telling what your code might be doing wrong as you haven't provided anything to review. If you are using GPIO 0 then you just need to ensure it is not LOW when powering up the esp32 otherwise it will enter bootloader download mode only, even in this mode it should still respond on the PC when plugged in.

ChristosPalisidis, 2019-09-26:
`void setup()
  Serial.begin(115200);

  if(!display.begin(SSD1306_SWITCHCAPVCC, 0x3C)) { 
    Serial.println(F("SSD1306 allocation failed"));
    for(;;); // Don't proceed, loop forever
  }
  display.display();
  delay(2000); // Pause for 2 seconds

  display.clearDisplay();
  
  attachInterrupt(BOOT_BUTTON, DisplayMeasurement, LOW);

  display.setTextSize(3);
  display.setTextColor(WHITE);
  display.setCursor(0,0);
  display.print( "ES");
  display.setTextSize(2);
  display.setCursor(0,20);
  display.print( "SYSTEMS");
  display.setTextSize(1);
  display.setCursor(0,37);
  display.print( "Sensors~IoT");
  drawThermometer();`
 This is the beggining of my setup() function which contains the part that i refer above
The interrupt routine is depicted below:
`void IRAM_ATTR DisplayMeasurement(){
  if(drawFlag=true){
     drawFlag=false;
    drawThermometer();
  }else{
    drawFlag=true;
    drawHygrometer();
  }
}`
The two draw functions above just display something on the OLED.

atanisoft, 2019-09-26:
You shouldn't be trying to access I2C from within the ISR callback, you should instead set a flag and do that call from the loop() function. But aside from that I don't see anything too alarming to indicate that it is failing due to the code.

stickbreaker, 2019-09-26:
@ChristosPalisidis You can't access the OLED from multiple threads without single threading the access.

You code shows you using the main thread to configure/init the display AND send updates to the display in your `DisplayMeasurement()`.  the OLED can only accept ONE command at a time. (also the I2C hardware can only do ONE thing at a time).

stickbreaker, 2019-09-26:
@atanisoft is correct, the I2C bus cannot be used from an interrupt.  The I2C code is not re-entrant, and the i2c protocol/hardware is transaction base.  An existing transaction cannot be paused, new transaction started, new transaction finished, existing transaction resumed.  

You will have to single thread all access to `Wire()`.

Chuck.

ChristosPalisidis, 2019-09-26:
Thanks for the imidiate responses!
Ok i got it, as far as the I2C access concerns. But how can i update a working firmware now, that my device is not detectable from the COM port?

stickbreaker, 2019-09-26:
@ChristosPalisidis can you see boot info on the Serial monitor when you manually reset the board?

stickbreaker, 2019-09-26:
@ChristosPalisidis describe you hardware, (What is connected to each pin).

stickbreaker, 2019-09-26:
specifically pins: from [Esptool wiki](https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection)
*  1,3 (Serial)
* 0 (boot mode selection)
* 2 (enable serial boot loader)
* 12 (FLASH Power Voltage)
* 15 (Boot message enable)
* 6..11 (FLASH SPI bus)
* 16,17 (PSRAM controls)
* En (RESET)
 

Chuck.

ChristosPalisidis, 2019-09-26:
I can not open the serial monitor.Instead,i get the message "Board at COM3 is not available".
There is nothing extra connected to the esp32 azure IoT kit. I just only have the device connected to the COM3 port.

atanisoft, 2019-09-26:
Sounds like it is *NOT* on COM3. Is it showing up possibly on a different port?

stickbreaker, 2019-09-26:
That is a PC operating system failure, It is not recognizing the USB device as com3.  
When you select the com port the Arduino Environment will show you all available com ports.  I have multiple ESP32 connected on my system, the com port name change based on how many boards I have plugged in.  What com ports show up on your system?  @atanisoft is probably correct.  It has been renamed by Windows.

stickbreaker, 2019-09-26:
@ChristosPalisidis 
>I just only have the device connected to the COM3 port.

Are you saying you have the IOT board physically connected through a DB9M plug(RS232 port) to COM3?

Or have you plugged in a USB cable into a USB port and the PC O/S is creating a Virtual Serial port and naming it COM3?

Chuck.

ChristosPalisidis, 2019-09-26:
No other ports are showing in the device manager and this device is the only one connected to the PC. Actually, when i plug it in, it is showing in the device manager in COM POTR 3 for a second and then it loses connection.

ChristosPalisidis, 2019-09-26:
@stickbreaker i have the device connected through a USB cable

stickbreaker, 2019-09-26:
@ChristosPalisidis It could be that you have an illegal clone FT232 chip on the IOT board and the Official FT232 driver "bricked" the illegal clone.

Read this page [FTDI drivers Brick Fake Chips](https://hackaday.com/2016%2F02%2F01%2Fftdi-drivers-break-fake-chips-again%2F)

Chuck.

stickbreaker, 2019-09-26:
do a youtube search on "recovering Bricked Arduinos" . you will find many videos on recovering.

Chuck.

ChristosPalisidis, 2019-09-26:
The esp32 has a cp2102 usb-to-uart bringe built in. But your statement is interesting @stickbreaker.
Thank you very much. I'll update as long as i have something new!

stickbreaker, 2019-09-26:
@ChristosPalisidis Read this page on CP2102 diagnostics [silicon labs](https://www.silabs.com/community/interface/knowledge-base.entry.html/2014/10/03/troubleshooting_apr-10iL)

Chuck.

ChristosPalisidis, 2019-09-27:
Hello guys!
Another question. Is there a way to erase the chip's flash memory at the moment, that i have no connection of the device with my PC?

ChristosPalisidis, 2019-09-27:
Ok guys, problem solved!
I just forced the ESP32 to the serial bootloader mode holding boot button down (GPIO0=LOW) while reseting simultaneously. That made the device detectable again.
I will be more cautious when dealing with the oled display next time!
Thank you all for your time and assistance!

closed 2019-09-27 by ChristosPalisidis