espressif/arduino-esp32 #3264: Serial2 and RockBLOCKv2 Satelite Modem

vangalvin, 2019-09-24:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.3
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    ?no? ?yes? default
Upload Speed:					?115200?
Computer OS:                                        Windows 10

### Description:
I am trying to set up a RockBLOCKv2 Sat modem on an ESP32.

ESP32 WROVER
ESP32 Arduino 1.0.3 (also tried earlier versions with the same result)

The ESP32 keeps rebooting.

I am using the example code from the library
[https://github.com/mikalhart/IridiumSBD](url)

The main error that seems to appear when the modem is started seems to be this.
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.

I tried to change the hardware serial to another set of pins, that did not seem to work. I then tried to use espSoftwareSerial and the had the same issue.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

`#include <IridiumSBD.h>
#include <time.h>
/*
 * Time
 */

#define IridiumSerial Serial2
#define DIAGNOSTICS false // Change this to see diagnostics

// Declare the IridiumSBD object
IridiumSBD modem(IridiumSerial);

void setup()
{
  int err;
  
  // Start the console serial port
  Serial.begin(115200);
  while (!Serial);

  // Start the serial port connected to the satellite modem
  IridiumSerial.begin(19200);

  // If we're powering the device by USB, tell the library to 
  // relax timing constraints waiting for the supercap to recharge.
  modem.setPowerProfile(IridiumSBD::USB_POWER_PROFILE);

  // Begin satellite modem operation
  Serial.println("Starting modem...");
  err = modem.begin();
  if (err != ISBD_SUCCESS)
  {
    Serial.print("Begin failed: error ");
    Serial.println(err);
    if (err == ISBD_NO_MODEM_DETECTED)
      Serial.println("No modem detected: check wiring.");
    return;
  }
}

void loop()
{
   struct tm t;
   int err = modem.getSystemTime(t);
   if (err == ISBD_SUCCESS)
   {
      char buf[32];
      sprintf(buf, "%d-%02d-%02d %02d:%02d:%02d",
         t.tm_year + 1900, t.tm_mon + 1, t.tm_mday, t.tm_hour, t.tm_min, t.tm_sec);
      Serial.print("Iridium time/date is ");
      Serial.println(buf);
   }

   else if (err == ISBD_NO_NETWORK)
   {
      Serial.println("No network detected.  Waiting 10 seconds.");
   }

   else
   {
      Serial.print("Unexpected error ");
      Serial.println(err);
      return;
   }

   // Delay 10 seconds
   delay(10 * 1000UL);
}

#if DIAGNOSTICS
void ISBDConsoleCallback(IridiumSBD *device, char c)
{
  Serial.write(c);
}

void ISBDDiagsCallback(IridiumSBD *device, char c)
{
  Serial.write(c);
}
#endif`

### Debug Messages:
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9564
ho 0 tail 12 room 4
load:0x40080400,len:6320
entry 0x400806a8
Starting modem...
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x00000000  PS      : 0x00060f30  A0      : 0x800d14a2  A1      : 0x3ffb1f00  
A2      : 0x3ffc007c  A3      : 0x3f401074  A4      : 0x00000011  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x800d1058  A9      : 0x3ffb1f00  
A10     : 0x3ffc007c  A11     : 0x00000043  A12     : 0x00000002  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000001c  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb  

Backtrace: 0x00000000:0x3ffb1f00 0x400d149f:0x3ffb1f20 0x400d1682:0x3ffb1f60 0x400d0edf:0x3ffb1f80 0x400d2a43:0x3ffb1fb0 0x4008c5f1:0x3ffb1fd0

Rebooting...

stickbreaker, 2019-09-24:
@vangalvin you need to decode your [backtrace](https://github.com/me-no-dev/EspExceptionDecoder).
Just a guess, but, based on the message "InstrFetchProhibited" I would say your Modem library is calling an uninitialized callback in the `modem.begin()` function.

Chuck.

lbernstone, 2019-09-24:
You should use HardwareSerial, but with a wrover module, you will need to reassign the pins.  I would recommend you use Serial1, and set the pins to GPIO21&22.

stickbreaker, 2019-09-24:
@lbernstone I think he is currently using `Serial2` with default pins 16,17.  At least I can't see anywhere he is defining other pins? 

What is your reasoning for using `Serial1` instead of `Serial2`?  I don't know of any functional difference?

Chuck.

stickbreaker, 2019-09-24:
@lbernstone Sorry, I missed the WROVER comment in the body, I just saw "ESP32 Dev Module" and "PSRAM default".  That is why 17,16 are not available for `Serial2`  If he truly has a WROVER with PSRAM then 16,17 are unavailable.  

Other than the PSRAM default pins, is there any reason to use `Serial1` over `Serial2`?

Chuck.

vangalvin, 2019-09-24:
@stickbreaker I did try decoding the backtrace however for some reason the decoding tool will not work on my machine and it returns noting. 

@lbernstone Using Serial1 as it was caused other errors, however i have not tried to reassign the pins on that as yet so will give that a try.

vangalvin, 2019-09-24:
@stickbreaker Yea, its a WROVER. sorry lack of sleep down here trying to get this thing working. It is supposed to be heading off to Antartica in a couple of weeks and the modem only just turned up.

stickbreaker, 2019-09-24:
@vangalvin yea, use @lbernstone recommendation with:

` IridiumSerial.begin(19200,SERIAL_8N1,21,22); // rx=21, tx=22;`

stickbreaker, 2019-09-24:
@vangalvin Don't you worry about falling off the Earth when you go to the bottom of the World?  Every time I put something on the bottom of my globe, it falls off! :grinning: 

Chuck.

lbernstone, 2019-09-24:
Chuck, the ice keeps things from sliding off the flat earth.
I have a long memory back to when Serial2 had buffer issues.  Probably fine now.

vangalvin, 2019-09-24:
@stickbreaker Bwahaha. Thanks I needed a giggle for the morning :) 
Im more worried about the thing blowing away or melting. Its being stuck on the side of Mt Erebus. wind speed is something like 200Km/h, air temperature gets down to -50 however the ground is +60 degrees. 

I set the port settings in HardwareSerial.cpp. still getting the same result.

stickbreaker, 2019-09-24:
@lbernstone Well, maybe, But I think it has something to do with Solar wind pressure holding them on.  All the pictures I see of Antarctica are always with the Sun high in sky It's never night down there.  And Probably the Wind too as @vangalvin says:
> Its being stuck on the side of Mt Erebus. wind speed is something like 200Km/h, air temperature gets down to -50 however the ground is +60 degrees.

See, Full Sun pressure and Windy! that's why they don't fall off.

Chuck.

stickbreaker, 2019-09-24:
@vangalvin On a more serious note, the "instrFetchProhibit" is telling you the processor is attempthing to load a program instruction from an invalid memory location.  This is usually cause by an uninitialized function pointer.  I saw from your code that you defined two callback, but I never saw were you "installed" them.
```c++
#if DIAGNOSTICS
void ISBDConsoleCallback(IridiumSBD *device, char c)
{
  Serial.write(c);
}

void ISBDDiagsCallback(IridiumSBD *device, char c)
{
  Serial.write(c);
}
#endif
```
Chuck.

vangalvin, 2019-09-24:
lol, I think stuff stays in place just like getting your tongue stuck to an iceblock :)

@stickbreaker that was part of the demo code, thought it best to start with that before I start messing around with other things. 

I removed the modem.setPowerProfile(IridiumSBD::USB_POWER_PROFILE); line and the cyclic reboot stopped so I suspect you are right and whatever it is causing it is located in that routine.

vangalvin, 2019-09-24:
sorry, that was //err = modem.begin(); that I removed.. and everyting in the loop.

stickbreaker, 2019-09-24:
@vangalvin Yea, read through that modem library and see how it uses callbacks. It is probably echoing some diagnostic about the configuration state through the unconfigured functions.

Good luck with reading/understanding someone else's code.
Chuck.

vangalvin, 2019-09-24:
Thanks, I will have a look and see if anyting sticks out. 
This is the library that Rock7 recomend, I was tempted to just use AT commands but handling the binary data that way is a little harder. 
Thanks for all the help!

stickbreaker, 2019-09-24:
@vangalvin I looked through https://github.com/mikalhart/IridiumSBD/ It seems to be that the`#ifdef` in your example needs an `#else` case.

```c++
#if DIAGNOSTICS
void ISBDConsoleCallback(IridiumSBD *device, char c)
{
  Serial.write(c);
}

void ISBDDiagsCallback(IridiumSBD *device, char c)
{
  Serial.write(c);
}
#else
void ISBDConsoleCallback(IridiumSBD *device, char c)
{
 // do nothing
}

void ISBDDiagsCallback(IridiumSBD *device, char c)
{
// do Nothing
}
#endif
```

Chuck.

vangalvin, 2019-09-24:
I will add that however for the moment I just removed the diagnostics and everyting in the loop() leaving just what was in setup() to see if i could isolate what was causing the issue.

stickbreaker, 2019-09-24:
@vangalvin add my code and a `#define DIAGNOSTICS` at the top.  You should see a all the Modem commands echoed through the Serial Monitor.

Chuck.

vangalvin, 2019-09-24:
Well, the second i added it the cyclic boot stopped. :) WOOHOO!!!

stickbreaker, 2019-09-24:
@vangalvin Yep those two callbacks are "weakly" bound in the library.  The library does not provide default functions, so unless the user app creates some, the "uninitialized state" NULL would cause the fault.

Good luck with the rest of your code.

So, if you are happy with this result, you can close this issue.

Chuck.

vangalvin, 2019-09-24:
Thank you so much!!! Just reconnecting a longer cable to see if i can get a signal from the sat :)
will let you know how it goes.

vangalvin, 2019-09-24:
@stickbreaker IT WORKS!!!!!! even got the pass through working!!!!
Sereiously thank you so much!!!!
I owe you a beer!

stickbreaker, 2019-09-24:
@vangalvin Post a picture of the the installation when you get down there!

vangalvin, 2019-09-24:
@stickbreaker I wish i was going, I missed the training this year. so may end up going down next year if I am lucky. they are only doing the one node this year however next year they are planning about 8 more I think. but will be sure to get a picture of the install :)

closed 2019-09-25 by stickbreaker