espressif/arduino-esp32 #3268: ESP32 I2s DMA ADC buffer Interrupt question.

zekageri, 2019-09-25:
### Hello guys!

**I'am using the i2s DMA for reading a 4Khz frequency from a signal generator.
My plan is to plot the measured data to web.
All is working fine.**
 _-I can read the measured ADC data.
 -I can plot it to web via canvasjs._

_The problem is that when i "reading" the adc data from buffer, the data overflows sometimes.
I belive that means that when i read the data from the i2s dma buffer it gets filled again and again while i read from it.
So i think the solution might be **to get an interrupt when the i2s dma buffer is full**, and stop the adc, read the buffer and start it again._

**But i don't know how to setup an interrupt to it and i can't find any working example online.**

### My i2s init code:

```

#define I2S_NUM         (0)
void configure_i2s(){
  i2s_config_t i2s_config = 
    {
    .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN),  // I2S receive mode with ADC
    .sample_rate = samplingFrequency,                                             // sample rate
    .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,                                 // 16 bit I2S
    .channel_format = I2S_CHANNEL_FMT_ALL_LEFT,                                   // only the left channel
    .communication_format = (i2s_comm_format_t)(I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB),   // I2S format
    .intr_alloc_flags = 1,                                                        // none
    .dma_buf_count = 2,                                                           // number of DMA buffers
    .dma_buf_len = NUM_SAMPLES,                                                   // number of samples
    .use_apll = 0,                                                                // no Audio PLL
  };
  adc1_config_channel_atten(ADC_CHANNEL, ADC_ATTEN_11db);
  adc1_config_width(ADC_WIDTH_12Bit);
  i2s_driver_install(I2S_NUM_0, &i2s_config, 0, NULL);
 
  i2s_set_adc_mode(ADC_UNIT_1, ADC_CHANNEL);
  SET_PERI_REG_MASK(SYSCON_SARADC_CTRL2_REG, SYSCON_SARADC_SAR1_INV);
  i2s_adc_enable(I2S_NUM_0);
}
```

### My "reading and plotting" function:

```

#define DUMP_INTERVAL 2000
int32_t  last_millis = DUMP_INTERVAL;

static const inline void ADC_Sampling(){
  uint16_t i2s_read_buff[NUM_SAMPLES];
  String data;
  i2s_read_bytes(I2S_NUM_0, (char*)i2s_read_buff,  NUM_SAMPLES * sizeof(uint16_t), portMAX_DELAY);
  long Millis_Now = millis();
  if (((Millis_Now - Start_Sending_Millis) >= DUMP_INTERVAL)&& Chart_is_ok_to_Send_Data && olvasunk_e)
  {
    Start_Sending_Millis = Millis_Now;
    i2s_adc_disable(I2S_NUM_0);
    for (int i=0;i<NUM_SAMPLES;i++) {
      data = String((i2s_read_buff[i])/(float)40.95);
      data += "}";
      webSocket.broadcastTXT(data.c_str(), data.length());
    }
    i2s_zero_dma_buffer(I2S_NUM_0);
    i2s_adc_enable(I2S_NUM_0);
  }
}
```

Proof from data overflow (attachments). 
(BTW. The signal generator square signals are looking good on an oscilloscope.)

[image]
[image]
[image]

zekageri, 2019-10-02:
Closing this and opening an other one for better explanation.

closed 2019-10-02 by zekageri

Dominik204, 2022-03-09:
hello how did you solve this problem?

zekageri, 2022-03-09:
Hi! I was not able to solve this.
The esp adc is garbage for this.
Also for continous adc sampling the RTOS framework is not suitable.
If you create a task, you must delay at least a few millisecs for task switching.
This delay can kill the readings and you can see it on the graph.

I abandoned this project a long time ago.

Also the adc pins has a lot of noise coming from the wifi and other components. Just no way to work reliably.