espressif/arduino-esp32 #3271: cannot download from Fehler beim Herunterladen von https://github.com/espressif/arduino-esp32/releases/download/package_esp32_dev_index.json Fehler beim Herunterladen von https://dl.espressif.com/dl/package_esp32_dev_index.json Fehler beim Herunterladen von https://github.com/espressif/arduino-esp32/releases/download/package_esp32_index.json

mikewolf, 2019-09-25:
in Arduino ide
Fehler beim Herunterladen von https://github.com/espressif/arduino-esp32/releases/download/package_esp32_dev_index.json
Fehler beim Herunterladen von https://dl.espressif.com/dl/package_esp32_dev_index.json
Fehler beim Herunterladen von https://github.com/espressif/arduino-esp32/releases/download/package_esp32_index.json

reddden, 2019-09-25:
me too but in Russia

pfeerick, 2019-09-25:
In Australia, I can't access `https://dl.espressif.com/dl/package_esp32_index.json` either. If this URL has changed, this means the instructions at [docs/arduino-ide/boards_manager.md](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md) are incorrect. 

I'm not familiar with the other two links, but they don't work for me either. I was, however, able to update to 1.0.3 by using the [package_esp32_index.json
](https://github.com/espressif/arduino-esp32/releases/download/1.0.3/package_esp32_index.json) included in that release.

leng-yue, 2019-09-26:
me too in the United States

atmosuwiryo, 2019-09-26:
Yup, https://dl.espressif.com/dl/ is not accessible.

sogeniusio, 2019-09-26:
Is there a country where this is working?

leng-yue, 2019-09-26:
> Is there a country where this is working?

I think no

me-no-dev, 2019-09-26:
will be fixed today, I've been told.

atmosuwiryo, 2019-09-26:
It's working right now. Should do mirroring of this while alive

pfeerick, 2019-09-26:
Yup, https://dl.espressif.com/dl/package_esp32_dev_index.json is working again. Is there a particular reason it's not hosted in the repo, or in it's own if you don't want updates to it polluting the changelog? Not getting the benefits and reliability of AWS hosting this way... ;)

closed 2019-12-09 by stale[bot]