espressif/arduino-esp32 #3287: v1.0.3: Network stack never recovers without a hard reset after receiving closely-spaced UDP packets

ssilverman, 2019-09-26:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

ssilverman, 2019-09-26:
I did some more testing and for me, it always fails at 32 packets.

ssilverman, 2019-09-26:
After even more testing, even for very slow packet rates, eg. 0.5Hz (`sleep 2` in the bash script), it always stops receiving packets at 32. Is it overrunning the dynamic buffers? Is it a coincidence that CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM is 32?

ssilverman, 2019-09-26:
Yep. When I change this value to something else, the failure always happens at that number of packets. For example, at 43, it fails at 43 packets, etc.

stickbreaker, 2019-09-26:
@ssilverman try calling the packet destructor when you are finished with the packet:
```c++
void onPacket(AsyncUDPPacket &packet) {
  Serial.printf("%d: %d\n", ++counter, packet.length());
  delete packet; // all done free resources
}
```
Chuck.

ssilverman, 2019-09-26:
@stickbreaker That doesn't work and causes crashes. (I changed to `delete &packet`, BTW.)

lbernstone, 2019-09-26:
This is not at about 32 packets, but at 32 packets exactly.  Not even closely spaced.  Are sockets not getting closed?

stickbreaker, 2019-09-26:
@ssilverman 

It look like asynchUdp is not freeing the buffer after you use it.  I added a call to `pbuf_free()` after it calls your `onPacket()` handler.

This is the code where your handler is called:
[asyncUdp.cpp](https://github.com/espressif/arduino-esp32/blob/9bbd720d4c6424af3f3731189c74a7a0e0c55a88/libraries/AsyncUDP/src/AsyncUDP.cpp#L676)

```c++
void AsyncUDP::_recv(udp_pcb *upcb, pbuf *pb, const ip_addr_t *addr, uint16_t port, struct netif * netif)
{
    while(pb != NULL) {
        pbuf * this_pb = pb;
        pb = pb->next;
        this_pb->next = NULL;
        if(_handler) {
            AsyncUDPPacket packet(this, this_pb, addr, port, netif);
            _handler(packet);
++            pbuf_free(this_pb); // try adding this line
        } else {
            pbuf_free(this_pb);
        }
    }
}
```

ssilverman, 2019-09-26:
I solved the problem by changing the `onPacket` method signature to: `void onPacket(AsyncUDPPacket packet)`. i.e. it doesn't use a reference. However, that disagrees with AsyncUDP's definition of `AuPacketHandlerFunction` using a reference.

ssilverman, 2019-09-26:
@lbernstone You're right. I'm adjusting the issue title...

stickbreaker, 2019-09-26:
@ssilverman so, what does your working code look like?

ssilverman, 2019-09-26:
@stickbreaker I changed the `onPacket` function like so:

```c++
void onPacket(AsyncUDPPacket packet) {
  Serial.printf("%d: %d\n", ++counter, packet.length());
}
```

stickbreaker, 2019-09-26:
@ssilverman Yes, that was my question.  
Can you modify `asyncUdp.cpp` by addint the `pbuf_free() ` call and rerun your tests?

ssilverman, 2019-09-26:
Thinking about why (per my update two comments above) the AsyncUDPPacket destructor gets called when I pass a copy, but not the reference. The AsyncUDPPacket object drops out of scope after the handler is called. Why wouldn't the destructor get called?

stickbreaker, 2019-09-26:
@ssilverman that buffer needs to be freed somewhere.  If I am reading the code correctly it only frees the buffer element if there is no callback?  Without a callback how does the data get returned to the user?

my recommendation for `delete packet;` was in error.  

I think the `else` case is the problem.  Can you modify the `asyncUDP.cpp` and see if your results change?

Or, are you saying by changing the function you can receive continuous packet streams?

Chuck.

stickbreaker, 2019-09-26:
@ssilverman 
>Update: Turns out the AsyncUDPPacket destructor calls this, but I wonder why passing the reference doesn't work, but passing a copy does? I mean, after the handler is called, the AsyncUDPPacket object drops out of scope.

the `packet()` object's scope is above your `onPacket()` so it does not loose scope in your function. Scope if from create to deletion. I think `packet()` is a member of the `upd()` object.

lbernstone, 2019-09-26:
Pass pointers, not references.  That makes a copy of the object and holds it open.

stickbreaker, 2019-09-26:
@ssilverman I think I figured it out. `packets()` scope is:
```c++
if(_handler) {
            AsyncUDPPacket packet(this, this_pb, addr, port, netif);
            _handler(packet);
        }
```
yes @lbernstone It finally donned on me!  you are correct, the copy object was the guilty party!
It only took an hour to sink through my head.

Chuck.

lbernstone, 2019-09-26:
The literal definition of a fnord.  It took me 3 looks to see it.

ssilverman, 2019-09-26:
I _think_ I get what's happening:
1. The regular constructor calls `pbuf_ref` and the destructor calls `pbuf_free`. This means that when calling the handler with a reference, it's missing a `pbuf_free` because (and I'm guessing here because I haven't looked at the `pbuf_ref` code yet) there's still one reference count outstanding and so we have a leak.
2. The copy constructor doesn't have an extra reference so when we use the handler with the copied object, `pbuf_free` gets called twice, which would also cause a leak eventually? Once when the copied object gets freed and another time when `packet` drops out of scope.

So:
1. To use the reference version, we need an additional call to `pbuf_free`.
2. To use the copy version, we need an additional call to `pbuf_ref` in the copy constructor?

In other words, no matter which version we're using there's a leak. Does my assessment look correct? If not, which part is incorrect?

stickbreaker, 2019-09-26:
@lbernstone Thank you, you just gave me a new word **fnord**  I had never heard of it before.
@ssilverman you can close this issue if you satisfied. 
Chuck.

ssilverman, 2019-09-26:
@stickbreaker looks like we commented at the same time :)
I still believe there's a leak in both the reference and copy-constructor versions.

ssilverman, 2019-09-26:
Here's my proposed fix:
1. Add a call to `pbuf_ref` in the copy constructor, and
2. Call `pbuf_free` regardless, outside the `if` and not in an `else`.

stickbreaker, 2019-09-26:
@ssilverman 
the definition of the `onPacket()` is :+1: 
```c++
typedef std::function<void(AsyncUDPPacket& packet)> AuPacketHandlerFunction;
```
So, to me the call in `_recv()` should be:
```c++
_handler(&packet);
```
I hate the special/random treatment of objects as compared to standard c types

I know that the base type a object is a pointer but it confuses me.

I don't actually know the correct answer.

stickbreaker, 2019-09-26:
@ssilverman I'm studying your fix.

ssilverman, 2019-09-27:
@stickbreaker the troubling fact is that the code still works even when compiling with a copy constructor for `AuPacketHandlerFunction` instead of the defined reference version. So I think the attempted fix should accommodate both. It is my belief that my above proposed fix solves the problem, but I'd definitely like other opinions.

When writing the copy constructor, it sucks that we can't just create a default and call `pbuf_ref` inside. The code must have all the copying done manually. :)

ssilverman, 2019-09-27:
@stickbreaker (re. https://github.com/espressif/arduino-esp32/issues/3287#issuecomment-535727939). The `&packet` syntax means "take the address of". It's a little confusing when the `&` character is used to declare a reference. In other words the `&` is used for both reference declarations and "take the address of".

stickbreaker, 2019-09-27:
@ssilverman the confusing part for me is:
```c++
void doIt( char * buf){
   Serial.print(buf);
}
void doit( char &buf){
   Serial.print(buf);
}
```

ssilverman, 2019-09-27:
@stickbreaker message me offline, I'd be happy to chat about this. I don't want to dilute this issue too much. :)

ssilverman, 2019-10-01:
I’m failing to see why the logic in #3288 and the actual fix in #3290 wouldn’t work. Could anyone come up with cases where it doesn’t work?

ssilverman, 2019-12-13:
There's still no explanation about why the proposed fix in #3290 doesn't work. The pass-by-reference version in AsyncUDP is still broken.

closed 2020-02-26 by stale[bot]

ssilverman, 2021-08-04:
My fix was taken, but uncredited: https://github.com/espressif/arduino-esp32/pull/3290#issuecomment-821741102