espressif/arduino-esp32 #3294: WiFi.scanNetworks() not working on ESP32 and no viable workaround yet?

zamorae, 2019-09-28:
### Hardware:
Board: ESP32 Dev Module
Core Installation/update date:?
IDE name: Arduino IDE 1.8.5
Flash Frequency: 80Mhz
PSRAM enabled: no
Upload Speed: 921600
Computer OS:  Windows 7 Ultimate SP1

### Description:
Hi all,

I have bumped into a major problem in my project. When trying the WiFi.scanNetworks() operation it crashes the entire system. I tested it using the following chips:
 
 ESP-WROOM-32
 ESP-WROOM-32U

Before someone suggests visiting an existing thread/topic, I have visited all these and no one seems to have a clear workaround, plus all these are closed topics:

#758 Closed
#242 Closed
#791 Closed
#918 Closed
#1617 Closed

The closest work around is to run WiFiscan.ino adding a couple of lines of code that actually disconnects Wifi then calls a non-existent AP, but for me it doesn't work at all:

### Sketch: 
```
/*
 *  This sketch demonstrates how to scan WiFi networks.
 *  The API is almost the same as with the WiFi Shield library,
 *  the most obvious difference being the different file you need to include:
 */
#include "WiFi.h"

void setup()
{
    Serial.begin(115200);

    // Set WiFi to station mode and disconnect from an AP if it was previously connected
    WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    WiFi.begin("AccessPoint", "Password");  //tried the "lolllool","loooooooool" in #758 too
    delay(100);

    Serial.println("Setup done");
}

void loop()
{
    Serial.println("scan start");

    // WiFi.scanNetworks will return the number of networks found
    int n = WiFi.scanNetworks();
    Serial.println("scan done");
    if (n == 0) {
        Serial.println("no networks found");
    } else {
        Serial.print(n);
        Serial.println(" networks found");
        for (int i = 0; i < n; ++i) {
            // Print SSID and RSSI for each network found
            Serial.print(i + 1);
            Serial.print(": ");
            Serial.print(WiFi.SSID(i));
            Serial.print(" (");
            Serial.print(WiFi.RSSI(i));
            Serial.print(")");
            Serial.println((WiFi.encryptionType(i) == WIFI_AUTH_OPEN)?" ":"*");
            delay(10);
        }
    }
    Serial.println("");

    // Wait a bit before scanning again
    delay(5000);
}
```

I keep getting a fatal watchdog error on core 0:

### Debug Messages:
```
⸮E (23527) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (23527) task_wdt:  - IDLE0 (CPU 0)
E (23527) task_wdt: Tasks currently running:
E (23527) task_wdt: CPU 0: wifi
E (23527) task_wdt: CPU 1: IDLE1
E (23527) task_wdt: Aborting.
abort() was called at PC 0x400d9a7f on core 0

Backtrace: 0x4008c7e0:0x3ffbe160 0x4008ca11:0x3ffbe180 0x400d9a7f:0x3ffbe1a0 0x40081789:0x3ffbe1c0 0x40132fb6:0x3ffb5290 0x40133005:0x3ffb52b0 0x401334b5:0x3ffb52d0 0x401303bd:0x3ffb5370 0x40130692:0x3ffb53f0 0x401310fa:0x3ffb5440 0x40131f50:0x3ffb5460 0x40132408:0x3ffb5480 0x400d9382:0x3ffb5540 0x400d9632:0x3ffb5570 0x4010d28d:0x3ffb55a0 0x4010d392:0x3ffb55d0 0x4010d68e:0x3ffb5600 0x4010d82b:0x3ffb5630 0x4008f8cc:0x3ffb5650 0x4008877d:0x3ffb5690

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6400
entry 0x400806a8
E (21412) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (21412) task_wdt:  - IDLE0 (CPU 0)
E (21412) task_wdt: Tasks currently running:
E (21412) task_wdt: CPU 0: wifi
E (21412) task_wdt: CPU 1: IDLE1
E (21412) task_wdt: Aborting.
abort() was called at PC 0x400d9a7f on core 0

Backtrace: 0x4008c7e0:0x3ffbe160 0x4008ca11:0x3ffbe180 0x400d9a7f:0x3ffbe1a0 0x40081789:0x3ffbe1c0 0x40132fb6:0x3ffb5290 0x40133005:0x3ffb52b0 0x401334b5:0x3ffb52d0 0x401303bd:0x3ffb5370 0x40130692:0x3ffb53f0 0x401310fa:0x3ffb5440 0x40131f50:0x3ffb5460 0x40132408:0x3ffb5480 0x400d9382:0x3ffb5540 0x400d9632:0x3ffb5570 0x4010d28d:0x3ffb55a0 0x4010d392:0x3ffb55d0 0x4010d68e:0x3ffb5600 0x4010d82b:0x3ffb5630 0x4008f8cc:0x3ffb5650 0x4008877d:0x3ffb5690
```
Please let me know if someone has tried a different workaround than the ones stated on the previous issues. I find it incredible that such a nice chip as the ESP32 has this major flaw that hinders its potential. I can get nothing from the Espressif forums on this, so I presume this is not a major problem for most people, the previous workarounds work fine for the majority, nobody uses the scan function (really?), or I am doing someting wrong.
 
I see from the previous postings that the issue started with an update on the libraries of the ESP and also that the threads go back to 2017 so its pretty worrisome that nobody is talking of this issue in 2019.

Thanks in advance 
-EZ

a-c-sreedhar-reddy, 2019-09-28:
Why are you going to STA mode when you want AP mode?

atanisoft, 2019-09-28:
please decode your backtrace as I can't reproduce this behavior.

zamorae, 2019-09-28:
> After disconnecting from a station why are you again connecting?
> `WiFi.begin("AccessPoint", "Password");` ?

Yes, this clears out any previous connection to WiFi, then re-connects from 0 using WiFi.begin(). I use the disconnect function along with autoConnect and autoReconnect on ESP8266 and it works fine.

zamorae, 2019-09-28:
@atanisoft 
Here is the backtrace decode:

### Original Backtrace code
Backtrace: 0x4008c7e0:0x3ffbe160 0x4008ca11:0x3ffbe180 0x400da01b:0x3ffbe1a0 0x40081789:0x3ffbe1c0 0x40133552:0x3ffb5290 0x401335a1:0x3ffb52b0 0x40133a51:0x3ffb52d0 0x40130959:0x3ffb5370 0x40130c2e:0x3ffb53f0 0x40131696:0x3ffb5440 0x401324ec:0x3ffb5460 0x401329a4:0x3ffb5480 0x400d991e:0x3ffb5540 0x400d9bce:0x3ffb5570 0x4010d829:0x3ffb55a0 0x4010d92e:0x3ffb55d0 0x4010dc2a:0x3ffb5600 0x4010ddc7:0x3ffb5630 0x4008f8cc:0x3ffb5650 0x4008877d:0x3ffb5690 
### Backtrace decoded:
 Decoding stack results
0x4008c7e0: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x4008ca11: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x400da01b: task_wdt_isr at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/task_wdt.c line 174
0x400d991e: esp_phy_rf_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/phy_init.c line 153
0x400d9bce: esp_phy_load_cal_and_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/phy_init.c line 626
0x4008877d: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143_

atanisoft, 2019-09-28:
It isn't even getting to the wifi scan when it fails. It is very likely failing due to a bad partition table or corrupted entry in nvs. I'd suggest erase flash and reupload

tablatronix, 2019-09-28:
Ditto, also what lib version ?

definetly erase NVS

tablatronix, 2019-09-28:
Reproduced

tablatronix, 2019-09-28:
ok looks like it has more to do with the ap not found background connecting maybe.

```PHP
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
scan done
no networks found

[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
```

tablatronix, 2019-09-28:
SCAN_DONE event never fires

tablatronix, 2019-09-28:
Here you go

`W (8367) wifi: sta_scan: STA is connecting, scan are not allowed!`

tablatronix, 2019-09-28:
Is this related? 
https://github.com/espressif/esp-idf/issues/2125

zamorae, 2019-09-28:
> It isn't even getting to the wifi scan when it fails. It is very likely failing due to a bad partition table or corrupted entry in nvs. I'd suggest erase flash and reupload

I have reuploaded the file on both chips several times. Is there any other way to clear the NVS? or do you mean format the chip somehow?

zamorae, 2019-09-28:
> Is this related?
> [espressif/esp-idf#2125](https://github.com/espressif/esp-idf/issues/2125)

Not sure. I cannot really get into scanning, an the espressif/esp-idf#2125 talks about AP not disconnecting after being powered off.
@tablatronix  I am using a version of you scanNetworks program that works quite reliably on the ESP8266 but ESP32 even using the original WiFiScan.ino should yield results. Do you think it is a hardware related issue?

zamorae, 2019-09-28:
> Is this related?
> [espressif/esp-idf#2125](https://github.com/espressif/esp-idf/issues/2125)

This threads talk about the same topic, but these are closed w/no clear fix:

#758 Closed
#242 Closed
#791 Closed
#918 Closed
#1617 Closed

lbernstone, 2019-09-28:
[erasing the flash](https://github.com/espressif/esptool#erase-flash-erase_flash--erase-region)

tablatronix, 2019-09-28:
you have to disconnect before scanning

you can reconnect after 

```
    // WiFi.disconnect(); // esp_wifi_disconnect();
    int n = WiFi.scanNetworks();
```

tablatronix, 2019-09-28:
you should be checking int N for `WIFI_SCAN_FAILED`

zamorae, 2019-09-28:
> you should be checking int N for `WIFI_SCAN_FAILED` also
> 
> #define WIFI_SCAN_RUNNING (-1)
> #define WIFI_SCAN_FAILED (-2)
> 
> ```c++
>     int n = WiFi.scanNetworks();
>     Serial.println("scan done");
>     if(n == WIFI_SCAN_FAILED) Serial.println("Scan Failed!");
>     if (n = 0) {
>         Serial.println("no networks found");
>     } else if(n>0) {
>         Serial.print(n);
>         Serial.println(" networks found");
> ```

Ok, but I think the fatal error occurrs before being able to produce any meaningful output (reaching the WiFi.scanNetworks() line). 

Added the code to search for scan failed and result is the same. Backtrace is:

Decoding stack results
0x4008c7e0: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x4008ca11: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x400d9e4b: task_wdt_isr at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/task_wdt.c line 174
0x400d974e: esp_phy_rf_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/phy_init.c line 153
0x400d99fe: esp_phy_load_cal_and_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/phy_init.c line 626
0x4008877d: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

tablatronix, 2019-09-28:
Yeah I cannot reproduce that part.
Did you try to disconnect first or not use wifi at all and remove begin ?

zamorae, 2019-09-28:
> Yeah I cannot reproduce that part.
> Did you try to disconnect first or not use wifi at all and remove begin ?

Yes I've tried it only using  WiFi.scanNetworks() without the disconnect() and no WiFi.begin. I only placed it as a workaround proposed on #758 and @242

zamorae, 2019-09-28:
@atanisoft
Will erase NVS and report back, as I am on mobile now. The chips execute code correctly with any other program, but I have not tried direct wifi connection to a known ssid, will try that shortly too

atanisoft, 2019-09-29:
If you skip the call to WiFi.begin() it should work. It is 100% untrue that you must disconnect() for scanNetworks() to work. The only part that is critical is that you are not actively connecting to an SSID as the scan will be aborted and an error returned.

tablatronix, 2019-09-29:
If you are connecting a non existent ap then yeah you have to disconnect first.. I thought that was what we are talking about??? Because it is always connecting..

atanisoft, 2019-09-29:
If you are trying to connect to a network (existing or otherwise) you *WILL* need a disconnect *IF* the connection process has not completed. If you skip the call to WiFi.begin() entirely then it won't be trying to connect to a network and therefore the WiFi.scanNetworks() call *WILL* work as expected. The problem with the code in this issue is that it is forcing a connect to a non-existing network and after 10seconds (default timeout inside WiFi.begin() call) it is still not connected and it keeps trying to connect in the background since it can't find the requested network. Therefore the call to WiFi.scanNetworks() *WILL* fail and *SHOULD* return an error code accordingly, since the code does no error checking whatsoever (WiFi.begin() will return the connection status!) the code is missing the returned error codes. The for loop in loop() is being short circuited as the return code from WiFi.scanNetworks() is less than zero ([ref](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiType.h#L26-L27)). The loop() code is only explicitly checking for ZERO networks found and otherwise assuming that something was found which is incorrect since the scan request failed.

zamorae, 2019-09-29:
> If you are trying to connect to a network (existing or otherwise) you _WILL_ need a disconnect _IF_ the connection process has not completed. If you skip the call to WiFi.begin() entirely then it [...]

I skip the WiFi.begin() and just leave the disconnect(), skip the WiFimode as well, skip everything including WiFi.disconnect(), and the results are the same. The system aborts before I can get an error code.

This code below considers the following conditions:
**_a) define WIFI_SCAN_RUNNING (-1) disabled_**
**_b) WiFi.mode(WIFI_STA) disabled_**
**_c) WiFi.begin() disabled_**

I re-arranged the example loops for more clarity. You can check that the -2 condition is inspected on the 1st loop therefore there shouldn't be an exception thrown for negatives. The backtrace is attached at the end:

```
/*
 *  This sketch demonstrates how to scan WiFi networks.
 *  The API is almost the same as with the WiFi Shield library,
 *  the most obvious difference being the different file you need to include:
 */
#include "WiFi.h"
//#define WIFI_SCAN_RUNNING (-1)
#define WIFI_SCAN_FAILED (-2)

void setup()
{
    Serial.begin(115200);
    //Set WiFi to station mode and disconnect from an AP if it was previously connected
    //WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    delay(100);

    Serial.println("Setup done");
}

void loop()
{
    Serial.println("scan start");

    // WiFi.scanNetworks will return the number of networks found

    int n = WiFi.scanNetworks();
    Serial.println("scan done");
    if(n == WIFI_SCAN_FAILED)     //<-- This code checks the -2 condition
      {
        Serial.println("Scan Failed!");
      }
      else
        {
          if(n == 0) 
            {
              Serial.println("no networks found");
            } 
            else 
              if(n>0) 
                {
                  Serial.print(n);
                  Serial.println(" networks found");
        
                  for (int i = 0; i < n; ++i) {
                      // Print SSID and RSSI for each network found
                      Serial.print(i + 1);
                      Serial.print(": ");
                      Serial.print(WiFi.SSID(i));
                      Serial.print(" (");
                      Serial.print(WiFi.RSSI(i));
                      Serial.print(")");
                      Serial.println((WiFi.encryptionType(i) == WIFI_AUTH_OPEN)?" ":"*");
                      delay(10);
                  }
                 }
            Serial.println("");
        }

    // Wait a bit before scanning again
    delay(5000);
}
```
I even removed the if conditionals altogether, same situation. See this code:

```
/*
 *  This sketch demonstrates how to scan WiFi networks.
 *  The API is almost the same as with the WiFi Shield library,
 *  the most obvious difference being the different file you need to include:
 */
#include "WiFi.h"
//#define WIFI_SCAN_RUNNING (-1)
#define WIFI_SCAN_FAILED (-2)

void setup()
{
    Serial.begin(115200);
    //Set WiFi to station mode and disconnect from an AP if it was previously connected
    //WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    delay(100);

    Serial.println("Setup done");
}

void loop(){

    Serial.println("scan start");

    // WiFi.scanNetworks will return the number of networks found

    int n = WiFi.scanNetworks();
    Serial.println("scan done");    //<-- Never get this line, means it crashes inside the previous function
    
                  for (int i = 0; i < n; ++i) 
                     {
                      // Print SSID and RSSI for each network found
                      Serial.print(i + 1);
                      Serial.print(": ");
                      Serial.print(WiFi.SSID(i));
                      Serial.print(" (");
                      Serial.print(WiFi.RSSI(i));
                      Serial.print(")");
                      Serial.println((WiFi.encryptionType(i) == WIFI_AUTH_OPEN)?" ":"*");
                      delay(10);
                     }
            Serial.println("");

    // Wait a bit before scanning again
    delay(5000);
}
```

### EXCEPTION DETAIL
scan start
E (21018) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (21018) task_wdt:  - IDLE0 (CPU 0)
E (21018) task_wdt: Tasks currently running:
E (21018) task_wdt: CPU 0: wifi
E (21018) task_wdt: CPU 1: IDLE1
E (21018) task_wdt: Aborting.
abort() was called at PC 0x400d9e4f on core 0

Backtrace: 0x4008c7e0:0x3ffbe160 0x4008ca11:0x3ffbe180 0x400d9e4f:0x3ffbe1a0 0x40081789:0x3ffbe1c0 0x40133386:0x3ffb5290 0x401333d5:0x3ffb52b0 0x40133885:0x3ffb52d0 0x4013078d:0x3ffb5370 0x40130a62:0x3ffb53f0 0x401314ca:0x3ffb5440 0x40132320:0x3ffb5460 0x401327d8:0x3ffb5480 0x400d9752:0x3ffb5540 0x400d9a02:0x3ffb5570 0x4010d65d:0x3ffb55a0 0x4010d762:0x3ffb55d0 0x4010da5e:0x3ffb5600 0x4010dbfb:0x3ffb5630 0x4008f8cc:0x3ffb5650 0x4008877d:0x3ffb5690

###STACK DECODING
Decoding stack results
0x4008c7e0: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x4008ca11: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x400d9e4f: task_wdt_isr at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/task_wdt.c line 174
0x400d9752: esp_phy_rf_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/phy_init.c line 153
0x400d9a02: esp_phy_load_cal_and_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/phy_init.c line 626
0x4008877d: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

atanisoft, 2019-09-29:
You must call WiFi.mode() to setup the call to scanNetworks().

Also you need to check for the other error code as well. Basically if scanNetworks () returns less than zero an error occurred.

Also your backtrace points to a wifi system init failure. Possibly power related even. How are you powering the esp32? It will require at least 600mA during wifi init.

zamorae, 2019-09-29:
@atanisoft 
I call WiFI.mode() and same error. I added a switch to sort out all possible error codes.
```
/*
 *  This sketch demonstrates how to scan WiFi networks.
 *  The API is almost the same as with the WiFi Shield library,
 *  the most obvious difference being the different file you need to include:
 */
#include "WiFi.h"

void setup()
{
    Serial.begin(115200);

    // Set WiFi to station mode and disconnect from an AP if it was previously connected
    WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    delay(100);

    Serial.println("Setup done");
}

void loop()
{
    Serial.println("scan start");

    // WiFi.scanNetworks will return the number of networks found
    int n = WiFi.scanNetworks();
    
 /*WL_CONNECTED: assigned when connected to a WiFi network;
WL_NO_SHIELD: assigned when no WiFi shield is present;
WL_IDLE_STATUS: it is a temporary status assigned when WiFi.begin() is called and remains active until the number of attempts expires (resulting in WL_CONNECT_FAILED) or a connection is established (resulting in WL_CONNECTED);
WL_NO_SSID_AVAIL: assigned when no SSID are available;
WL_SCAN_COMPLETED: assigned when the scan networks is completed;
WL_CONNECT_FAILED: assigned when the connection fails for all the attempts;
WL_CONNECTION_LOST: assigned when the connection is lost;
WL_DISCONNECTED: assigned when disconnected from a network;
*/
    switch(WiFi.status()){
                          case WL_NO_SHIELD: 
                               Serial.println("No WiFi shield is present.");
                               break;
                          case WL_IDLE_STATUS: 
                               Serial.println("Attempting to connect...");
                               break;
                          case WL_NO_SSID_AVAIL: 
                               Serial.println("No SSID available.");
                               break;
                          case WL_SCAN_COMPLETED: 
                               Serial.println("Scan Networks is complete.");
                               break;
                          case WL_CONNECT_FAILED: 
                               Serial.println("Connection FAILED.");
                               break;
                          case WL_CONNECTION_LOST: 
                               Serial.println("Connection LOST.");
                               break;
                          case WL_DISCONNECTED: 
                               Serial.println("Device has been DISCONNECTED from the Network.");
                               break;
                          default: 
                               Serial.println("UNKNOWN ERROR");
                               break;
                        }

                     
                     
    Serial.println("scan done");
    if (n == 0) {
        Serial.println("no networks found");
    } else {
        Serial.print(n);
        Serial.println(" networks found");
        for (int i = 0; i < n; ++i) {
            // Print SSID and RSSI for each network found
            Serial.print(i + 1);
            Serial.print(": ");
            Serial.print(WiFi.SSID(i));
            Serial.print(" (");
            Serial.print(WiFi.RSSI(i));
            Serial.print(")");
            Serial.println((WiFi.encryptionType(i) == WIFI_AUTH_OPEN)?" ":"*");
            delay(10);
        }
    }
    Serial.println("");

    // Wait a bit before scanning again
    delay(5000);
}
```
You may have a point on power. I am using a PC serial port to power the ESP32, but before I had it hooked to a Lab regulated DC Power supply at the Vin and 6.5V of power, and was doing the same. 
Will try to power it again separately using the Lab PS and send the messages via UDP to the PC.

zamorae, 2019-09-29:
@Ibernstone
Installed esptool.py and erased flash as you mentioned. Did not work.
[image]

zamorae, 2019-09-30:
@atanisoft 
Connected mi chip onto the Lab PS directly. Placed code to blink a led if the wifiScan was successful. Went from +6.5V to +8 V but It keeps rebooting.

[image]

 I damaged the chip going up to 11 volts on the power using the ESP32's +5V input. I know, I took the risk. Had another one available so I gave it a go.
The spare one right now its loaded with the code but still resets and has the same behavior, so I think I can safely say its not low power related.

What I am sure at this point its that the WiFi.h has a problem to run on EPS32. The ESP32 chip runs fine if I disable everything but WiFi.disconnect(). As soon as I use WiFi.begin(), WiFi.mode() or WiFi.scan Networks() it starts faulting faulting like crazy.

### Setup() 
WiFi.begin() or WiFi.mode(WIFI_STA) <-- first LED 50 blink test at setup() doesn't happen
WiFi.disconnect() <-- first LED 50 blink test at setup() DOES happen

### Loop()
WiFi.scanNetworks()  <-- Everything falls apart when this method is reached.....

@me-no-dev @tablatronix @copercini @tobozo 
_I'm running out of ideas..._ 
Any other recommendations/suggestions?
Perhaps my settings are incorrect?
Can anyone actually get the ESP32 to scan some networks at al without exceptions?

### The problem in a nutshell: can't run WiFi.scanNetworks on ESP32.

Setttings/Code/Stack trace added below.

[image]

```
/*
 *  This sketch demonstrates how to scan WiFi networks.
 *  The API is almost the same as with the WiFi Shield library,
 *  the most obvious difference being the different file you need to include:
 */
#include "WiFi.h"
//WiFiUDP Udp;                                         //enable Udp service
//unsigned int localPort = 9999;

void setup()
{
    Serial.begin(115200);
    pinMode(13, OUTPUT);

    // Set WiFi to station mode and disconnect from an AP if it was previously connected
    //WiFi.begin();
    WiFi.disconnect();
    //WiFi.mode(WIFI_STA);
    delay(100);
    
    for(int i = 0; i <= 50; i++)
       {
         digitalWrite(13, HIGH);
         Serial.println("13 ON");
         delay(100);
         digitalWrite(13, LOW);
         Serial.println("13 OFF");
         delay(100);
       }
    Serial.println("Setup done");
}

void loop()
{
   Serial.println("scan start");

    // WiFi.scanNetworks will return the number of networks found
    int n = WiFi.scanNetworks();
    
    Serial.println("scan done");
    if (n == 0) {
         Serial.println("no networks found"); 
         digitalWrite(13, HIGH);
         delay(300);
         digitalWrite(13, LOW);
    } 
          else
            {
              Serial.print(n);
              Serial.println(" networks found");
              for (int i = 0; i < n; ++i) 
                  {
                    // Print SSID and RSSI for each network found
                    Serial.print(i + 1);
                    Serial.print(": ");
                    Serial.print(WiFi.SSID(i));
                    Serial.print(" (");
                    Serial.print(WiFi.RSSI(i));
                    Serial.print(")");
                    Serial.println((WiFi.encryptionType(i) == WIFI_AUTH_OPEN)?" ":"*");
                    delay(10);
                  }
                  for(int i = 0; i <= 20; i++)
                     {
                       digitalWrite(13, HIGH);
                       delay(250);
                       digitalWrite(13, LOW);
                       delay(100);
                     }
            }
    Serial.println("");

    // Wait a bit before scanning again
    delay(5000);
}
```
Decoding stack results
```
0x4008c7e0: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x4008ca11: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x400d9ecb: task_wdt_isr at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/task_wdt.c line 174
0x400d97ce: esp_phy_rf_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/phy_init.c line 153
0x400d9a7e: esp_phy_load_cal_and_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/phy_init.c line 626
0x4008877d: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```

zamorae, 2019-09-30:
### Geez... I was looking at the last pictures I posted and I found the problem. Can you see guys what that is?
I'll post it tomorrow in case anyone else wants to point it  out.

_**The module is working fine now**...._

zamorae, 2019-10-01:
This was the problem:
[image]
Solved when switching to a WiFi abled speed:
[image]

Noob mistake...sorry, guys. Thanks for your support!

Closing this issue. Hopefully others will benefit from this.

Regards
-EZ

closed 2019-10-01 by zamorae

lbernstone, 2019-10-01:
I wonder if it is useful or even practical to have guards around WiFi.begin and setCpuFrequencyMHz to give some debug or error around this?

atanisoft, 2019-10-01:
@lbernstone I think it would be a good idea to add guards for that. It will save a lot of headaches of odd failures like this one.

tablatronix, 2019-10-01:
Do not use the dev board definition as the default for esp32, its is asking for trouble, too many options, I use the doit module

peperoca116, 2020-12-15:
This is an old issue but hope this still helps anyone, it is used right after the ESP32 begins. It just works the first time but its enough for me.

I run the following code and works 100% of the times:

    Serial.begin ( BAUD );

    uint8_t WIFI_SSIDs = WiFi.scanNetworks();     
    if( WIFI_SSIDs != WIFI_SCAN_FAILED) {
        if( WIFI_SSIDs == 0 ) Serial.println("No networks found!");
        else {
            Serial.println("  wifi config -> Networks found: ");
            for( uint8_t i = 0; i < WIFI_SSIDs; ++i ){
                Serial.println("    Network -> rssi: " + (String)WiFi.RSSI(i) + "  |  name: " + (String)WiFi.SSID(i));            
            }
        }
    }

    WiFi.mode ( WIFI_STA );
    delay( 1000 );
   WiFi.disconnect( true );
   delay( 1000 ); 
   WiFi.begin();

tablatronix, 2020-12-15:
I was going to test this systematically since I had 30+ aps, and my work replaced all aps and removed 2.4ghz..

shrug

I have been changing my channel and change bandwidth to 20mhz and set country properly, you can try changing mode and power also , having much better luck at everything esp32

patience4711, 2021-02-06:
I have a nicely working captive portal that starts automatically when no wifi connection could be made. That means that a failed wifi.begin() ​​preceded this, and that prevents the scanning. 

It also means that i cannot use wifi.disconnect() because when a grid powerfailure occurs, the esp starts up and cannot connect because the (slow) router has not been started yet. So it enters the accesspoint mode. After a few minutes the portal times out and then a reboot follows so that it can still connect when the router is "in the air".  This would not work when there was a wifi.disconnect executed.

So what i did was when entering the portal: 
  `WiFi.mode(WIFI_OFF); // otherwise wifi,Scannetworks fails
   delay(5000);`
This seems to stop the processes that prevent the scanning.
After that an immediate scan works without problems.

tablatronix, 2021-02-06:
Thats interesting observation, so scannetworks is failing if wifi begin fails? But what is your failure mode? SSID_not avail? or some other bug like WL_DISCONNECT failure to assoc etc.

I cannot think of anything that would be breaking scannetworks on assoc failure.

Seems easy enough to test, ill mess with it