espressif/arduino-esp32 #3295: Ubuntu no module pyserial

comarius, 2019-09-28:
OS:  4.18.0-25-generic #26~18.04.1-Ubuntu
IDE: Arduino 1.8.9

Fresh installed

```
 python -m pip install pyserial
...
You are using pip version 10.0.1, however version 19.2.3 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
 gone wit it
```
Then
```
pip install pyserial
Traceback (most recent call last):
  File "/home/marius/.local/bin/pip", line 7, in <module>
    from pip._internal import main
ModuleNotFoundError: No module named 'pip._internal'
```
Then

```
marius@pegpc:~$ python3 -m pip install pyserial
Collecting pyserial
  Using cached https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl
Installing collected packages: pyserial
Successfully installed pyserial-3.4
```
Then edit esptool.py and did
```
#!/usr/bin/env python3
#
#
```
Still the error persists

```

Traceback (most recent call last):
  File "/home/marius/.arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py", line 39, in <module>
    import serial
ImportError: No module named serial
exit status 1
Error compiling for board ESP32 Wrover Module.
```

lbernstone, 2019-09-28:
Shebang is a hint- the app calling the script does not necessarily respect that.  Did you try installing through apt? `sudo apt install python-serial`

comarius, 2019-09-29:
Resolved.

closed 2019-09-29 by comarius

Kyungje-cho, 2020-11-11:
how to solved this problem ??

hary66, 2020-11-25:
It would have been nice to tell us how you solved this issue !

wrmthorne, 2021-04-05:
For anyone else coming here, I tried everything in [this](https://github.com/espressif/esptool/issues/528) which didn't work for me but the command above (instead for python3) fixed it:
```bash
sudo apt install python3-serial
```
Worked for me on Ubuntu 20.0.4