espressif/arduino-esp32 #3298: WiFiServer class inside another class do not works

lagunax, 2019-09-29:
Hi. When i make cod like
```
file.cpp
...
WiFiServer Server(port);
class SomeClass{
somwhere using Server.begin();
```
in this case all works well. But when i move WiFiServer Server(port) into class like this
```
file.cpp
...
class SomeClass{
WiFiServer Server(port);
somwhere using Server.begin();
```
then connections do not works. Why this happens?

ddwood, 2019-10-09:
So many things.. Does that even compile? 
WiFiServer is a class - not a pointer to a class. 
Server is a private method of SomeClass above is somwhere the call to Server.begin within SomeClass?
If you convert Server to a pointer (WiFiServer *pServer) then you call to new will work but you would call pServer->begin() instead. (from within a SomeClass member function anyway)

lagunax, 2019-10-14:
> So many things.. Does that even compile?
> WiFiServer is a class - not a pointer to a class.
> Server is a private method of SomeClass above is somwhere the call to Server.begin within SomeClass?
> If you [...]

it compiles fine (edited my text above, because it was written fast in a way to work, sorry). but when i try to connect - nothing happens (no errors, no connection, just waiting for connection to server).
now i moved this on namespace but thinking about to continue with class soon and thinking what may be matter for this. Working with FTPServer code - later will place code when will get back for it

lagunax, 2019-10-14:
here is long code^
FtpServer.h
```
#ifndef FTP_SERVERESP_H
#define FTP_SERVERESP_H
#include <Arduino.h>

#ifndef ESP32
	#ifndef ESP8266
		#define ESP32
	#endif
#endif

//#include "Streaming.h"
#ifdef ESP8266
	#include <ESP8266WiFi.h>
#elif defined ESP32
	#include <WiFi.h>
	#include "SPIFFS.h"
#endif
#include <WiFiClient.h>
#include <FS.h>

#define FTP_SERVER_VERSION "FTP-2017-10-18"

#define FTP_CTRL_PORT    21          // Command port on wich server is listening  
#define FTP_DATA_PORT_PASV 50009     // Data port in passive mode

#define FTP_TIME_OUT  5           // Disconnect client after 5 minutes of inactivity
#define FTP_CMD_SIZE 255 + 8 // max size of a command
#define FTP_CWD_SIZE 255 + 8 // max size of a directory name
#define FTP_FIL_SIZE 255     // max size of a file name
#define FTP_BUF_SIZE 2*1460 //512   // size of file buffer for read/write

namespace Service {
	namespace FTP {

		class FtpServer;
		void Start(String user, String pass);
		void Handle();

		class FtpServer
		{
		public:
		  void    begin(String uname, String pword);
		  void    handleFTP();

		private:
		  	  WiFiServer *ftpServer;
		  	  WiFiServer *dataServer;
		  void    iniVariables();
		  void    clientConnected();
		  void    disconnectClient();
		  boolean userIdentity();
		  boolean userPassword();
		  boolean processCommand();
		  boolean dataConnect();
		  boolean doRetrieve();
		  boolean doStore();
		  void    closeTransfer();
		  void    abortTransfer();
		  boolean makePath( char * fullname );
		  boolean makePath( char * fullName, char * param );
		  uint8_t getDateTime( uint16_t * pyear, uint8_t * pmonth, uint8_t * pday,
							   uint8_t * phour, uint8_t * pminute, uint8_t * second );
		  char *  makeDateTimeStr( char * tstr, uint16_t date, uint16_t time );
		  int8_t  readChar();

		  IPAddress      dataIp;              // IP address of client for data
		  WiFiClient client;
[... 1062 lines cut ...]

		char * FtpServer::makeDateTimeStr( char * tstr, uint16_t date, uint16_t time )
		{
		  sprintf( tstr, "%04u%02u%02u%02u%02u%02u",
				   (( date & 0xFE00 ) >> 9 ) + 1980, ( date & 0x01E0 ) >> 5, date & 0x001F,
				   ( time & 0xF800 ) >> 11, ( time & 0x07E0 ) >> 5, ( time & 0x001F ) << 1 );
		  return tstr;
		}

	}
}
```

in this case:
```
$ ftp 192.168.1.5
Connected to 192.168.1.5.
```
and nothing else happens
If i move 
```
	  	  WiFiServer *ftpServer;
		  WiFiServer *dataServer;
```
out of the class, then connection continues with welcome message and user/pass auth.

ddwood, 2019-10-15:
In short: Nothing happens between your WiFiServer being inside or outside of a class.
What changes is ftpServer is allocated in two different memory spaces, When you "new" the ftpServer  (in case two) you are not validating that you successfully allocated the memory (which would create bigger problems that you don't appear to have),  Or possibly you have more than one declaration for the ftpServer name. Has WiFi been successfully started prior to ftpServer being created?

On another note, does anything change when you keep ftp server off the heap?

`class CTest {
  public:
  CTest(uint16_t port) : ftpServer(port) {}
  ~CTest() {}
  void begin() { ftpServer.begin(); }
private:
   WiFiServer ftpServer;
};

CTest aTest(21);
`

lagunax, 2019-10-17:
> Has WiFi been successfully started prior to ftpServer being created

FTP initialization is last thing that i start. and in all cases it is after wifi started.
```
	sys->WFAPInit();
	//sys->tft->println("WiFi");

#ifdef SERVICE_WEB
	Service::HTTP::Start();
	//sys->tft->println("HTTP");
#endif
#ifdef SERVICE_FTP
	Service::FTP::Start("user", "pass");
	//sys->tft->println("FTP");
#endif
```
and this code do not changes in all cases.
1) when i call Start("user", "pass"); - it calls ftpSrv.begin(user,pass);
2) ftpSrv.begin calls iniVariables();
3) iniVariables() initializes 
```
ftpServer=new WiFiServer( FTP_CTRL_PORT );
dataServer=new WiFiServer( FTP_DATA_PORT_PASV );
(if not - will be panic and reboot of device)
```
so, those servers initiates after wifi starts and before use of them in any case (and ftp connections freezes). but maybe i need to initiate it before wifi start... will check it later. because if i make like this:
```
#include "FtpServer.h"

namespace Service {
	namespace FTP {

		WiFiServer ftpServer( FTP_CTRL_PORT );//( FTP_CTRL_PORT );
		WiFiServer dataServer( FTP_DATA_PORT_PASV );//( FTP_DATA_PORT_PASV );
		FtpServer ftpSrv;
```
then ftpServer and dataServer initiates at the beginning of program run (befor wifi start) - and in this case all works well.

closed 2019-12-30 by stale[bot]