espressif/arduino-esp32 #3302: WiFi.scanComplete() always return -2  on ESP32

Rezaasghariir, 2019-09-30:
In esp32-wroom-32 when i want scan WiFi station WiFi.scanComplete() always return -2
my code is :
/*
 *  This sketch demonstrates how to scan WiFi networks.
 *  The API is almost the same as with the WiFi Shield library,
 *  the most obvious difference being the different file you need to include:
 */
#include "WiFi.h"

void setup()
{
    Serial.begin(115200);

    // Set WiFi to station mode and disconnect from an AP if it was previously connected
    WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    delay(100);

    Serial.println("Setup done");
}

void loop()
{
    Serial.println("scan start");

    // WiFi.scanNetworks will return the number of networks found
    int n = WiFi.scanComplete();
    Serial.println("scan done");
    if (n == 0) {
        Serial.println("no networks found");
    } else {
        Serial.print(n);
    }
    Serial.println("");

    // Wait a bit before scanning again
    delay(5000);
}
I thinks its a bug

DmitryBorisenko33, 2019-11-04:
Same situation. 

```
boolean RouterFind(String ssid) {

  int n = WiFi.scanComplete();
  Serial.print("status=");
  Serial.println(n);
  if (n == -2) {                       //Сканирование не было запущено, запускаем
    //WiFi.disconnect();
    //WiFi.begin();
    WiFi.scanNetworks (true, false);   //async, show_hidden
    Serial.println("->start scanning");
    return false;
  }
  if (n == -1) {                       //Сканирование все еще выполняется
    Serial.println("->scanning in progress");
    return false;

  }
  if (n > 0) {
    for (int i = 0; i <= n; i++) {
      if (WiFi.SSID(i) == ssid) {
        Serial.println("router found");
        WiFi.scanDelete();
        return true;
      } else {
        Serial.print(i);
        Serial.print(")");
        Serial.print(ssid);
        Serial.print("<=>");
        Serial.println(WiFi.SSID(i));
      }
    }
    WiFi.scanDelete();
    Serial.println("->scanning deleted");
    return false;
  }
}
```

I see that WiFi.scanComplete() return -2

atanisoft, 2019-11-04:
WiFi.scanComplete() will return WIFI_SCAN_FAILED (-2) *IF* you don't call WiFi.scanNetworks() (with or without arguments). The code which @Rezaasghariir provided doesn't call WiFi.scanNetworks() so it is rightfully returning WIFI_SCAN_FAILED since no scan has been initiated.

DmitryBorisenko33, 2019-11-04:
ok. i tested one more time. My problem is when i call WiFi.scanNetworks (true, false);  first time everything ok. After i make WiFi.scanDelete();. Than (after 30 sec) i again call WiFi.scanNetworks (true, false); and my code stop working. Probably code not going out from WiFi.scanNetworks function.

DmitryBorisenko33, 2019-11-04:
I try many times. When i call WiFi.scanNetworks (true, false) second time, and previous scanning was already completed, all freeze.

And i know exactly - previous scanning completed fully. I have this signal about it: 

`[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 1 - SCAN_DONE`

My full code:

```
bool StartAPMode() {

  Serial.println("WiFi up AP");
  IPAddress apIP(192, 168, 4, 1);
  IPAddress staticGateway(192, 168, 4, 1);
  IPAddress staticSubnet(255, 255, 255, 0);
  // Отключаем WIFI
  WiFi.disconnect();
  // Меняем режим на режим точки доступа
  WiFi.mode(WIFI_AP);
  // Задаем настройки сети
  WiFi.softAPConfig(apIP, staticGateway, staticSubnet);
  //Включаем DNS
  //dnsServer.start(53, "*", apIP);
  // Включаем WIFI в режиме точки доступа с именем и паролем
  // хронящихся в переменных _ssidAP _passwordAP
  String _ssidAP = jsonRead(configSetup, "ssidAP");
  String _passwordAP = jsonRead(configSetup, "passwordAP");
  WiFi.softAP(_ssidAP.c_str(), _passwordAP.c_str());
  jsonWrite(configJson, "ip", apIP.toString());

  //if (jsonReadtoInt(optionJson, "pass_status") != 1) {
    researching_wifi.attach(30, Async_researching_wifi);
  //}
  return true;
}

void Async_researching_wifi() {
  Serial.println("->try find router");
  if (RouterFind(jsonRead(configSetup, "ssid"))) {
    researching_wifi.detach();
    WIFI_init();
    //MQTT_init();
  }
}

boolean RouterFind(String ssid) {

  int n = WiFi.scanComplete();
  
  Serial.print("status=");
  Serial.println(n);
  
  if (n == -2) {                       //Сканирование не было запущено, запускаем  
    Serial.println("->test");
    WiFi.scanNetworks (true, false);   //async, show_hidden
    Serial.println("->start scanning");
    return false;
    
  }
  if (n == -1) {                       //Сканирование все еще выполняется
    Serial.println("->scanning in progress");
    return false;

  }
  if (n > 0) {                         //Найдено несколько сетей
    for (int i = 0; i <= n; i++) {
      if (WiFi.SSID(i) == ssid) {
        Serial.println("router found");
        WiFi.scanDelete();
        return true;
      } else {
        Serial.print(i);
        Serial.print(")");
        Serial.print(ssid);
        Serial.print("<=>");
        Serial.println(WiFi.SSID(i));
      }
    }
    WiFi.scanDelete();
    Serial.println("->scanning deleted");
    return false;
  }
}

void wifi_reset() {
  WiFi.mode(WIFI_STA);
  WiFi.disconnect(true, true);
}

```

My debug information:

```
WiFi up AP
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 14 - AP_STOP
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 14 - AP_STOP
WIFI_init
[I][ArduinoOTA.cpp:130] begin(): OTA server at: esp-async.local:3232
Web_server_init
Failed to obtain time
Time_Init
After loading memory used:37%
->try find router
status=-2
->test
->start scanning
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 1 - SCAN_DONE
->try find router
status=13
0)test<=>MGTS_GPON_7858
1)test<=>MGTS_GPON_1002
2)test<=>RT-WiFi_A78F
3)test<=>MGTS_GPON_7630
4)test<=>Beeline88
5)test<=>AKADO-1DB0
6)test<=>Beeline_2G_F12137
7)test<=>alexsandr
8)test<=>112
9)test<=>Thom_D0047687
10)test<=>MGTS_GPON_E6C7
11)test<=>Beeline-76
12)test<=>default
13)test<=>
->scanning deleted
->try find router
status=-2

```

btreecat, 2019-12-02:
I can confirm that additional calls to 
`WiFi.scanNetworks() `
Return -2 even with 120 seconds between them (plenty of time to let the scan finish). 

Currently this means I can scan once in the setup() method because if I scan again in the loop it will fail. Seems like a bug as you should always be able to rescan.

btreecat, 2020-01-31:
This is still an open issue.

btreecat, 2020-03-31:
Still a problem. Is any one else able to reproduce?

raphael-bmec-co, 2020-05-26:
Seeing the same issue.

everslick, 2021-06-21:
I see the scan failing, if AND ONLY IF, the configured SSID is not available. This used to work in older COREs, but fails with CORE 1.6.0.

The stale bot closing valid issues is REALLY annoying.

everslick, 2021-06-21:
Maybe this information helps someone, who has the same issue like me: https://github.com/espressif/esp-idf/issues/3044#issuecomment-463963433

YordanYanakiev, 2021-06-29:
Still a valid issue, which practically making ESP32 WiFi unreliable.

capedra, 2021-07-10:
@ivankravets @me-no-dev 
Pretty annoying. I'm going back to sync scan since it's really unreliable..

me-no-dev, 2021-07-16:
I can not confirm this issue with the latest master at all.
Example code used:
```cpp
#include "WiFi.h"

void setup()
{
    Serial.begin(115200);

    // Set WiFi to station mode and disconnect from an AP if it was previously connected
    WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    delay(100);

    Serial.println("Setup done");
}

void loop()
{
  int n = WiFi.scanComplete();
  if (n == -2) {
    Serial.println("->enter to scanning function");
    WiFi.scanNetworks (true, false);   //async, show_hidden
    Serial.println("->out of scanning function");
    
  }
  else if (n == -1) {
    delay(100);
  }
  else if (n > 0) {
    for (int i = 0; i <= n; i++) {
      Serial.print(i);
      Serial.print(")");
      Serial.println(WiFi.SSID(i));
    }
    WiFi.scanDelete();
    Serial.println("->scanning deleted");
  }
}
```

closed 2021-07-16 by me-no-dev

YordanYanakiev, 2021-07-16:
If this is referting to 1.0.6 - 1. it does not works for sure. 2. it is very slow.
the setup is - set a thread responsible for the WiFi only.
Setup a WifiMulti.
Run a scan before the Wifi multi.
Go to light sleep.
Resume from light sleep. 
Do a Wifi scan.
repeat the process.
- it hangs up after 1 or 2 repeats, and never returns anymore the networks, and it practically hangsup.
there is a case which it will work for quite longer, and when this happen - it is unknown for how long, yet the issue happens at some point.
Note that returning to 1.0.4 seems like to works if there is applied a few fixes from 1.0.5.
and 1.0.4 is much, much faster for conneting, which is absolutely vital thing when the esp32 jump from light sleep to light sleep in order to save the battery.
note that none of this happens in deep sleep.

kostinalex, 2022-10-18:
For those who still get this error. Maybe I will helps someone. 
Call 
`WiFi.disconnect();`
if you fail to connect the first time after scan. If you do not call it, every scan after will return -2.

WizardCell, 2024-04-02:
> For those who still get this error. Maybe I will help someone. Call `WiFi.disconnect();` if you tried to connect to one of the networks. Otherwise you will get -2 after.

I can vouch that this indeed helps, thanks heaps! @kostinalex

nmhtqkdk123, 2024-04-16:
> For those who still get this error. Maybe I will help someone. Call `WiFi.disconnect();` if you tried to connect to one of the networks. Otherwise you will get -2 after.

This comment help me alot. Many thanks. @kostinalex

brodrigz, 2024-10-01:
when async, scanComplete returns -2 right after logging '[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 1 - SCAN_DONE' how can this issue be closed?

StefanoTesla, 2025-07-11:
@brodrigz  same problem here, first scan 18 networks founded, after moving from AP_STA to STA, if I loose the wifi I tried to call `WiFi.disconnect();` I wait 500ms and I try to rescan, first result is 0, and at the next test, it's always to -2...

My code is this one:

```
    case L_LOOP:
        //check if wifi is disconnected
        if(WiFi.status() != WL_CONNECTED){
            WiFi.disconnect();
            Serial.println("[WiFiMgr L] WiFi connection lost.");
            _loopCycle = L_START_SCAN;
            _lms = millis();
        }
        break;
    case L_WAIT_BEFORE_SCAN:
        if(millis() - _lms > 500){
            _loopCycle = L_WAIT_SCAN;
        }
        break;
    case L_START_SCAN:
        startWiFiScan();
        _loopCycle = L_WAIT_SCAN;

        break;
            
    case L_WAIT_SCAN:
        if(asyncWaitWiFiScan()){
            _loopCycle = L_CHECK_CONFIGURED_WIFI;
        }
        break;

```

```
void startWiFiScan(){
    Serial.println("Scan in progress...");
    WiFi.scanDelete();
    WiFi.scanNetworks(true);
    _scanInProgress = true;
}
```

```
bool asyncWaitWiFiScan(){
    _wifiNetworkFound = WiFi.scanComplete();
    Serial.println(_wifiNetworkFound);
    
    if(_wifiNetworkFound < 0  ){
        return false;
    }
    Serial.print("[WiFiMgr] Scan during AP Completed: ");
    Serial.print(_wifiNetworkFound);
    Serial.println(" network found.");
    _scanInProgress = false;
    return true;
}
```

It doesn't work, but finally I decided to take a bad way, I killed the WiFi calling WiFi.disconnect(true); and set it again as WiFi.mode(WIFI_STA)...

```
    case L_LOOP:
        //check if wifi is disconnected
        if(WiFi.status() != WL_CONNECTED){
            WiFi.disconnect(true); //DISCONNECT WITH TRUE (SHUOLD TURN OFF THE RADIO)
            Serial.println("[WiFiMgr L] WiFi connection lost.");
            _loopCycle = L_START_SCAN;
            _lms = millis();
        }
        break;
    case L_WAIT_BEFORE_SCAN:
        if(millis() - _lms > 500){
            _loopCycle = L_WAIT_SCAN;
            WiFi.mode(WIFI_STA); //CALLING THE WIFI MODE AS STATION
        }
        break;
    case L_START_SCAN:
        startWiFiScan();
        _loopCycle = L_WAIT_SCAN;

        break;
            
    case L_WAIT_SCAN:
        if(asyncWaitWiFiScan()){
            _loopCycle = L_CHECK_CONFIGURED_WIFI;
        }
        break;
```

StefanoTesla, 2025-07-11:
I think there's someting strange in the WiFi library 2.0, WiFi.scanComplete(); retunr -1 for a while, and then change to -2 but after some seconds returns the WiFi lfound number...I'm going to try the latest version of arduino-espressif32....