espressif/arduino-esp32 #3306: esp_partition_get_sha256() mismatches PHP's hash() function in HTTPUpdate

mcpicoli, 2019-09-30:
Disclaimer: It may be that this question belongs to a PHP forum instead of this one. However, this question is _also_ very ESP32/Arduino related. Please correct me if this is the case.
Another disclaimer: I also posted this issue at [arduino.stackexchange.com](https://arduino.stackexchange.com/questions/69014/esp-partition-get-sha256-mismatches-phps-hash-function-in-httpupdate)

The scenario:

 - Implementing self-update routines for my ESP32 project, using HTTPUpdate
Using a server-side PHP script to decide which binary image to serve to the device, or return HTTP 304 return code if no updates available.
 - Using the sketch's MD5 **and** SHA-256 hashes provided by the HTTPUpdate library as request headers, server-side, to check binary version.

The problem:

 - The MD5 hash contained in the request header "x-ESP32-sketch-md5" _is the same_ as the MD5 hash calculated by a simple ```md5()``` call using the raw binary for the currently running version.
 - The SHA256 hash contained in the request header "x-ESP32-sketch-sha256" **is not the same** as the hash obtained by calling PHP's ```hash("sha256", $v)``` (where ```$v``` contains the same raw binary).

Example:

 - MD5 (HTTPUpdate): c93d814989b53a24a0164e561b6a8bf8
 - MD5 (PHP): c93d814989b53a24a0164e561b6a8bf8 (no difference)
 - SHA256 (HTTPUpdate): 5634674253E095600CD863A34287FCA742BCFCF9B5BF409870DD65AF0ADEF7A0
 - SHA256 (PHP): 1ECDDF3215520C79AAD13BBC1D116D9286910490DF316B9B203FCBFC89CBCCC2 (different)

Am I doing something wrong?

Note: The ```esp_partition_get_sha256()``` function is called internally by the ```update()``` method of the HTTPUpdate library.

PHP Code (for calculating the hashes):
```PHP
<?php

$v = file_get_contents("fw.bin");

print("MD5: ". md5($v)."\r\n");
print("SHA256: ".hash("sha256", $v)."\r\n");
```

Arduino code (the relevant part):
```Arduino
// Called once or twice a day
void update_fw()
{
    Serial.println("Checking for updates...");
    
    t_httpUpdate_return ret = httpUpdate.update(client, "http://my_server/fw/check_fw.php");  // For now it only logs the headers and returns HTTP 304 code

    switch (ret)
    {
      case HTTP_UPDATE_FAILED:
        Serial.printf("HTTP_UPDATE_FAILED Error (%d): %s\n", httpUpdate.getLastError(), httpUpdate.getLastErrorString().c_str());
        break;

      case HTTP_UPDATE_NO_UPDATES:
        Serial.println("HTTP_UPDATE_NO_UPDATES");
        break;

      case HTTP_UPDATE_OK:
        Serial.println("HTTP_UPDATE_OK");
        break;
    }
}
```

PHP code (logs headers and returns HTTP 304 code):
```PHP
<?php

$headers = getallheaders();

if (!file_exists("log.txt"))
{
	file_put_contents("log.txt", print_r($headers, true));
}

http_response_code(304);			// No updates.

exit;
```

Thanks in advance.

atanisoft, 2019-09-30:
Can you check if ```hash_file("sha256", "fw.bin");``` returns a different value? If not then this issue may need to be transferred to https://github.com/espressif/esp-idf as the ```esp_partition_get_sha256``` function is implemented there and the HttpUpdate code is leveraging that implementation.

mcpicoli, 2019-09-30:
Unfortunately, the ```hash_file()``` (PHP) function returns the same hash value as the ```hash()``` function (also PHP) - but different from ```esp_partition_get_sha256()``` function (esp-idf?).

atanisoft, 2019-09-30:
I think what is happening is that ```esp_partition_get_sha256()``` is calculating for the entire partition and not just the "used" part of the partition. The MD5 is calculated using only the "used" portion of the active partition.
@me-no-dev thoughts on how to fix this?

me-no-dev, 2019-10-01:
yup exactly.

mcpicoli, 2019-10-01:
Hi, and thanks to all comments.

The mistery is solved.

**_TL;DR_**

MD5 hash is calculated from the whole _file_ uploaded, in my case, "fw.bin".
SHA256 hash is calculated from the app contained inside

**_Long explanation_**

The compiled binary files for the ESP32 are composed (as of 2019-09) of two parts. The first is the app image, and the second is the SHA256 hash for integrity verification. The hash is appended to the end of the app image in binary form.
Legacy compiled binary files do not have this hash appended to the end, and rely on the MD5 hash for integrity verification.

Therefore, since the MD5 hash is calculated _always_ from the whole data, they'll match.
The SHA256 hash _will not match_ because the SHA256 hash of the app image with the hash from this same image appended at the end will _never_ be the same value as the hash itself (excluding theorethical heat-death-of-the-universe-level of probabilities),

However, if the SHA256 hash is stripped from the end of the whole binary, the hash _must_ match. This can be (and was) tested using a simple modification: ```$new_image = substr($whole_image, 0, -32);``` (PHP)

_There is no relation of the hashes (MD5 or SHA256) with the unused flash space._

In the end it seems that the MD5 hash is somewhat a legacy mechanism for firmware image verification and the SHA256 is the currently used one, and the httpUpdate library provides both hashes for legacy uses.

**_How I discovered this?_**

While my research before posting this issue could've been more thorough, since this behaviour [is described](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/storage/spi_flash.html) deep into the ESP-IDF documentation, as pointed by user "Majenko" from [Stackexchange](https://arduino.stackexchange.com/questions/69014/esp-partition-get-sha256-mismatches-phps-hash-function-in-httpupdate), I followed a hunch and opened the firmware file in an hex editor and I soon recognized familiar bytes at the end (the hash).

**_What could be improved?_**

Documentation, of course. While I greatly appreciate the vast contributions of the httpUpdate library maintainers, the documentation is (in my opinion), poor. Maybe it's time for me to start contributing?

I am closing this issue now.

closed 2019-10-01 by mcpicoli

lbernstone, 2019-10-01:
It belongs [here](https://github.com/espressif/esp-idf/blob/master/components/spi_flash/include/esp_partition.h#L297).  While this does mention it in passing, a clear description that the hash is appended to the end would be useful.