espressif/arduino-esp32 #3307: No Arduino options in menuconfig when using as a component of ESP-IDF

wijnsema, 2019-09-30:
When I follow the instructions in

[To use as a component of ESP-IDF](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md)

no additional options for Arduino are show in ```make menuconfig```

This is what I did:
I followed the instructions precisely:

- I used the ESP-IDF Tools Installer version 2.1 to install the master branch of ESP-IDF and other tools. Python 2.7 was already installed.

- Next, I created a project copied  from the ```blink``` example. This builds without problems.

- Then the Arduino component, translated to Windows 10 command prompt.
```
md components
cd components 
git clone https://github.com/espressif/arduino-esp32.git arduino
cd arduino 
git submodule update --init --recursive
cd ..
cd ..
idf.py menuconfig
```
According to the documentation extra options like "Autostart Arduino setup and loop on boot" should appear in menuconfig. However no extra option are shown in menuconfig.

How can I set these options?

Using the latest builds on Windows 10

Kinds regards,
Ronald Wijnsema

lbernstone, 2019-09-30:
arduino-esp32 currently builds against IDF v3.2.  You can use https://github.com/espressif/esp32-arduino-lib-builder/tree/master/tools to get all the necessary components at the right versions.

wijnsema, 2019-10-02:
Thanks for your help!

I installed IDF 3.2, but the result is the same, no extra options in menuconfig.

Is the buildsystem scanning for Kconfig.projbuild files in sub directories or something like that? There is a lot of magic going on, at least with my knowlegde.

Not really sure what to do with the tools you mentioned. These are all Linux shell commands, and I'm running Windows 10. I tried to run them the Windows Subsystem for Linux, but a couple of packages could not be found.

lbernstone, 2019-10-02:
I don't use Windows, since it takes about 4-5x longer to compile code on the same hardware.  However, the IDF is based on msys32, and WSL may not behave the same.  There shouldn't be much more to it than putting arduino in the components folder.  menuconfig will then inherit the options there.  Make sure you are following the directions at https://docs.espressif.com/projects/esp-idf/en/v3.2/get-started/ .

wijnsema, 2019-10-02:
I can confirm that using IDF 3.2 AND the msys32 toolchain indeed solves the problem, Arduino options are shown in menuconfig.

I believe this should be mentioned in the documentation.

Thanks!

closed 2019-10-02 by wijnsema