espressif/arduino-esp32 #3310: ESP32 PICO RFM69 (SPI) - communication problems arduino IDE

LDavidFr, 2019-10-01:
Hello,
I use a esp32 pico with an RFM69.

Everything works fine except with the use of energy saving mode.

Solution 1:
Deel Sleep
The frames are sent but not processed by ESP32 (server).
I restart esp32 server to receive a correct frame. But the next trame are not processed.

Solution 2:
Light Sleep
Same code and lib, I use Light Sleep. It works.
But obligation to reduce energy consumption. So Deep Sleep.

 **SERVER** 

```
#include <RHReliableDatagram.h>
#include <RH_RF69.h>
#include <SPI.h>

#define CLIENT_ADDRESS 2
#define SERVER_ADDRESS 1
RH_RF69 driver(5, 26);
RHReliableDatagram manager(driver, SERVER_ADDRESS);

void setup()
{
  Serial.begin(9600);
  while (!Serial)
    ;
    delay(10);
    if (!manager.init())
    Serial.println("init failed");
    delay(10);
}

uint8_t data[] = "And hello back to you";
uint8_t buf[RH_RF69_MAX_MESSAGE_LEN];

void loop()
{
  manager.temperatureRead();
  if (manager.available())
  {
    uint8_t len = sizeof(buf);
    uint8_t from;
    if (manager.recvfromAck(buf, &len, &from))
    {
      Serial.print("got request from : 0x");
      Serial.print(from, HEX);
      Serial.print(": ");
      Serial.println((char*)buf);
     
      if (!manager.sendtoWait(data, sizeof(data), from)){
        Serial.println("sendtoWait failed");
    }else{
    }
    }
  }
}

```

 **CLIENT DEEP SLEEP**  After reboot SERVER, first trame is good (receive and ack) and next trame KO

```
#include <RHReliableDatagram.h>
#include <RH_RF69.h>
#include <SPI.h>

RTC_DATA_ATTR  uint8_t data[] = "Hello World!";
RTC_DATA_ATTR  uint8_t buf[RH_RF69_MAX_MESSAGE_LEN];
RTC_DATA_ATTR  uint8_t len = sizeof(buf);
RTC_DATA_ATTR  uint8_t from;  

void setup()
{
RH_RF69 driver(5, 13);
RHReliableDatagram manager(driver, 2);
Serial.begin(9600);
  if (!manager.init())
      if (!manager.init())
    Serial.println("init failed");
    Serial.println("Sending to rf69_reliable_datagram_server");
   
  if (manager.sendtoWait(data, sizeof(data), 1))
  {
    if (manager.recvfromAckTimeout(buf, &len, 2000, &from))
    {
      Serial.print("got reply from : 0x");
      Serial.print(from, HEX);
      Serial.print(": ");
      Serial.println((char*)buf);
    }
    else
    {
      Serial.println("No reply, is rf69_reliable_datagram_server running?");
    }
  }
  else
    Serial.println("sendtoWait failed");
    Serial.println("sendtoWait test");
    delay(500);
    esp_sleep_enable_timer_wakeup(5 * 100000);
    esp_deep_sleep_start();
}

void loop()
{
}
```

 **CLIENT LIGHT SLEEP**  Good for all trame (receive and ACK) but consumption is not ideal
```
#include <RHReliableDatagram.h>
#include <RH_RF69.h>
#include <SPI.h>

RTC_DATA_ATTR  uint8_t data[] = "Hello World!";
RTC_DATA_ATTR  uint8_t buf[RH_RF69_MAX_MESSAGE_LEN];
RTC_DATA_ATTR  uint8_t len = sizeof(buf);
RTC_DATA_ATTR  uint8_t from;  
RH_RF69 driver(5, 13);
RHReliableDatagram manager(driver, 2);

void setup()
{
Serial.begin(9600);
}

void loop()
{
  if (!manager.init())
      if (!manager.init())
    Serial.println("init failed");
    Serial.println("Sending to rf69_reliable_datagram_server");
   
  if (manager.sendtoWait(data, sizeof(data), 1))
  {
    if (manager.recvfromAckTimeout(buf, &len, 2000, &from))
    {
      Serial.print("got reply from : 0x");
      Serial.print(from, HEX);
      Serial.print(": ");
      Serial.println((char*)buf);
    }
    else
    {
      Serial.println("No reply, is rf69_reliable_datagram_server running?");
    }
  }
  else
    Serial.println("sendtoWait failed");
    Serial.println("sendtoWait test");
    esp_sleep_enable_timer_wakeup(5000000);
    esp_light_sleep_start();
}

```

me-no-dev, 2019-10-01:
check the reset lines when you are waking up from deep sleep. maybe possible that the RFM is reset?

LDavidFr, 2019-10-01:
Humm

I tested not inclued rf69_reliable_datagram_client
its work. 

You have a idea ?

**CLIENT**
```
#include <SPI.h>
#include <RH_RF69.h>
RH_RF69 rf69(5, 13);
void setup()
{
  Serial.begin(9600);
  if (!rf69.init())
    Serial.println("init failed");
  if (!rf69.setFrequency(433.0))
    Serial.println("setFrequency failed");
  rf69.setTxPower(14, false);

  uint8_t key[] = { 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08,
                    0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08};
  rf69.setEncryptionKey(key);
}
void loop()
{
  Serial.println("Sending to rf69_server");
  uint8_t data[] = "Hello World!";
  rf69.send(data, sizeof(data));
  rf69.waitPacketSent();
  uint8_t buf[RH_RF69_MAX_MESSAGE_LEN];
  uint8_t len = sizeof(buf);
  if (rf69.waitAvailableTimeout(500))
  {
    if (rf69.recv(buf, &len))
    {
      Serial.print("got reply: ");
      Serial.println((char*)buf);
      Serial.print("RSSI: ");
      Serial.println(rf69.lastRssi(), DEC);     
    }
    else
    {
      Serial.println("recv failed");
    }
  }
  else
  {
    Serial.println("No reply, is rf69_server running?");
  }
    esp_sleep_enable_timer_wakeup(100000);
    gpio_config_t io_conf;
    Serial.println("Going to sleep now");
    esp_deep_sleep_start();
   
}
```

**SERVER**
```
#include <SPI.h>
#include <RH_RF69.h>
RH_RF69 rf69(5, 26);

void setup()
{
  Serial.begin(9600);
  if (!rf69.init())
    Serial.println("init failed");
  if (!rf69.setFrequency(433.0))
    Serial.println("setFrequency failed");

  rf69.setTxPower(14, false);
  uint8_t key[] = { 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08,
                    0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08};
  rf69.setEncryptionKey(key);
 
/*
  // For compat with RFM69 Struct_send
  rf69.setModemConfig(RH_RF69::GFSK_Rb250Fd250);
  rf69.setPreambleLength(3);
  uint8_t syncwords[] = { 0x2d, 0x64 };
  rf69.setSyncWords(syncwords, sizeof(syncwords));
  rf69.setEncryptionKey((uint8_t*)"thisIsEncryptKey");
#endif*/
}
void loop()
{
  if (rf69.available())
  {
    // Should be a message for us now  
    uint8_t buf[RH_RF69_MAX_MESSAGE_LEN];
    uint8_t len = sizeof(buf);
    if (rf69.recv(buf, &len))
    {
//      RH_RF69::printBuffer("request: ", buf, len);
      Serial.print("got request: ");
      Serial.println((char*)buf);
      Serial.print("RSSI: ");
      Serial.println(rf69.lastRssi(), DEC);
     
      // Send a reply
      uint8_t data[] = "121212121212;1234565434;60;";
      rf69.send(data, sizeof(data));
      rf69.waitPacketSent();
      Serial.println("Sent a reply");
    }
    else
    {
      Serial.println("recv failed");
    }
  }
}
```

I write mikem @airspayce

closed 2019-12-14 by stale[bot]