espressif/arduino-esp32 #3312: Are interrupts queued after call to noInterrupts()?

alexsr, 2019-10-01:
Hello everyone,
I am using an ESP32 DevKit C to communicate with an IMU.
This external device sends data when a data-ready pin is active. Therefore I trigger an interrupt on this pin.
However, I need to read the data that is sent by the IMU and send it to a different device. This would be susceptible to **race conditions** though.
I was wondering if using `noInterrupts()` followed by a data copy and `interrupts()` works here.
Especially, whether or not an interrupt, that is triggered while `noInterrupts()` is active, is **queued and triggered once interrupts** are enabled again.

Thank you.
Alex

atanisoft, 2019-10-01:
noInterrupts() etc are all non-implemented stubs: https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Arduino.h#L76-L79

Interrupts will still fire even after calling those functions.

alexsr, 2019-10-01:
Oh, that's unfortunate. Thank you though.

closed 2019-10-01 by alexsr