espressif/arduino-esp32 #3317: Simpe WiFiClient example does not work after versiom 1.02

bill-orange, 2019-10-04:
I am trying to run the simple Wifi example after discovering that my more complicated sketch would not connect.  Am I doing something wrong? Is this a know bug?

Board manager 1.02 works fine. there is no connection with board manager 1.03 or 1.0.4.
I am using Arduino IDE 1.8.9 in a Windows 10 environment with  Wemos WiF&Bluetooth Battery (default setting).

The example sketch is probably the shortest possible code to reproduce the error.  There is no error message.

There are no errors, simply a failure to connect.

I did not include the sketch in this post since it is the example sketch with my SSID and Password added.

Is there a fix for this behavior?

stickbreaker, 2019-10-04:
@bill-orange did you erase flash when you upgraded versions?  Sometimes there are incompatible config values stored in NVS. [erase flash using esptool,py](https://github.com/espressif/esptool#erase-flash-erase_flash--erase-region)

Chuck.

bill-orange, 2019-10-04:
Good catch @stickbreaker .  That would not have occurred to me to try for a long time.  

Yes, that fixed it.  User name and password must be stored out of program memory in a slightly different location or format.

closing.....

closed 2019-10-04 by bill-orange